ros-navigation/navigation2 #1317: Add documentation for lifecycle manager

yathartha3, 2019-10-29:
Currently, there is no documentation for lifecycle manager.

shivaang12, 2019-10-29:
Do you mean doxygen? Cause I have open pull request for it.

yathartha3, 2019-10-29:
@shivaang12 Thanks for adding the doxygen.
However, I was not referring to that. I was referring to some documentation about the lifecycle manager and lifecycle nodes in general in the front page readme of the package. I am planning to do that soon.

shivaang12, 2019-10-29:
@yathartha3 No worries!

SteveMacenski, 2019-11-21:
@yathartha3 any update here? I see you self-assigned

yathartha3, 2019-11-21:
@SteveMacenski Hadn't really worked on it. I will write it up this week, or by monday and submit a PR.

SteveMacenski, 2019-11-21:
No, just trying to see if I poked the bear if it would get done faster ;-)

yathartha3, 2020-01-22:
Addressed by #1392

closed 2020-01-22 by yathartha3