ros-navigation/navigation2 #1318: Multi-robot collision avoidance

orduno, 2019-10-29:
I'd like to request your input on the best way to implement collision avoidance between robots.

Background
---
The default LiDAR scanner on the [tb3](http://www.robotis.us/turtlebot-3/) is not sufficient to detect the complete footprint of other robots.

A depth-camera could solve this problem. However, we would do better if robots communicate their position and footprint to others. Possibly, even their current motion or future intention.

Proposal
---
Create a `nav2_costmap_2d::RobotsLayer`.

The layer is provided with a list of robot namespaces to extract footprints from, for example, robot1/footprint, robot2/footprint, etc. The layer marks as occupied the cells used by other robots. This gets squashed, together with the other layers, into the master costmap.

Later, if needed, we could incorporate accounting robots' future locations.

shivaang12, 2019-10-29:
@orduno I think robots communicating their position would be a more of a centralized approach. While trying to avoid obstacles using depth sensor or lidar at an individual level would be a decentralized approach. We should make room for both the approach considering the generality of the applications.

orduno, 2019-10-30:
>  I think robots communicating their position would be a more of a centralized approach

Wouldn't it depend on who they're communicating to? Robots conveying information to other robots would still be considered distributed as there's no centralized component.

yathartha3, 2019-10-30:
I see the proposed `nav2_costmap_2d::RobotsLayer` would be very useful for avoidance and planning.
And especially having some kind of _intent_ such as the robots' future locations could be very useful.
The alternate idea was using TF, which I think can be used in a similar way; i.e., if we know the footprint, of the robot, we can populate the master costmap using the TF data and robot footprint.

I believe for a much broader use case such as graph search for multi-robot path planning, area coverage, orienteering, etc, having a TF tree would actually be more powerful.

shivaang12, 2019-10-30:
> Wouldn't it depend on who they're communicating to? Robots conveying information to other robots would still be considered a distributed architecture as there's no centralized component.

That's actually true! Sorry, I kind off assumed that it would go through central server.

SteveMacenski, 2019-10-30:
I think having located footprints makes most sense (either robot footprint points and then a position, or a set of posed-footprints). I don't want to tie TF to this as that would make it pretty un-useful for real robot applications where they're not going to be running on the same network directly and are probably chatting to each other over a production cloud service that is aggregating and distributing the data. 

Take in `N` topics (like costmaps do) that have some `footprint` type and populate them into the **global** costmap. That **global** is really important because their odometry frames are going to differ, this layer should only work in a globally consistent frame shared by all assets or you're going to be doing alot of annoying transformations and have some information loss

mattcenoo, 2019-10-31:
The social_navigation_layers in [navigation_layers](https://github.com/DLu/navigation_layers.git) could be a good start.

vinnnyr, 2020-06-27:
Is there anyone working on this? I would like to take a shot at this issue, if not.

SteveMacenski, 2021-03-10:
There hasn't been much interest / involvement from folks in awhile around multiple robot features. I'm closing  these tentatively due to this lack of interest. I'm sure if there's groups in the future that would like to consider this, we can open a ticket and scope out the work at that junction. However many of these projects don't very clearly fit in the framework and will have a relatively small impact on the userbase comparatively.

There's still very large and poorly-undefined questions questions around how ROS2 robots should talk to each other over the network to enable something like this beyond a very basic full-simulation demo (e.g. if we make this, how will 5 robots from 5 vendors practically talk to each other? share TF? a custom topic? Something centralized in the cloud? How do we standardize this?) 

As such, closing, since this involves more industry involvement and standardization before we can do it and meaningfully hit the target for a wide range of robots and applications outside of a proof of concept demo

closed 2021-03-10 by SteveMacenski

Parv-Maheshwari, 2021-08-24:
hey @SteveMacenski , I am trying to figure out a method of using other robots' footprint to mark the cells as occupied on the costmap. This would enable me to just not depend on the laser scan data for obstacle avoidance.

I did think of one approach that I could add another laserscan type observation source in the obstacle layer - and I can write a node which converts robots footprints to laserscan data. This would make sure we do not need to change the obstacle layer plugin. Is this a valid idea or should I try to implement something else?

iLanceS, 2021-11-18:
we need this too,but i have no ideal how could i finish it.

ShreekrushnB, 2022-04-05:
I would like to know about Multi-floor Multi-robot coordination.

sandilyasg, 2023-04-24:
> The social_navigation_layers in [navigation_layers](https://github.com/DLu/navigation_layers.git) could be a good start.

How do I use the navigation_layers package? The github page and ROS page both do not have published/subscribed topics information and more