ros-navigation/navigation2 #1325: Change all references from Local Planner -> Controller

SteveMacenski, 2019-11-03:
The local planner actually describes a controller and that nomenclature is widely used in vernacular. 

This also was a point of discussion at ROSCon & universally agreed upon

mikeferguson, 2019-11-04:
+1 from me as well -- much of the code internally in ROS1 navigation actually says "controller" in the poorly named "local planners"

SteveMacenski, 2019-11-04:
I changed a bunch of places, but I greped through and realized there's a few too many places it still exists for me to comfortably say that "[checkmark] did completely remove it".

Changing the pluginlib header name is a little late at this point, but I can have the LocalPlanner header derive from the Controller with an additional depreciation warning for F-turtle.

closed 2019-11-26 by SteveMacenski