ros-navigation/navigation2 #1333: Can't visualize voxels from voxel layer

orduno, 2019-11-06:
## Bug report

**Required Info:**

- Operating System:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue
Follow instructions from:
https://github.com/ros-planning/navigation2/blob/master/nav2_costmap_2d/README.md#to-visualize-the-voxels-in-rviz

#### Expected behavior
We should be able to view voxel markers in RVIZ.

#### Actual behavior
No markers shown in RVIZ.

#### Additional information

crdelsey, 2019-11-07:
@orduno Can you work with Yathartha to figure out what's going on with this?

orduno, 2019-11-07:
*Update:* Didn't realize we're using a robot model without a depth camera, let's try again once #1249 is merged.

mlherd, 2019-11-14:
@orduno #1249 is merged, so it should solve this issue.

SteveMacenski, 2019-11-21:
Has this been brought to conclusion?

closed 2019-11-25 by SteveMacenski