ros-navigation/navigation2 #1339: Add action server feedback to all action tasks

SteveMacenski, 2019-11-11:
Right now none of the action servers provide feedback on the state of the action in progress

orduno, 2019-11-11:
Maybe we can start by discussing the types of feedback messages that make the most sense for each [action type](https://github.com/ros-planning/navigation2/tree/master/nav2_msgs/action)?

SteveMacenski, 2019-11-12:
Backup: distance travelled

spin: angle traveled 

Navigate to pose: time of task elapsed and distance traveled?

Follow path: time elapsed, number of waypoints passed, distance left to goal

Compute path to pose: is very fast, I dont know that would be meaningful

fmrico, 2019-11-14:
In the case of Navigate to pose, maybe this information should be provided by extending `nav2_core::LocalPlanner` with a method to get this information. Something like:

`void getFeedback(rclcpp::Duration * elapsed_time, float * distance_to_goal)` (better distance_to_goal instead of distance_traveled?
or 
`std::tuple<rclcpp::Duration, float> getFeedback()`

and to implement it in dwb_core::DWBLocalPlanner

SteveMacenski, 2019-11-14:
Well I think step 1 is: what are the metrics that would make sense for each task.

Step 2 is: how do we get those metrics.

Anything other than the metrics I posted that you'd want? I know most of those are knee-jerk and there's probably something more insightful than that available

mlherd, 2019-11-14:
+ Estimated time of arrival
+ Task completion percentage
+ Path length
+ Number of performed recovery behaviors
+ Total distance traveled since launch
+ Total travel time since launch

SteveMacenski, 2019-11-14:
> Estimated time of arrival

Cannot be computed with current architecture. We're not using a time dependent planner. We could make a rough guess but it would be so rough I think it wouldn't have alot of direct value. But maybe I'm wrong -- would a really rough estimate be interesting to people?

> Task completion percentage

For which of the actions?

> Path length

Integrating this accurately isn't as trivial as it sounds and if we're replanning often, what's this tell you? Interesting metric, what's your planned use for it?

> Number of performed recovery behaviors

That's a good one. I'm not sure how I'd use it or how we'd track it but that's an interesting metric.

> total [value] since launch

Mhm, this is valuable information, but I don't know that this makes sense to put as a feedback item for the navigation task. Maybe it does. If I have a scheduler calling the action server to go somewhere, does that scheduler need this information or someone else... mhm, I'll have to think about that one.

mlherd, 2019-11-14:
> Estimated time of arrival

I think ETA information can be useful in several different use cases.

- One robot: if the user knows how long the delivery of a package could take using a specific robot and the ETA is much later than what they can tolerate, they can cancel the navigation or use a different robot, etc.

- Multi-robot: Let's say there are several different robots in the building, ETA can be used when picking a robot for the task.

> For which of the actions?

Navigate to pose?

> Path length
 
It is needed for ETA calculation and can be used in the multi-robot scenario. It can be used when checking the current battery level is enough to complete task etc...

> Number of performed recovery behaviors

It can be used to discover which paths cause fewer recovery behaviors.

> Total travel time since launch

total travel time and total distance can tell about the efficiency of the robot with its current configurations

SteveMacenski, 2019-11-14:
> Estimated time of arrival

I was on board with this potential, but if those are the reasons, I'd say not to do it. That information is not what you should be doing for multirobot coordination or scheduling. 

> [paraphrased] completion % for nav to pose

How would you calculate this? if you're replanning and have to go different routes, this is going to fluxuate wildly

> total travel time and total distance can tell about the efficiency of the robot with its current configurations

Ok, but again, is the action caller really the right entity to be receiving that information? Moreover, navigation shouldn't be retaining information from prior action calls. If anything, the action caller should be the one computing and storing that information.

shivaang12, 2019-11-14:
May be considering to make an interface, build on some basic feedback operation. Through this interface we can custom define the way we want to extract this information.

SteveMacenski, 2019-11-14:
This is for a `.action`, we can't build an interface, its going to be hardcoded into a serialized message

mlherd, 2019-11-15:
After thinking about it I came to the conclusion that
- Is Navigation2 the correct place to calculate ETA? Probably not.
- Can we use the ETA information anywhere in Navigation2? I don't know. Maybe for informing the user.
- Should it be calculated at the application level then? Perhaps.

> How would you calculate this?
  - This metric would be hard to calculate accurately especially if the environment is dynamic, but if the environment is static
something like (100*traveled_distance/(traveled distance + current_path_length)) may work with some accuracy. I don't have much experience in this subject, so I can't say anything for sure.

I give up on all the metrics except,
  - **Path length**
  - It is going to fluctuate if the environment is dynamic, but if it is a static controlled environment, the path length could be useful. It could be used in system tests or log data when there is a failure. It could be used if the robot needs to call an elevator or open an automatic door before reach to the goal pose.

- **Number of performed recovery behaviors**
  - I am not sure what it could be useful for. It could be used when dynamically changing the max velocity parameter.

SteveMacenski, 2019-11-15:
So I think very rough ETA information could be useful, but it has to be with the understanding and framing to the user that its VERY rough and shouldn’t be relied against. Similar with % complete. 

They’re interesting prototypes, but I don’t think we can casually include, that needs some real design discussion on how to do well and how its represented to a user. 

I basically always assume every environment is the worst case situation with people everywhere and kids jumping in front of the robot. That opinion was hardened by working on a robot where kids jumped in front of it all the time. Maybe not the best baseline. 

I think the # recoveries is a good metric. Hard to track potentially, but maybe the BT makes that easier? In the general case, how do we mark what’s a “recovery”? It would at minimum require a new base class for recovery BT nodes to use to distinguish

SteveMacenski, 2020-03-07:
Working on this now. So far I'm thinking:

Navigate to pose:
- pose
- time for task
- number of recoveries executed
- distance left (? follow path has, but also useful to an application client I imagine)

Follow path:
- distance left

Compute path:
- None, very short process without a loop

Backup:
- Distance
- time processed

Spin:
- distance
- time processed

Wait:
- time left

closed 2020-05-15 by SteveMacenski