ros-navigation/navigation2 #1342: Move AMCL/map_server into new repository

SteveMacenski, 2019-11-11:
Right now the 2 sore thumbs in the navigation stack are AMCL/map_server. They're pretty use-case specific and really give the impression for a user that navigation is only useable in 2D and with a laser scanner, which isn't true. 

AMCL is the only bit of code here that isn't Apache/BSD so moving it out of this project would be helpful on the licensing front. Additionally, we aren't really touching it as part of the re-architecture.

I propose that I make a new ros-planning repo for AMCL where it will live. This will also help keep the AMCL-tickets in a different place and hopefully get a real make-over / upgrade. Specifically I'd like an alternative localization method that works better and works with 3D lidars.

sweetquiet, 2019-11-12:
I agree with your thoughts and look forward to new localization  methods.

SteveMacenski, 2019-11-12:
As always, PRs welcome with implementations!

SteveMacenski, 2020-08-11:
This is getting more integrated into the stack with the map_server supporting semantic info, 3D maps, etc. So I think these should stay. AMCL is being replaced with a 2D/3D method so we can remove this from the queue.

closed 2020-08-11 by SteveMacenski