ros-navigation/navigation2 #1345: Specifying the order for bringing up or shutting down the nodes.

orduno, 2019-11-11:
`nav2::LifecycleManager` resets/shutdowns the nodes in the inverse order that these were brought up. We currently store the order on a vector:

https://github.com/ros-planning/navigation2/blob/43bace3bbb9ddec4d2b362b56af05957194233b7/nav2_lifecycle_manager/src/lifecycle_manager.cpp#L40-L42

And use the same method for transitioning the nodes, the only difference is what we pass to `reverse_order`:

https://github.com/ros-planning/navigation2/blob/43bace3bbb9ddec4d2b362b56af05957194233b7/nav2_lifecycle_manager/src/lifecycle_manager.cpp#L156-L174

However, that is prone to errors. We actually forgot to set the flag to true here:

https://github.com/ros-planning/navigation2/blob/43bace3bbb9ddec4d2b362b56af05957194233b7/nav2_lifecycle_manager/src/lifecycle_manager.cpp#L215-L216

Should define a more explicit way to transition the nodes in the right order.

crdelsey, 2019-11-12:
Are you saying reverse order is the wrong order? If so, what's the right one?

fmrico, 2020-01-07:
Dear @orduno 

If I have understood correctly, the order (reverse or not) depends on the type of transition. Maybe the order could be implicit in the function:

Something like:

```
LifecycleManager::changeStateForAllNodes(std::uint8_t transition) 
 { 
   if (transition == Transition::TRANSITION_CONFIGURE || 
      transition == Transition::TRANSITION_ACTIVATE) { // and all the transitions to apply in order
     for (auto & node_name : node_names_) { 
       if (!changeStateForNode(node_name, transition)) { 
         return false; 
       } 
     } 
   } else { // the other transitions that must be applied in reverse order
     std::vector<std::string>::reverse_iterator rit; 
     for (rit = node_names_.rbegin(); rit != node_names_.rend(); ++rit) { 
       if (!changeStateForNode(*rit, transition)) { 
         return false; 
       } 
     } 
   } 
  
   return true; 
 } 
```

Would it be a solution?

Best
Francisco

SteveMacenski, 2020-01-13:
@fmrico I think that resolved the concern, yes

orduno, 2020-01-16:
@fmrico could you submit a PR for this change?

fmrico, 2020-01-16:
@orduno Yes. I hope to do it this evening.

closed 2020-01-16 by SteveMacenski