ros-navigation/navigation2 #1346: Start using different default Gazebo world models

mlherd, 2019-11-11:
We have been using the same Turtlebot 3 world model for a very long time. Since now the voxel layer is enabled, I think that we should start using more complex Gazebo worlds. I created a repo with several different candidate Gazebo world files. We can change the Gazebo world every few months or so.

https://github.com/mlherd/gazebo_worlds_models_maps_for_testing_navigation

I'll add more Gazebo world models to the repo.

SteveMacenski, 2019-11-12:
I love those first 3

crdelsey, 2019-11-12:
The question to me is how best to run them. Do we update the  model used in CI, or do we leave CI with just a simple world? Should CI be a thorough test of the system, or is it just a smoke test to catch egregious errors. If the latter, how do we run more thorough tests?

mlherd, 2019-11-12:
My initial thought was leaving CI with TB3 world for now and using some of these worlds in nav2_bringup for a while first at least until we know which worlds are the most suitable for CI testing.

SteveMacenski, 2019-11-12:
I think it would be most ideal to change our basic demo + CI to use one of those 3 worlds. 

Then once there’s some generally liked platform we also change from TB3 to that platform. I think showing a pseudo-realistic room with a (more) useful robot sends good signals to people able the state of the project and its usefulness than a toy example world & robot for their tutorial demos. Plus those worlds could have more realistic obstables placed in them for testing that wouldnt make sense to build on the TB3 world (like people moving around).

SteveMacenski, 2019-11-19:
After some thought @mlherd I think you should add a default map and world to one of these maps. I think leaving CI as it is for now is fine, but as we build more CI tests we migrate to it as well.

mlherd, 2019-11-19:
Ok. I will test one these worlds/maps and submit a PR sometime this week then.

SteveMacenski, 2020-02-15:
Closing out - we don't have the engineering time to support multiple or varying environments. We should stick to what we know for the time being.

If a PR appeared to change the environments that works, I'd merge it, but I don't believe its an action item as a bug or required feature to keep a ticket open for.

closed 2020-02-15 by SteveMacenski