ros-navigation/navigation2 #1351: Move bt plugin registration into their respective headers

SteveMacenski, 2019-11-13:
Allow for single-file definition of a new BT node and plugin, I think that would fully check my box that the behavior tree is modifiable without requiring a hard fork. This has similar methods then with how we deal with pluginlib plugins.

https://github.com/ros-planning/navigation/issues/941

Ah, ye ol' bookmarks finally got me. Mixed up navigation1 and 2

crdelsey, 2019-11-18:
I'm not sure I understand this. Are you wanting to get rid of the .hpp files and move their contents into the .cpp file? For instance, the `SpinAction` class definition in spin_action.hpp could be moved to the spin_action.cpp file and we could get rid of the header.

SteveMacenski, 2019-11-18:
Yes, see PR

closed 2019-11-21 by SteveMacenski