ros-navigation/navigation2 #1353: Move nav2_msgs -> common_interfaces/nav_msgs

SteveMacenski, 2019-11-13:
Right now we have a bunch of duplicated messages in nav2_msgs that are in nav_msgs. Slowly I think we've been deleting them and homologating back, but we've hit the critical mass where we should just move entirely our messages to nav_msgs in common_interfaces.

crdelsey, 2019-11-18:
I'm OK with this. What about the nav_2d_msgs as well?

SteveMacenski, 2019-11-18:
That's been completely isolated to DWB, I think that should just stick where it is.

SteveMacenski, 2019-11-20:
@crdelsey on naming, how attached to you are `ComputePathToPose` vs `ComputePath`?

The `ToPose` seems too much information to me. The `FollowPath` isn't `FollowPathToPose`.

crdelsey, 2019-11-20:
I can see someone wanting to compute other types of paths. They could have a ComputePathToCoverArea, ComputePathToNearestPub, etc. We may never integrate it into the nav stack, but we wouldn't conflict with them. 

I think it's slightly preferable to be specific, but I'm not strongly attached. I'll ping the others here and see how they feel.

FollowPath is a different. We don't really FollowPathToPose. It only receives a path and follows it to the end. It could be FollowPathToEnd, I suppose. Are there any other alternatives to what you might do with a path? FollowPathTillToldToStop for a looping path, FollowPathBackAndForth maybe.

SteveMacenski, 2019-11-21:
You’re going the opposite direction than I intended. I want shorter, cleaner names. Not longer

SteveMacenski, 2019-11-21:
I think these are still actually in enough flux that I think its worth keeping them separated for the moment. I think in a release or two when we've definitely settled on something and have more users we can re-evaluate.

I think for now, the action item is to change all the uses of `nav2_msgs` to their `nav_msgs` counter part when possible (costmap metadata, etc)

closed 2020-06-09 by SteveMacenski