ros-navigation/navigation2 #1355: Does nav2 costmap_2d currently support multiple obstacle layers?

zachfang, 2019-11-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Version or commit hash:
  - ros-dashing-nav2-costmap-2d	0.2.6-1bionic.20191017.211223
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
YAML file used: 
```
global_costmap:
  global_costmap:
    ros__parameters:
      use_sim_time: False
      plugin_names: ["static_layer", "obstacle_layer_1", "obstacle_layer_2", "inflation_layer"]
      plugin_types: ["nav2_costmap_2d::StaticLayer", "nav2_costmap_2d::ObstacleLayer", "nav2_costmap_2d::ObstacleLayer", "nav2_costmap_2d::InflationLayer"]
      always_send_full_costmap: True
      rolling_window: False
      update_frequency: 30.0
      publish_frequency: 1.0
      resolution: 0.05
      global_frame: map
      robot_base_frame: base_link
      footprint: "[[0.5, 0.5], [0.5, -0.5], [-0.5, -0.5], [-0.5, 0.5]]"
      lethal_cost_threshold: 200
      obstacle_layer_1:
        enabled: True
        observation_sources: points_1
        max_obstacle_height: 2.5
        points_1:
          data_type: PointCloud2
          topic: /points_1
          max_obstacle_height: 2.5
          min_obstacle_height: 0.05
          raytrace_range: 10.0
          obstacle_range: 10.0
          clearing: True
          marking: True
      obstacle_layer_2:
        enabled: True
        observation_sources: points_2
        max_obstacle_height: 2.5
        points_2:
          data_type: LaserScan
          topic: /points_2
          max_obstacle_height: 2.5
          min_obstacle_height: 0.0
          raytrace_range: 10.0
          obstacle_range: 10.0
          clearing: True
          marking: True
      inflation_layer:
        inflation_radius: 0.2
        cost_scaling_factor: 10.0
  global_costmap_rclcpp_node:
    ros__parameters:
      use_sim_time: False
local_costmap:
  local_costmap:
    ros__parameters:
      use_sim_time: False
      plugin_names: ["static_layer", "obstacle_layer_1", "obstacle_layer_2", "inflation_layer"]
      plugin_types: ["nav2_costmap_2d::StaticLayer", "nav2_costmap_2d::ObstacleLayer", "nav2_costmap_2d::ObstacleLayer", "nav2_costmap_2d::InflationLayer"]
      always_send_full_costmap: True
      rolling_window: True
      update_frequency: 30.0
      publish_frequency: 3.0
      resolution: 0.05
      global_frame: odom
      robot_base_frame: base_link
      footprint: "[[0.5, 0.5], [0.5, -0.5], [-0.5, -0.5], [-0.5, 0.5]]"
      lethal_cost_threshold: 99
      width: 10
      height: 10
      obstacle_layer_1:
        enabled: True
        observation_sources: points_1
        max_obstacle_height: 2.5
        points_1:
          data_type: PointCloud2
          topic: /points_1
          max_obstacle_height: 2.5
          min_obstacle_height: 0.05
          raytrace_range: 10.0
          obstacle_range: 10.0
          clearing: True
          marking: True
      obstacle_layer_2:
        enabled: True
        observation_sources: points_2
        max_obstacle_height: 2.5
        points_2:
          data_type: LaserScan
          topic: /points_2
          max_obstacle_height: 2.5
          min_obstacle_height: 0.0
          raytrace_range: 10.0
          obstacle_range: 10.0
          clearing: True
          marking: True
      inflation_layer:
        inflation_radius: 0.2
        cost_scaling_factor: 5.0
  local_costmap_rclcpp_node:
    ros__parameters:
      use_sim_time: False
```

#### Expected behavior
The costmap should have two obstacle layers with separate observation sources and inflation layer inflates difference sensor reading in the final local/global costmap. 

#### Actual behavior
Only static and inflation layers are initialized properly, and obstacle layers are missing topic and other parameters.
[image]
(For illustration use only, the robot was not localized correctly.)

```
[dwb_controller-15] [INFO] [local_costmap.local_costmap]: Using plugin "static_layer"
[dwb_controller-15] [INFO] [local_costmap.local_costmap]: Subscribing to the map topic (/map) with transient local durability
[dwb_controller-15] [INFO] [local_costmap.local_costmap]: Using plugin "obstacle_layer_1"
[dwb_controller-15] [INFO] [local_costmap.local_costmap]: Subscribed to Topics: 
[dwb_controller-15] [INFO] [local_costmap.local_costmap]: Using plugin "obstacle_layer_2"
[dwb_controller-15] [INFO] [local_costmap.local_costmap]: Subscribed to Topics: 
[dwb_controller-15] [INFO] [local_costmap.local_costmap]: Using plugin "inflation_layer"
[dwb_controller-15] [INFO] [local_costmap.local_costmap]: StaticLayer: Resizing static layer to 340 X 572 at 0.050000 m/pix
```

```
ros2 param get /local_costmap/local_costmap plugin_names
String values are: ['static_layer', 'obstacle_layer_1', 'obstacle_layer_2', 'inflation_layer']
ros2 param get /local_costmap/local_costmap obstacle_layer_1.enabled
Boolean value is: True
ros2 param get /local_costmap/local_costmap obstacle_layer_1.max_obstacle_height
Double value is: 2.5
ros2 param get /local_costmap/local_costmap obstacle_layer_1.observation_sources
Parameter not set.
```
#### Additional information
Follow ROS2 layer configuration example https://github.com/ros-planning/navigation2/tree/master/nav2_costmap_2d.

By looking into the code, https://github.com/ros-planning/navigation2/blob/master/nav2_costmap_2d/src/costmap_2d_ros.cpp#L93 should not be in the top level as it's for each obstacle layer specifically, also it's not used anywhere in this file.

From here https://github.com/ros-planning/navigation2/blob/dashing-devel/nav2_costmap_2d/plugins/obstacle_layer.cpp#L91, looks like namespacing is not propagate properly and thus obstacle layer is looking for top level observation_sources rather than observation_sources specified for each of them.

mlherd, 2019-11-13:
What robot do you use? Do you use Turtlebot 3 in Gazebo? Can you check if the point cloud messages are published?

zachfang, 2019-11-13:
Hi @mlherd,
Thank you for checking on this. We are using a proprietary robot and the point cloud data is being published from the forward facing laser. The example I have is the above screenshot of the real robot in RVIZ that you can see the sensor is actually outputting data.

I can have one obstacle layer working by specifying `observation_sources` on the top level, however, we are trying to have multiple obstacle layers with different observation_sources and we failed to do so.

SteveMacenski, 2019-11-13:
Ditto on making sure the topics line up and transformations are available to get past the message filter. Make sure to build from source, our last sync was awhile ago and some work has occurred on the costmap side since then.

yathartha3, 2019-11-13:
@zachfang Try with 'data_type: PointCloud2' and same for laserscan

SteveMacenski, 2019-11-13:
Aorry to pile on, but also check the QoS settings of your publisher if you see them publishing with transforms & the above suggestions. 

Yathartha have you tested multiple layers before?

yathartha3, 2019-11-13:
@SteveMacenski I have tested with one obstacle layer, and one voxel layer. But not two obstacle layers.

SteveMacenski, 2019-11-13:
Got time to do that today? If you see the issue, I bet its something stupid simple. If you don't, that at least lets us narrow down @zachfang problem to user-specific or debians. 

I'd do it but I'm working from home today (and my home machine isn't 18.04 with ROS2)

zachfang, 2019-11-13:
Hi @SteveMacenski and @yathartha3 I can get more screenshots for different yaml configurations shortly to show what I think is the problem. Basically, I think currently if you have multiple obstacle layers, they WILL share the same observation_sources because the code only reads the param on the top namespace (/local_costmap/local_costmap/observation_sources) but not the individual obstacle layer (/local_costmap/local_costmap/obstacle_layer_1.observation_sources) which is what it's designed for in ROS1. I can be wrong but this is my current thought.

I will post more screenshots with different config in an hour if that helps. Thanks

SteveMacenski, 2019-11-13:
I don't follow that logic.

The plugins are initialized here: 
https://github.com/ros-planning/navigation2/blob/a227539ea9a9d979baa91e1139cbcf60499ad564/nav2_costmap_2d/src/costmap_2d_ros.cpp#L148

Being given their name, ei `obstacle_layer_2` and then the plugin initialization called here: 
https://github.com/ros-planning/navigation2/blob/c530cd89eb2e23cd4e9928c9df9dcdf21eaa042a/nav2_costmap_2d/src/layer.cpp#L47

That then has the `name_` set. Then its getting the observation sources locally ei `obstacle_layer_2.observation_sources` here:

https://github.com/ros-planning/navigation2/blob/master/nav2_costmap_2d/plugins/obstacle_layer.cpp#L91

To then read in. It looks to me like everything is local. That's the "ROS1 way" you describe, but maybe you see something I'm missing for your point that you think its reading in the global space?

SteveMacenski, 2019-11-13:
woah @yathartha3, not sure if this is the problem they're seeing, but there's a problem here

https://github.com/ros-planning/navigation2/blob/master/nav2_costmap_2d/plugins/obstacle_layer.cpp#L118

We have declared parameters in what looks like the wrong namespace. We declare `source + "." + "topic"` but then get `name_ + "." + source + "." + "topic"`. 

So how the streaming in ROS1 worked is that you could set SOME global variables to be used if a specific observation source didn't provide something. An example is max height or observation persistence, but not required fields like topic or datatype.

SteveMacenski, 2019-11-13:
> By looking into the code, https://github.com/ros-planning/navigation2/blob/master/nav2_costmap_2d/src/costmap_2d_ros.cpp#L93 should not be in the top level as it's for each obstacle layer specifically, also it's not used anywhere in this file.

Correct. That should be removed. 

> From here https://github.com/ros-planning/navigation2/blob/dashing-devel/nav2_costmap_2d/plugins/obstacle_layer.cpp#L91, looks like namespacing is not propagate properly and thus obstacle layer is looking for top level observation_sources rather than observation_sources specified for each of them.

Annnd you are correct. Ignore my first comment. That's 100% the issue.

zachfang, 2019-11-13:
@SteveMacenski I found the branch I was reading is `dashing-devel` rather than `master`.
```
node_->get_parameter(name_ + "." + "observation_sources", topics_string);
```
-> does make sense to me.

SteveMacenski, 2019-11-13:
Yes, it looks like we're on the same page, I just caught that myself.

SteveMacenski, 2019-11-13:
If you would submit a PR to dashing devel, we can fix that up. 

Right now we've moved _so far_ ahead of dashing, its hard for us to cherry pick commits over from master (Eloquent) to Dashing.

zachfang, 2019-11-13:
Thanks @SteveMacenski. Plan is to move to Eloquent for sure and I just want to make sure Nav2 will have multiple obstacle layers functionality in Eloquent. I can also help to address that if everyone's cycle is full.

SteveMacenski, 2019-11-13:
I'm not sure many of us spend too much time thinking about the dashing branch. I think our unofficial stance on it is "fix it only when someone reports something horribly wrong".

yathartha3, 2019-11-13:
@zachfang @SteveMacenski so everything okay now?

zachfang, 2019-11-14:
@yathartha3 Fine by me to close this issue. I will move to Eloquent to test and maybe submit a PR for multiple obstacle layers if it's not there yet. Thank you for all your help!

closed 2019-11-14 by SteveMacenski