ros-navigation/navigation2 #1358: Refactor navfn

shivaang12, 2019-11-14:
I understand that navfn is working properly and all. But I think that it's code is quite old as it is written in old c++ (< c++11) and now that in ROS 2 we can leverage the power of modern c++ we should refactor its code.

Also, there is some code which is not being used any where. Refactor can help reduced the code and that way we would have less code to maintain.

Let me know if you think the same.

SteveMacenski, 2019-11-14:
I dont think that's a wise move. People have tried and failed (enter: `global_planner` in ros1). Its a huge and buggy project with not much marginal benefit. I'd agree if there was anyone that wanted to change the navfn library, but I think its not going to be modified ever again.

(plus I just scheduled 6 weeks in early 2020 to delete navfn and replace with hybrid A* with plugin motion models)

shivaang12, 2019-11-14:
Actually, I think replacing is fine too. Hybrid A* sounds cool already. Let me know if you need any help on that.

closed 2019-11-14 by SteveMacenski