ros-navigation/navigation2 #1360: Travis build frequently times out now

crdelsey, 2019-11-14:
## Bug report

The Travis build now times out frequently, exceeding the 50 min limit. I tried reducing the number of tests it runs, but it even times out just doing the build step.

To continue using Travis, we either need to reduce the amount of code we build, or figure out how to cache build objects.

The question is, do we keep making Travis work or is it time to stop using it.

SteveMacenski, 2019-11-14:
+1 for getting rid of it at this point

SteveMacenski, 2019-11-14:
Update name from discussion

mkhansen-intel, 2019-11-15:
I just turned it off in the Travis settings. We shouldn't see it build any more. I think all that needs to be done is remove the status badge from the front 'README' page and optionally remove the .travis.yml file.

PR #1364 (2019-11-18): changing docker file & removing travis

---

## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | fixes #1360  |
| Primary OS tested on | (Ubuntu, MacOS, Windows) |
| Robotic platform tested on | (Steve's Robot, gazebo simulation of Tally, hardware turtlebot) |

---

## Description of contribution in a few bullet points
* Changed docker references from my docker images I set up to the rosplanning version (can't believe that was still there, I did that so long ago)
* Removed travis badge and travis file

---

## Future work that may be re [... cut]
closed 2019-11-18 by SteveMacenski