ros-navigation/navigation2 #1362: Unable to build Navigation2 Dependencies

srsidd, 2019-11-15:
## Bug report
I followed instructions from [the wiki](https://github.com/ros-planning/navigation2/blob/master/doc/BUILD.md). I did a src build of the latest ros2 from [here](https://index.ros.org/doc/ros2/Installation/Eloquent/Linux-Development-Setup/#linux-latest). Tried the `ros1_bridge`, and am able to send msgs back and forth, which leads me to believe my `ros2` installation is ok.. However I'm having trouble building `navigation2.`

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Version or commit hash:
  - ```
	$ vcs custom --args rev-parse HEAD
	.....
	=== ./BehaviorTree.CPP (git) ===
	47d770c38623be39b677efa0a6e164a9085500e3
	=== ./angles (git) ===
	60cdaa34b81b2dba7938546f011db1ac09238339
	=== ./gazebo_ros_pkgs (git) ===
	c7ea2dbb2ffa52b60e308fd0a344f5b4f6fabdcf
	=== ./image_common (git) ===
	e3f482846435eb65e299e9e5955788dc5bcf73c6
	=== ./vision_opencv (git) ===
	bb2ee014d1f5683d00d2edfe7712cf03185ebd6b
    ```
- DDS implementation:
  -  RTI Connext

#### Steps to reproduce issue

```
$ mkdir -p ~/ros2_nav_dependencies_ws/src
$ cd ~/ros2_nav_dependencies_ws
$ wget https://raw.githubusercontent.com/ros-planning/navigation2/master/tools/ros2_dependencies.repos
$ vcs import src < ros2_dependencies.repos
$ rosdep install -y -r -q --from-paths src --ignore-src --rosdistro <ros2-distro-name>
e.g- ros2-distro-name = dashing
$ colcon build --symlink-install
```

#### Expected behavior
Code should compile and build should succeed.
 
#### Actual behavior
Build fails with the following error -
```---
Failed   <<< image_transport	[ Exited with code 1 ]
Aborted  <<< camera_calibration_parsers                                                    
--- stderr: image_geometry                                                                              
In file included from /media/exteral_hdd/sidd/workspace/ros2/ros2_nav/deps_ws/src/vision_opencv/image_geometry/include/image_geometry/stereo_camera_model.h:4:0,
                 from /media/exteral_hdd/sidd/workspace/ros2/ros2_nav/deps_ws/src/vision_opencv/image_geometry/src/stereo_camera_model.cpp:1:
/media/exteral_hdd/sidd/workspace/ros2/ros2_nav/deps_ws/src/vision_opencv/image_geometry/include/image_geometry/pinhole_camera_model.h:6:10: fatal error: sensor_msgs/msg/camera_info.hpp: No such file or directory
 #include <sensor_msgs/msg/camera_info.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/image_geometry.dir/src/stereo_camera_model.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /media/exteral_hdd/sidd/workspace/ros2/ros2_nav/deps_ws/src/vision_opencv/image_geometry/src/pinhole_camera_model.cpp:1:0:
/media/exteral_hdd/sidd/workspace/ros2/ros2_nav/deps_ws/src/vision_opencv/image_geometry/include/image_geometry/pinhole_camera_model.h:6:10: fatal error: sensor_msgs/msg/camera_info.hpp: No such file or directory
 #include <sensor_msgs/msg/camera_info.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/image_geometry.dir/src/pinhole_camera_model.cpp.o] Error 1
make[1]: *** [CMakeFiles/image_geometry.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
---
```
#### Additional information

SteveMacenski, 2019-11-15:
Try fresh using our script after sourcing /opt/ros/dashing:

https://github.com/ros-planning/navigation2/blob/master/tools/initial_ros_setup.sh

mlherd, 2019-11-15:
@srsidd did you set the rosdistro to eloquent
- rosdep install -y -r -q --from-paths src --ignore-src --rosdistro eloquent

srsidd, 2019-11-15:
@SteveMacenski I'm using `eloquent` built from src. I will give the script a try.

@mlherd Yes, from my bash history I have `rosdep install -y -r -q --from-paths src --ignore-src --rosdistro eloquent`.

mlherd, 2019-11-15:
@srsidd the build instructions should work too. I successfully built and tested everything last night.

The build fails because it can't find `/sensor_msgs/msg/camera_info.hpp`. It is part of `ros2/common_interfaces`. I would say make sure that you built ROS2 successfully and sourced your ros2 workspace.

mlherd, 2019-11-19:
@srsidd were you able to build Navigation2?

mlherd, 2019-11-20:
@srsidd you can check the recommended steps at #1369 which is a build failure issue for the same package.
I close this issue but feel free to reopen this issue if you still can't build the stack successfully.

closed 2019-11-20 by mlherd

srsidd, 2019-11-20:
@mlherd Sorry abt the delay, I couldn't get to this till today. I was able to build the navigation stack and the `turtlebot3_ws` without any errors. I think it had something to do with the way I was sourcing my workspaces, (I have multiple ros1 overlays, and I think the env variables were fighting with each other). If I can find something more concrete, I will post it here.