ros-navigation/navigation2 #1363: Planner and Controller server die with message - [PARTICIPANT Error] Type : sensor_msgs::msg::dds_::PointCloud2_ Not Registered

crdelsey, 2019-11-15:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Version or commit hash:
  - f80356380f765 (master from 2 days ago - Wednesday)
- DDS implementation:
  - FastRTPS

#### Steps to reproduce issue
Run the nav stack interactively or through the system tests. Cause "Clear Costmap" recoveries to happen

#### Expected behavior
Costmap gets cleared.

#### Actual behavior

Within about 5 minutes of testing, I get an error like below. I've seen this come from both the "controller server" and the "planner server"

```
13: [controller_server-7] [INFO] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
13: [controller_server-7] 2019-11-15 14:12:30.357 [PARTICIPANT Error] Type : sensor_msgs::msg::dds_::PointCloud2_ Not Registered -> Function createSubscriber
13: [ERROR] [controller_server-7]: process has died [pid 17331, exit code -11, cmd '/home/crdelsey/src/ros_dev_workspace/navigation2_ws/install/lib/nav2_controller/controller_server --ros-args --params-file /tmp/tmpby1s9gda        '].
```

#### Additional information

yathartha3, 2019-11-18:
This might be related to the voxel layer PR since we start using point cloud from that point on.

crdelsey, 2019-11-19:
You are correct. If I disable the voxel layer, this seems to go away.

SteveMacenski, 2019-11-21:
Does this appear outside of fastRTPS? I suspect this is a RMW layer bug since this is happening on a layer reset which is unsubscribing and resubscribing, clearly the message did at one point exist.

SteveMacenski, 2019-11-21:
Does this appear outside of fastRTPS? I suspect this is a RMW layer bug since this is happening on a costmap layer reset which is unsubscribing and resubscribing to a topic quickly, clearly the message did at one point exist. 

@JaimeMartin does this look like something lower in the stack?

yathartha3, 2019-11-21:
<details>
```yathartha@11900298-desk1:~$ ros2 launch nav2_bringup nav2_tb3_simulation_launch.py
[INFO] [launch]: All log files can be found below /home/yathartha/.ros/log/2019-11-21-13-44-32-568876-11900298-desk1-7592
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [7609]
[INFO] [gzclient-2]: process started with pid [7610]
[INFO] [rviz2-3]: process started with pid [7611]
[INFO] [robot_state_publisher-4]: process started with pid [7612]
[INFO] [lifecycle_manager-5]: process started with pid [7613]
[INFO] [map_server-6]: process started with pid [7614]
[INFO] [amcl-7]: process started with pid [7615]
[INFO] [controller_server-8]: process started with pid [7616]
[INFO] [planner_server-9]: process started with pid [7617]
[INFO] [recoveries_server-10]: process started with pid [7618]
[INFO] [bt_navigator-11]: process started with pid [7628]
[robot_state_publisher-4] Initialize urdf model from file: /home/yathartha/ros2_all_ws/turtlebot3_ws/install/turtlebot3_description/share/turtlebot3_description/urdf/turtlebot3_waffle.urdf
[robot_state_publisher-4] Parsing robot urdf xml string.
[robot_state_publisher-4] Link base_link had 7 children
[robot_state_publisher-4] Link camera_link had 2 children
[robot_state_publisher-4] Link camera_depth_frame had 1 children
[robot_state_publisher-4] Link camera_depth_optical_frame had 0 children
[robot_state_publisher-4] Link camera_rgb_frame had 1 children
[robot_state_publisher-4] Link camera_rgb_optical_frame had 0 children
[robot_state_publisher-4] Link caster_back_left_link had 0 children
[robot_state_publisher-4] Link caster_back_right_link had 0 children
[robot_state_publisher-4] Link imu_link had 0 children
[robot_state_publisher-4] Link base_scan had 0 children
[robot_state_publisher-4] Link wheel_left_link had 0 children
[robot_state_publisher-4] Link wheel_right_link had 0 children
[robot_state_publisher-4] got segment base_footprint
[robot_state_publisher-4] got segment base_link
[robot_state_publisher-4] got segment base_scan
[robot_state_publisher-4] got segment camera_depth_frame
[robot_state_publisher-4] got segment camera_depth_optical_frame
[robot_state_publisher-4] got segment camera_link
[robot_state_publisher-4] got segment camera_rgb_frame
[robot_state_publisher-4] got segment camera_rgb_optical_frame
[robot_state_publisher-4] got segment caster_back_left_link
[robot_state_publisher-4] got segment caster_back_right_link
[robot_state_publisher-4] got segment imu_link
[robot_state_publisher-4] got segment wheel_left_link
[robot_state_publisher-4] got segment wheel_right_link
[robot_state_publisher-4] Adding fixed segment from base_footprint to base_link
[robot_state_publisher-4] Adding fixed segment from base_link to camera_link
[robot_state_publisher-4] Adding fixed segment from camera_link to camera_depth_frame
[robot_state_publisher-4] Adding fixed segment from camera_depth_frame to camera_depth_optical_frame
[robot_state_publisher-4] Adding fixed segment from camera_link to camera_rgb_frame
[robot_state_publisher-4] Adding fixed segment from camera_rgb_frame to camera_rgb_optical_frame
[robot_state_publisher-4] Adding fixed segment from base_link to caster_back_left_link
[robot_state_publisher-4] Adding fixed segment from base_link to caster_back_right_link
[robot_state_publisher-4] Adding fixed segment from base_link to imu_link
[robot_state_publisher-4] Adding fixed segment from base_link to base_scan
[robot_state_publisher-4] Adding moving segment from base_link to wheel_left_link
[robot_state_publisher-4] Adding moving segment from base_link to wheel_right_link
[lifecycle_manager-5] [INFO] [lifecycle_manager]: Creating
[lifecycle_manager-5] [INFO] [lifecycle_manager]: Creating and initializing lifecycle service clients
[amcl-7] [INFO] [amcl]: Creating
[map_server-6] [INFO] [map_server]: Creating
[bt_navigator-11] [INFO] [bt_navigator]: Creating
[planner_server-9] [INFO] [planner_server]: Creating
[controller_server-8] [INFO] [controller_server]: Creating controller server
[planner_server-9] [INFO] [global_costmap.global_costmap]: Creating Costmap
[controller_server-8] [INFO] [local_costmap.local_costmap]: Creating Costmap
[rviz2-3] [INFO] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [rviz2]: OpenGl version: 4.4 (GLSL 4.4)
[rviz2-3] [INFO] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-3] [WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-3] [INFO] [rviz2]: Waiting for the lifecycle_manager/is_active service...
[rviz2-3] [INFO] [rviz2]: Sending lifecycle_manager/is_active request
[... 1201 lines cut ...]
[recoveries_server-10] [INFO] [recoveries_server]: Spin completed successfully
[recoveries_server-10] [INFO] [recoveries_server]: Attempting Wait
[... cut ...]
[ERROR] [controller_server-8]: process has died [pid 7616, exit code -11, cmd '/home/yathartha/ros2_all_ws/navigation2_ws/install/nav2_controller/lib/nav2_controller/controller_server --ros-args --params-file /tmp/tmpxm5qm3ye        '].
```

@crdelsey I still see this after deleting the voxel layer plugin name, and the voxel layer params.

This is triggered mostly if you do the following:
- Give a goal pose to navigate
- Drop an obstacle on its path right in front of the robot
- repeat 2-3 times

mlherd, 2019-11-21:
I can reproduce this issue if I drop an obstacle in Gazebo very close to the robot while the robot is navigating. 

I tested it without enabling the voxel layer.

SteveMacenski, 2019-12-10:
Filed ticket https://github.com/ros2/rmw_fastrtps/issues/341

SteveMacenski, 2019-12-10:
@mlherd I'm curious - has anyone on your end tested this with Cyclone or any other DDS vendor? It would be good to know if its vendor specific or generally problematic

mlherd, 2019-12-10:
I tested it only with FastRTPS.

crdelsey, 2019-12-11:
Not yet. It's on the todo list.

closed 2020-01-04 by SteveMacenski