ros-navigation/navigation2 #1366: Nav2 bringup does not work with namespaced tb3 hardware

lbegani, 2019-11-18:
## Bug report

**Required Info:**

- Operating System: Ubuntu 18.04
- Version or commit hash: 8c1a8e6e15a6b7ea52d32dc0a2e4abd70e4f9e12
- DDS implementation: Fast-RTPS

#### Steps to reproduce issue

```
1. Run tb3 bringup(with code changes) in turtlebot3 waflle hardware with namespace
    $ ros2 launch turtlebot3_bringup robot.launch.py namespace:=tb3_0
    $ ros2 topic list
    /parameter_events
    /rosout
    /tb3_0/battery_state
    /tb3_0/cmd_vel
    /tb3_0/imu
    /tb3_0/joint_states
    /tb3_0/magnetic_field
    /tb3_0/odom
    /tb3_0/parameter_events
    /tb3_0/robot_description
    /tb3_0/rosout
    /tb3_0/scan
    /tb3_0/sensor_state
    /tb3_0/tf
    /tb3_0/tf_static
2. Run nav2 stack with namespace, returns error
    $ ros2 launch nav2_bringup nav2_bringup_launch.py map:=$HOME/map.yaml namespace:=tb3_0
[WARN] [rcl]: Found remap rule 'namespace:=tb3_0'. This syntax is deprecated. Use '--ros-args --remap namespace:=tb3_0' instead.
[INFO] [launch]: All log files can be found below /home/nuc5-1804/.ros/log/2019-11-18-11-56-09-368035-nuc51804-NUC8i7BEH-2702
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [lifecycle_manager-1]: process started with pid [2714]
[INFO] [map_server-2]: process started with pid [2715]
[INFO] [amcl-3]: process started with pid [2716]
[INFO] [controller_server-4]: process started with pid [2717]
[INFO] [planner_server-5]: process started with pid [2718]
[INFO] [recoveries_server-6]: process started with pid [2719]
[INFO] [bt_navigator-7]: process started with pid [2720]
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Creating
[planner_server-5] [INFO] [planner_server]: Creating
[planner_server-5] [INFO] [global_costmap.global_costmap]: Creating Costmap
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Creating and initializing lifecycle service clients
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Starting managed nodes bringup...
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Configuring map_server
[amcl-3] [INFO] [amcl]: Creating
[controller_server-4] [INFO] [controller_server]: Creating controller
[controller_server-4] [INFO] [local_costmap.local_costmap]: Creating Costmap
[map_server-2] [INFO] [map_server]: Creating
[bt_navigator-7] [INFO] [bt_navigator]: Creating
[map_server-2] [INFO] [map_server]: Configuring
[map_server-2] [INFO] [map_server]: OccGridLoader: Creating
[map_server-2] [INFO] [map_server]: OccGridLoader: Configuring
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Configuring amcl
[amcl-3] [INFO] [amcl]: Configuring
[amcl-3] [INFO] [amcl]: initTransforms
[amcl-3] [INFO] [amcl]: initPubSub
[amcl-3] [INFO] [amcl]: Subscribed to map topic.
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Configuring controller_server
[controller_server-4] [INFO] [controller_server]: Configuring local controller interface
[controller_server-4] [INFO] [local_costmap.local_costmap]: Configuring
[controller_server-4] [INFO] [local_costmap.local_costmap]: Using plugin "static_layer"
[controller_server-4] [INFO] [local_costmap.local_costmap]: Subscribing to the map topic (/map) with transient local durability
[controller_server-4] [INFO] [local_costmap.local_costmap]: Initialized plugin "static_layer"
[controller_server-4] [INFO] [local_costmap.local_costmap]: Using plugin "obstacle_layer"
[controller_server-4] [INFO] [local_costmap.local_costmap]: Subscribed to Topics: 
[controller_server-4] [INFO] [local_costmap.local_costmap]: Initialized plugin "obstacle_layer"
[controller_server-4] [INFO] [local_costmap.local_costmap]: Using plugin "inflation_layer"
[... 57 lines cut ...]
[INFO] [bt_navigator-7]: process has finished cleanly [pid 2720]
[bt_navigator-7] [INFO] [rclcpp]: signal_handler(signal_value=2)
[bt_navigator-7] [INFO] [bt_navigator]: Destroying
[map_server-2] [INFO] [rclcpp]: signal_handler(signal_value=2)
[lifecycle_manager-1] [INFO] [rclcpp]: signal_handler(signal_value=2)
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Destroying
[amcl-3] [INFO] [rclcpp]: signal_handler(signal_value=2)
[amcl-3] [INFO] [amcl]: Destroying
[recoveries_server-6] [INFO] [rclcpp]: signal_handler(signal_value=2)

```

#### Expected behavior
Nav2 should run without errors
#### Actual behavior
Error in running nav2 with namespaced tb3 waffle hardware
#### Additional information
The experiment is run with TB3 waffle hardware. Changes were done in turtlebot3 ros2 packages to enable namespace in tb3 topics

----
## Feature request
Support nav2 with namespace as argument
#### Feature description

#### Implementation considerations

SteveMacenski, 2019-11-18:
Feel free to submit a PR.

if they do something to breaks basic compatibility with this stack, someone who has an invested interest in the TB3 should work to resolve.

crdelsey, 2019-11-21:
I think you just need to do a clean build. This library `libnav2_behavior_tree_nodes.so` no longer exists. The code shouldn't be referring to it.

lbegani, 2019-11-26:
@SteveMacenski @crdelsey I am able to fix this issue by creating a new launch file that includes  nav2_bringup_launch file and calls PushRosNamespace action. Have tested it on simulation, shall test on hardware and create a PR for the same.

shivaang12, 2019-12-04:
I am getting this error without providing the namespace. Is someone happen to know the problem?

crdelsey, 2019-12-09:
The only place the name `nav2_behavior_tree_nodes` exists is in the default value for the `plugin_lib_names` parameter on nav2_bt_navigator. So, if you get the error

```
Could not load library: libnav2_behavior_tree_nodes.so: cannot open shared object file: No such file or directory
```

it means that the `plugin_lib_names` parameter is not getting set properly through your launch file.

I will try to reproduce.

closed 2019-12-19 by SteveMacenski