ros-navigation/navigation2 #1367: Incorrect behavior when navigating to current pose

SteveMacenski, 2019-11-18:
When you navigate to the current pose, or within the success criteria of the controller of the current position, rather than returning success it returns failure.

```
[bt_navigator-10] [INFO] [bt_navigator]: Begin navigating from current location to (0.58, -0.55)
[controller_server-7] [INFO] [controller_server]: Received a goal, begin computing control effort.
[controller_server-7] [INFO] [controller_server]: Reached the goal!
[controller_server-7] [WARN] [controller_server_rclcpp_node]: [FollowPath] [ActionServer] A preemption request was available before succeeding on the current goal.
[controller_server-7] [INFO] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[controller_server-7] [INFO] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[bt_navigator-10] [WARN] [bt_navigator_client_node]: Node timed out while executing service call to local_costmap/clear_entirely_local_costmap.
[bt_navigator-10] [ERROR] [bt_navigator]: Navigation failed
```

It should return success. This looks like a bug in the simple action server.

crdelsey, 2019-11-18:
How often do you see this? How do you reproduce this?

mkhansen-intel, 2019-11-18:
Can you post more of the log? This doesn't make sense by itself. There's no global planner output here, did the global_planner output a path? How did we get to the recovery behavior at all?

SteveMacenski, 2019-11-18:
That's all the log there is, I didn't clip anything. 

I see this every time. Reproduce by just having it go to its current position in a navigate command

mlherd, 2019-11-19:
I am not so familiar with this part of the stack, but I guess somewhere in [simple_action_server](https://github.com/ros-planning/navigation2/blob/46b329571d3b9f9a60bf9da7158bcdfa64992f23/nav2_util/include/nav2_util/simple_action_server.hpp)  when there is a request for goal acceptance, the [isGoalReached()](https://github.com/ros-planning/navigation2/blob/ea0524fb6a75b9ab8f113556ff5d6e492fab6e0c/nav2_core/include/nav2_core/goal_checker.hpp) can be used to check if the new goal is within the success criteria and reject that goal with an info message.

SteveMacenski, 2019-11-19:
Careful on language, it shouldn’t reject, it should succeed.

crdelsey, 2019-11-19:
FYI. We discussed possible causes yesterday. It's possible that #1371 will also fix this. I'm gonna test soon.

crdelsey, 2019-11-19:
I don't see this problem if I run the behavior tree in #1371.

I believe the problem is, when you say nav to the current location, the "GoalReached" condition node in the behavior tree responds with true, preventing the ComputePathToPose node from ever running and producing a path. Then FollowPath is told to follow an empty path and fails.

SteveMacenski, 2019-11-19:
Did you confirm it before doing in 1371? Happy to close when that’s in then.

crdelsey, 2019-11-20:
> Did you confirm it before doing in 1371?

No. I'm not able to reproduce what you see exactly. I either get a Navigation success or a controller server crashes.

Controller server crashes for me because we end up sending a path with 0 poses, I believe.

SteveMacenski, 2019-11-20:
That's probably not good that DWB can't handle that condition

crdelsey, 2019-11-25:
@SteveMacenski #1371 is integrated. Can you confirm this is fixed?

SteveMacenski, 2019-11-25:
I will not be able to get to it before the new year

mlherd, 2019-12-03:
I just tested this. This is still an issue.

<details>
    <summary> Terminal Log </summary>

```
[bt_navigator-11] [INFO] [bt_navigator]: Begin navigating from current location to (-0.83, -0.58)
[controller_server-8] [INFO] [controller_server]: Received a goal, begin computing control effort.
[controller_server-8] [INFO] [controller_server]: Reached the goal!
[controller_server-8] [WARN] [controller_server_rclcpp_node]: [follow_path] [ActionServer] A preemption request was available before succeeding on the current goal.
[controller_server-8] [INFO] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[bt_navigator-11] [WARN] [bt_navigator_client_node]: Node timed out while executing service call to local_costmap/clear_entirely_local_costmap.
[controller_server-8] [INFO] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[bt_navigator-11] [WARN] [bt_navigator_client_node]: Node timed out while executing service call to local_costmap/clear_entirely_local_costmap.
[bt_navigator-11] [ERROR] [bt_navigator]: Navigation failed
```

</details>

SteveMacenski, 2019-12-03:
You sure its not just a special case of #1402 ?

mlherd, 2019-12-03:
I am not sure. It quite is possible. It is hard to tell why navigation fails since it returns Navigation Failure almost for any goal pose. However, I have seen it printing success a a few times (~1 in 10) after sending the robot's current pose as its goal pose.

SteveMacenski, 2020-02-11:
This is no longer seen

closed 2020-02-11 by SteveMacenski