ros-navigation/navigation2 #1369: Navigation2 Dependencies build fail

rafaelhuang-intel, 2019-11-19:
I followed instructions from the github (https://github.com/ros-planning/navigation2/blob/master/doc/BUILD.md) to build navigation2, ros2 source code is  build success. However I'm having trouble building navigation2.
## Bug report

**Required Info:**

- Operating System:
  - ubuntu 18.04
- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

```
intel@intel-desktop:~/ros2_nav_dependencies_ws$ colcon build --symlink-install
Starting >>> camera_calibration_parsers
Starting >>> cv_bridge
Starting >>> image_transport
Starting >>> gazebo_dev
Starting >>> gazebo_msgs                                                                                                                             
Starting >>> image_geometry
Starting >>> angles
Starting >>> behaviortree_cpp_v3
Finished <<< gazebo_dev [5.72s]                                                                                                                                                                                                                                                                           
Finished <<< angles [5.69s]                                                                                                                                                                                                                                                                               
--- stderr: camera_calibration_parsers                                                                                                                                                                                                                                                                    
/home/intel/ros2_nav_dependencies_ws/src/image_common/camera_calibration_parsers                                                                     /src/parse_yml.cpp:45:10: fatal error: rcpputils/filesystem_helper.hpp: No such                                                                      file or directory
 #include "rcpputils/filesystem_helper.hpp"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/camera_calibration_parsers.dir/src/parse_yml.cpp.o] Err                                                                     or 1
make[2]: *** Waiting for unfinished jobs....
/home/intel/ros2_nav_dependencies_ws/src/image_common/camera_calibration_parsers                                                                     /src/parse.cpp:43:10: fatal error: rcpputils/filesystem_helper.hpp: No such file                                                                      or directory
 #include "rcpputils/filesystem_helper.hpp"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/camera_calibration_parsers.dir/src/parse.cpp.o] Error 1
/home/intel/ros2_nav_dependencies_ws/src/image_common/camera_calibration_parsers                                                                     /src/parse_ini.cpp:52:10: fatal error: rcpputils/filesystem_helper.hpp: No such                                                                      file or directory
 #include "rcpputils/filesystem_helper.hpp"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/camera_calibration_parsers.dir/src/parse_ini.cpp.o] Err                                                                     or 1
make[1]: *** [CMakeFiles/camera_calibration_parsers.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
---
Failed   <<< camera_calibration_parsers [ Exited with code 2 ]
Aborted  <<< image_geometry                                                                                                                                                                                                                                                                               
Aborted  <<< cv_bridge                                                                                                                                                                                                                                                                                    
Aborted  <<< gazebo_msgs                                                                                                                             
Aborted  <<< image_transport                                                                                                                         
Aborted  <<< behaviortree_cpp_v3

Summary: 2 packages finished [40.2s]
  1 package failed: camera_calibration_parsers
  5 packages aborted: behaviortree_cpp_v3 cv_bridge gazebo_msgs image_geometry i                                                                     mage_transport
  1 package had stderr output: camera_calibration_parsers
  7 packages not processed

```

#### Expected behavior

#### Actual behavior

#### Additional information

I tried dashing or eloquent rosdistro, but always build fail.
```
rosdep install -y -r -q --from-paths src --ignore-src --rosdistro <ros2-distro-name>
```
if search missing file, I see this file already build in ROS2 core
```
/home/intel/ros2_dashing/install/rosbag2_storage/include/rosbag2_storage/filesystem_helper.hpp
/home/intel/ros2_dashing/install/pluginlib/include/pluginlib/impl/filesystem_helper.hpp
/home/intel/ros2_dashing/install/rcpputils/include/rcpputils/filesystem_helper.hpp
/home/intel/ros2_dashing/src/ros/pluginlib/pluginlib/include/pluginlib/impl/filesystem_helper.hpp
/home/intel/ros2_dashing/src/ros2/rosbag2/rosbag2_storage/include/rosbag2_storage/filesystem_helper.hpp
/home/intel/ros2_dashing/src/ros2/rcpputils/include/rcpputils/filesystem_helper.hpp
/home/intel/ros2_nav_dependencies_ws/src/image_common/camera_info_manager/src/filesystem_helper.hpp
```

## Feature request

#### Feature description

#### Implementation considerations

mlherd, 2019-11-19:
- try fresh build
  - make sure .bashrc is clean and ROS2 
  - get the lastest ROS2 packages (master branch)
  - build ROS2_master
  - source setup.bash in ROS2 workspace
  - run `rosdep install -y -r -q --from-paths src --ignore-src --rosdistro eloquent` in both navigation2_ws and nav2_dependencies_ws.
  - build nav2_dependencies_ws
  - source nav2_dependencies_ws/install/setup.bash
  - build navigation2_ws

rafaelhuang-intel, 2019-11-19:
@mlherd 
My fail is Navigation2 dependence, not navigation2, fail package is camera_calibration_parsers, this package include image_common.
 
I tried below step, but still fail,

> make sure .bashrc is clean
> get the lastest ROS 2 packages (master branch)
> build ROS 2
> source setup.bash in ROS 2 workspace
> run rosdep install -y -r -q --from-paths src --ignore-src --rosdistro eloquent 
> build nav2_dependencies

mlherd, 2019-11-19:
How did you do all that in 10 minutes? :) 

I just build the camera_calibration_parsers package to test it. I am have ROS2 master from 2 days ago.

[image]

rafaelhuang-intel, 2019-11-19:
I clean work space and tried again, still fail

intel@intel-desktop:~/ros2_nav_dependencies_ws$ source ~/ros2_dashing/install/local_setup.bash
intel@intel-desktop:~/ros2_nav_dependencies_ws$ colcon build --symlink-install --packages-select camera_calibration_parsers
Starting >>> camera_calibration_parsers
--- stderr: camera_calibration_parsers
/home/intel/ros2_nav_dependencies_ws/src/image_common/camera_calibration_parsers/src/parse_yml.cpp:45:10: fatal error: rcpputils/filesystem_helper.hpp: No such file or directory
 #include "rcpputils/filesystem_helper.hpp"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/camera_calibration_parsers.dir/src/parse_yml.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/intel/ros2_nav_dependencies_ws/src/image_common/camera_calibration_parsers/src/parse_ini.cpp:52:10: fatal error: rcpputils/filesystem_helper.hpp: No such file or directory
 #include "rcpputils/filesystem_helper.hpp"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/camera_calibration_parsers.dir/src/parse_ini.cpp.o] Error 1
/home/intel/ros2_nav_dependencies_ws/src/image_common/camera_calibration_parsers/src/parse.cpp:43:10: fatal error: rcpputils/filesystem_helper.hpp: No such file or directory
 #include "rcpputils/filesystem_helper.hpp"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/camera_calibration_parsers.dir/src/parse.cpp.o] Error 1
make[1]: *** [CMakeFiles/camera_calibration_parsers.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
---
Failed   <<< camera_calibration_parsers [ Exited with code 2 ]

Summary: 0 packages finished [7.08s]
  1 package failed: camera_calibration_parsers
  1 package had stderr output: camera_calibration_parsers
intel@intel-desktop:~/ros2_nav_dependencies_ws$

 I installed ROS2 master and Nav2/depend from 2 hours ago.

rafaelhuang-intel, 2019-11-19:
@mlherd 
what's the ROS2 repo you use it? dashing? eloquent?

mlherd, 2019-11-19:
I have ROS2 master from two days ago.

rafaelhuang-intel, 2019-11-19:
@mlherd 
got it, I re-build ROS2 core and try again.

mlherd, 2019-11-19:
ok let me know if it still doesn't build. I will try to build nav2_depend using ROS 2 Dashing tommorow.

SteveMacenski, 2019-11-19:
I’d recommend cleaning the ros2 workspace and rebuilding everything. I do this about monthly to keep up with the changes

rafaelhuang-intel, 2019-11-20:
@mlherd @SteveMacenski 
when I checkout ROS2 repo from dashing to master, clean, sync , re-build code again, this issue disappear.

closed 2019-11-20 by rafaelhuang-intel