ros-navigation/navigation2 #1373: Controller and Planner action server names CamelCase when should be snake_case

SteveMacenski, 2019-11-20:
The action servers for the planner and controller are named after the messages and not according to the rest of the servers and convention.

We have `NavigateToPose` when it should be `navigate_to_pose`. As well as compute path and follow path.

crdelsey, 2019-11-20:
Are you referring to the class name like here?
https://github.com/ros-planning/navigation2/blob/5fe111a9d65d4d64d2f1f328559b21def159655e/nav2_behavior_tree/include/nav2_behavior_tree/navigate_to_pose_action.hpp#L30

Or the name given to the node that gets used in the behavior tree?
https://github.com/ros-planning/navigation2/blob/5fe111a9d65d4d64d2f1f328559b21def159655e/nav2_bt_navigator/behavior_trees/navigate_w_replanning.xml#L11

I think the ROS2 style guide requires class names to be CamelCase, and methods snake_case.

For the actual node name in the behavior tree, CamelCase seems more consistent with what the behavior tree library uses.

SteveMacenski, 2019-11-20:
I'm referencing the actual action server name `/NavigateToPose` rather than `/navigate_to_pose`

Do a ros2 topic list and a ros2 service list, you'll see that the interfaces are supposed to be snake case

crdelsey, 2019-11-20:
Ah. I see. I've got no objection.

SteveMacenski, 2019-11-21:
@crdelsey do you prefer if I change the XMLs to be snake_case, or change the constructors to take in an action name and separately the XML tag to use?

SteveMacenski, 2019-11-21:
@shivaang12 if you're looking for another small task, this is another good candidate to get yourself familiar with the behavior tree stuff

crdelsey, 2019-11-23:
> do you prefer if I change the XMLs to be snake_case, or change the constructors to take in an action name and separately the XML tag to use? ( I assume the latter, but don't want to assume)

I think the latter as well. That way they can be namespaced if someone wants to have mutliple different actions of the same type.

SteveMacenski, 2019-11-23:
It is already implemented @crdelsey

SteveMacenski, 2019-11-25:
Its already done - why assign me?

closed 2019-11-27 by SteveMacenski