ros-navigation/navigation2 #1389: Default SLAM(s) For ROS2

SteveMacenski, 2019-11-25:
Per some discussions with the TSC, we're moving this discussion into Navigation. 

We need to come up with a default option that's well supported for ROS2. From the discourse thread and additional conversations at ROSCon, I propose the following as the 3 major options

- Cartographer 
- SLAM Toolbox
- LaMa

Each of these has their pros and cons. 

Cartographer: Lots of community traction, though very hard to setup or modify. Google has largely stopped working on it but some companies have really taken ahold of it. I'd argue that it isnt suitable for most (not all, not for everyone) industrial applications based on my experiences and discussions with a number of other companies. Works with 2D and 3D lidars, localization, and limited serialization. Lives in the Cartographer organization, but the temporary ROS2 fork in OSRF's organization, but not interested in maintaining it long term. To my knowledge, there are no excited maintainers for ROS2. 

SLAM Toolbox: Currently available under LGPL 2.1, it builds on top of the open karto library with additional optimizations and includes things like full serialization, localization, continue mapping, and works well for large / dynamic spaces. I would be a maintainer and welcome others who would be serious about it to join as well. Lives in my personal github account. 

LaMa: New SLAM based on distance function "attraction" as the author calls it. No ROS2 branch but we could make one without too much trouble. It runs much lower overhead than the other 2 options above, but I don't believe at this moment that anyone has tested how it would scale or do in large scale environments. There's a question around how well it works in changing environments or with the probabilistic grid generation. These problems don't seem unsolvable but are current things needing to be looked at. Lives in the IRIS LAMA github organization, the author would be willing to maintain and I would also probably take part. On a personal note, I've found it works well for some datasets I have but breaks down for others and I'm trying to figure out how or why. 

My proposition: I think we provide/support 2 SLAM implementations. I think those 2 should be SLAM Toolbox & LaMa. 

CC @mkhansen-intel  @mikeferguson @ablasdel

artivis, 2019-11-25:
I essentially agree with @SteveMacenski and, in the given context, vote for `SLAM Toolbox` because it is karto ('it just works') with goodies.

My only remark is that I wouldn't 'officially' provide/support 2 SLAM implementations for the reason that it pledges ROS 2 (maintainers, community etc) in to a lot of efforts. We better support one, do it well, and advertise other packages has 'community-provided' options. My concern is to avoid the ROS 1 situation where both karto and gmapping were somewhat the 'official' SLAM solutions but neither was fully working nor (at some point) properly maintained.
I also think that it might be confusing for new comers following tutorials and the main recommendations. They are not looking for options to tweak and compare but rather something 'that just maps'.

To summarize, my remark is about duplicate efforts and (mis)communication.

SteveMacenski, 2019-11-25:
A followup question since I wasn’t involved circa 2011 when these are made: 

Is the reason they weren’t “fully supported” because of duplicate efforts or just complete lack of any efforts? From my experience in the last 6 years, neither has had any support and I don’t believe that would have changed any even if there were only 1. Moreover, little maintenance is a fair claim, but both have very small ticket queues and haven’t required much updates (though total lack of additional development, maintainers own companies built solutions they use instead, and didn’t give back to larger community :frown:). 

Karto is old. I gave it (in my opinion) a really reasonable face lift internally and a bunch of tools on top of it. There’s objectively still work to do, but the biggest bang-for-the-buck tasks are done. Additional library development of open karto is time intensive and not massively impactful. I suspect the same will happen here, except maybe new shiny tools on top. 

With that said, I don’t want to stifle innovation and just go with what “just works”. I think LaMa represents the direction alot of the academic community that still cares about 2D/3D laser SLAM are going. Distance function approaches have taken over this space and as their own metrics show and I can confirm, are a solid over of magnitude decrease in compute time for basic use cases (as far as I know, no ones tested at scale). 

I see your point- and maybe that’s the way we should go. Lets get some more thoughts before we conclude. I want to make sure its not a situation where we’re using slam toolbox because its what we’re comfortable with as a community.

mkhansen-intel, 2019-12-05:
Let's discuss again today in the WG.

SteveMacenski, 2019-12-05:
@mkhansen-intel I have a speaking engagement during this slot, I can't come. Shouldn't stop you all if you would like to continue to discuss.

adamdbrw, 2020-01-15:
Was there any follow up to this?

SteveMacenski, 2020-01-15:
No updates at the moment. We have another working group meeting tomorrow, @mkhansen-intel can we add to the agenda?

SteveMacenski, 2020-01-16:
We talked again this week, the default was chosen to be `slam-toolbox`. 

We need to find now a set of maintainers other than me to be involved and have some more formality with its maintenance, release, and testing. 

Closing ticket, opening new ticket in SLAM Toolbox for this work

closed 2020-01-16 by SteveMacenski

JaccovdS, 2020-01-17:
For the public track record and for my information can you explain shortly why `slam-toolbox` was chosen over the other options you mentioned before?

SteveMacenski, 2020-01-17:
I'm going to pass the baton to @mkhansen-intel to explain. 

Frankly, I think any explanation coming from me is going to look full of impropriety in posterity. I will put a couple bullets but I'd like Matt to give his opinion on it independently.

- The goal is to have a sold out of box experience - and given this has roots in Karto and this version has received very good feedback from all users.
- We need to have a dedicated and knowledgeable maintainership staff for the long-haul, specifically in ROS2. 

We had 3 options identified: cartographer, slam toolbox, and lama slam. Cartographer met neither of these requirements. Lama, while very promising, is understood and tested in very limited environments, without a ROS2 wrapper. The maintainer expressed interest in additional work and maintainership responsibilities, but there's still a great deal of unknowns involved in the generalization of the approach for many environments and robots.

I'll make the note that I really think LAMA, whether in its current form or a later derivative is very promising. I personally would like to see that nurtured in the community and tested at scale. If it truly performs, its a good option to use, but I feel at this point it provides a fairly unknown out of box experience for a random user and that's one of the core things we are looking for.

eupedrosa, 2020-01-17:
LaMa SLAM author here.
I completely agree that my solution lacks testing and it is in fact an unknown out of box experience. Selecting  `slam-toolbox` was a good decision.

I will continue maintaining (and developing) LaMa - ROS2 included. Having choices is always a good thing.

doisyg, 2020-01-17:
@mkhansen-intel and @SteveMacenski , is there any qualitative tests comparing the performance of the `slam-toolbox` and `cartographer`? I am thinking for instance about tests conducted on a set of bags with different variables: size of the of the area scanned, quality of the odometry, power of the lidar (range, scan density), with/without IMU, etc... Ideally with some of the bags being challenging. 
This could be done collaboratively where people familiar with each SLAM tool will optimize the parameters for each bag and there favorite tool. In addition to provide a easy way to compare the performance of each SLAM tool, this will provides examples on how to use and tune them. I am more than willing to contribute to the `cartographer` part.
As discussed recently IRL @SteveMacenski , I am quite comfortable and happy with the performance of `cartographer` and I have yet to try `slam-toolbox`. If I have an example showing me that `slam-toolbox` performs at least as good as `cartographer`, I will happily invest the time needed to make the transition and contribute to the development (speaking for wyca robotics dev team).

SteveMacenski, 2020-01-17:
> I am thinking for instance about tests conducted on a set of bags with...

I don't think anyone's going to argue that comparing is good. But really, Cartographer is not a real option until at least one of the following occurs:
- Google stands back up and commits to maintaining cartographer. The last commit was in [May](https://github.com/googlecartographer/cartographer/commits/master).
- Someone that deeply and appreciably understands the codebase of cartographer and the ROS2 wrappers commit to maintaining and developing on it for the long-jaul
- Cartographer provides sufficient documentation and logical defaults to create a positive out of box experience.