ros-navigation/navigation2 #1390: Pull GoalChecker plugin out of DWB

crdelsey, 2019-11-25:
## Bug report

## Feature request

#### Feature description
There is currently a "isGoalReached" method in the controller plugin interface. DWB implements this method by loading a "GoalChecker" plugin and passing the parameters through to the plugin with virtually no change.

There is nothing about the "GoalChecker" plugins that is DWB specific, so it makes sense to pull them out and load the GoalChecker independently, so that the plugins can be reused with no changes needed by the controller plugin implementation.

#### Implementation considerations

1. We can load the plugin in the controller server with minimal changes.
2. We can load the plugin in the behavior tree. It already has something similar.

Between the two options, 1. is relatively simple. 2. is more in line with long term goals, but might have reliability implications.

SteveMacenski, 2019-11-25:
Its also possible we do both, where there is a controller level process and goal checker for local control level and correspondingly from a task level perceptive.

crdelsey, 2019-11-25:
- If we put it in the BT navigator, I'm thinking it would be like the existing "GoalReached" condition node ... in other words, it would just do the transforms it self to get current pose, and also do the math itself to determine if we've reached the goal. No need for a service call.

- I was thinking bad things would happen if BT navigator failed to cancel the action when the goal was reached (eg. network issues), but I guess the worst case is that the robot just keeps twitching around the goal pose if it never gets told to stop.

SteveMacenski, 2019-11-25:
> network issues

That really should never happen. I know back a year ago when things were less... usable that was a concern, but it shouldn't be anymore. I don't know how we can accomplish anything if we assume there's going to be lost service or action calls. 

Is GoalReached doing a service call? I've never looked at it. Yeah it should just have some rotation and position tolerance. I think adding those plugins to the controller server sounds reasonable but also trying to think about what they actually represent: progress checker is making sure the robot isn't stuck or isn't moving. Is that really a controller problem? Maybe. I'd argue its a  system level problem so I could see that plugin existing either in the behavior tree, or externally to the behavior tree in the BT navigator in a while loop. The goal reached condition I think I could argue for logistics reasons could be placed in the controller server, but also I could see it being valuable to have different goal reached conditionals for different behavior tree-able actions. I suppose that could be alternatively accomplished with a map of plugins like we have elsewhere but a goal condition reached node would be sufficiently light weight I think building into the BT (or BT navigator) would be fine

SteveMacenski, 2019-11-27:
Another note, this work shouldn't just be the goal checker, it should be the progress checker too. These are both generic things that should be pulled out at the same time and exposed.

siddhya, 2020-05-24:
So create a plugin folder under nav2_controller move add simple/stopped goal checker to it (and progress checker as well). Use this plugin in nav2_controller as well as nav2_behavior_tree where isGoalReached is needed?
In that case nav2_behavior_tree will have dependency on nav2_controller plugin, is that ok?
What if derived controllers like nav2_dwb_controller want to change the GoalChecker plugin?

SteveMacenski, 2020-05-26:
I think what it would be is:
- Create and move an abstract interfaces for the progress tracker to the `nav2_core` package: https://github.com/ros-planning/navigation2/tree/master/nav2_core/include/nav2_core
- Make the existing `SimpleProgressChecker` progress checker derive from this base class 
- In the nav2_controller, load it and use it
- Move the DWB goal checkers to nav2_controller package
- Remove goal checker from the DWB code, add/load in the nav2_controller and check. I think that would involve removing the `isGoalReached` from the controller abstract interface and having the `isGoalReached` inside of `nav2_controller` server call this new local plugin rather than the controller. 

In terms of where these should live; headers in `include/` and plugins in `plugins/` would work to be analog to `costmap_2d`. I'm not sure what you mean by the `nav2_behavior_tree`, currently the goal checking plugin in the BT is decoupled from the `nav2_controller`'s. Unless you're proposing having them use the same plugin or there's something else I'm missing. If we decided to make this plugin be used in both, then we should probably make a new package for these plugins, though I'm not sure what a reasonable name is off hand. 

If users want to change the GoalChecker, then they just need to select a new one to be loaded by plugin lib. If you look at the nav2_planner / nav2_controller, you'll see examples of how to load runtime plugins from a configuration file.

closed 2020-07-11 by SteveMacenski