ros-navigation/navigation2 #1396: Inconsistency in tutorials\documentation

srsidd, 2019-11-27:
Sort of related to #980. While getting started on the nav stack, I used instructions from [doc/build.md](https://github.com/ros-planning/navigation2/blob/master/doc/BUILD.md). Once completed, the tutorial redirects me to [nav2_bringup/readme.md](https://github.com/ros-planning/navigation2/blob/master/nav2_bringup/bringup/README.md). However that documentation is not as thorough\complete as what I found in [doc/tutorials/](https://github.com/ros-planning/navigation2/blob/master/doc/tutorials/navigation2_with_turtlebot3_in_gazebo.md).

I intend to create a PR rectifying this, but need on clarity on a few things - 
- Are we going to have individual README files within each meta package? If so do we want to update the README in [nav2_bringup/readme.md](https://github.com/ros-planning/navigation2/blob/master/nav2_bringup/bringup/README.md)?
- Is the  [nav2_bringup/readme.md](https://github.com/ros-planning/navigation2/blob/master/nav2_bringup/bringup/README.md) intentionally kept at a bare minimum (i.e., In the long run these will be the instructions, and what we have in the tutorial now is more of temporary measure?)
- Worst case, I think we should edit the hyperlink in the [doc/build.md](https://github.com/ros-planning/navigation2/blob/master/doc/BUILD.md) to point to [doc/tutorials/](https://github.com/ros-planning/navigation2/blob/master/doc/tutorials/navigation2_with_turtlebot3_in_gazebo.md), thoughts?

mlherd, 2019-11-27:
PR #1393 is going to update and move the build instructions to Github Pages. There is a link to the doc/tutorials in it but I am moving the tutorials to the website as well.

Starting from Eloquent release, the plan is to start using Github Pages instead, so there may be some readme files in the repo but probably no instructions and tutorials. Github pages is under construction [Navigation2 Documentation](https://ros-planning.github.io/navigation2/)

In the long run, I think [Getting started instruntions](https://ros-planning.github.io/navigation2/getting_started/index.html) will replace nav2_bringup/readme.md

Feel free to contribute to Github pages :)
https://ros-planning.github.io/navigation2/howtos/docs/index.html

srsidd, 2019-12-02:
Ah, I missed that PR. I'll wait for that to get merged, and then clean it up (if required). I'd rather wait and make sure I'm not duplicating effort\complicating things more.

SteveMacenski, 2019-12-02:
@srsidd if you have some concerns, please review @mlherd PR so that your issues can be reflected soon. 

We're pushing the Eloquent release this week so this is your last call to get changes in for the distribution

crdelsey, 2019-12-02:
> Are we going to have individual README files within each meta package? If so do we want to update the README in nav2_bringup/readme.md?

I'd say the intent is to move user documentation to the Github Pages docs. It looks so much nicer there and is easier to navigate. I think each package should still have a readme, but it is focused more on documentation for developers.

> Is the nav2_bringup/readme.md intentionally kept at a bare minimum (i.e., In the long run these will be the instructions, and what we have in the tutorial now is more of temporary measure?)

I'd say it's more a case of neglect due to focusing on other issues. I believe we want separate simple build instructions with add-ons or some such thing for more complicated variants (like building master). I think it's a bit easier to do this in the Github Pages (sphinx_doc directory). The tutorial is a more in-depth walkthrough for those who want it.

SteveMacenski, 2020-02-15:
Hi @srsidd,

I'd love more feedback on the tutorials and documentation, but it seems that the specific issues this ticket covers are no longer issues, primarily because those files no longer exist. We've transitioned our documentation to the website (https://ros-planning.github.io/navigation2/). If you have any feedback there or suggestions of how to improve, we certainly welcome that.

closed 2020-02-15 by SteveMacenski

srsidd, 2020-02-24:
Hi @SteveMacenski, thanks for the heads up. I'm excited abt the sphinx documentation which IMO is big stepup. I tried the new instructions recently and have the following comments - 

- I think it's worth specifying a `rosdep install` step in the src build instructions (I had a dependency issue that was resolved by running the rosdep cmd which is currently missing)
- I had trouble with my ros1\ros2 workspaces conflicting. So I think it'll help if we specify to users _not_ to source the ROS1 workspace throughout the entire build\install process (That's what ended up solving the problem for me)
- There's a small gap in the documentation when going from installing `navigation2` from src to `getting started` in that the step to install `turtlebot3` packages (from the src install) is missing. 

While the first two are minor comments and perhaps don't need a separate ticket the third point may need some discussion as to what the right path is and how to proceed with that. I can make these modifications and create a PR in the near future.

SteveMacenski, 2020-02-24:
rosdep is indeed in the source build instructions

```
rosdep install -y -r -q --from-paths src --ignore-src --rosdistro eloquent
```

> So I think it'll help if we specify to users not to source the ROS1 workspace throughout the entire build\install process

I'm not sure that that needs to be explicitly called out. We never ask for users to source or do anything with ROS1 in the instructions so there shouldn't be any reason a user would think they should source any ROS1 paths.

>  (from the src install) is missing.

Its not missing - its not included. This is how to install navigation2, which doesn't depend on turtlebot. If you would like to use turtlebot as your platform, that's a specific choice for your application, but the getting started and tutorial pages cover its install

srsidd, 2020-02-24:
Perhaps I was looking at the wrong page, but I used instructions from [here](https://ros-planning.github.io/navigation2/build_instructions/index.html#manually-build-master) to manually build from master and that doesn't have a rosdep install step. 

That's fair, my assumption is that (like myself) most people already would have a shortcut in their `.bashrc` to source their ros1 workspace, and do not explicitly attempt to remove that while performing the installation, which leads to the dependency errors while installing and having a cautionary statement may help with that. (I remember you addressing a question on ros answers fairly recently which I think originated due to this)

That makes sense. Perhaps I jumped the gun a little bit, once I've installation the navigation packages, I move on to the the tutorials, which point to using a turtlebot3, and they assume only installing from binaries, is it worth adding instructions (Or even a link) for a src install?