ros-navigation/navigation2 #1397: Critical Error: Robot performs unexpected recovery behaviors when navigating in the kitchen

mlherd, 2019-11-28:
[image]

## Bug report

**Required Info:**

- Operating System:
 Ubuntu 18.04

- Version or commit hash:
  ROS2 Eloquent
  Navigation2 Master

- DDS implementation:
  Fast-RTPS

#### Steps to reproduce the issue
Send a turkey robot to the kitchen

#### Expected behavior
Navigate to the kitchen and stay there for the Thanksgiving dinner preparation.

#### Actual behavior
It seems like the robot doesn't want to naviate to the kitchen and try to run away from the area for its life

#### Additional information

This is not an actual issue. Happy Thanksgiving to all Navigation2 Stack contributors.

SteveMacenski, 2019-11-28:
I think its a namespacing issue, your `turkeybot` should be `apathetic_cat` and I think you should be good. 

If not, try with cyclonedds and tag @joespeed in the greetings.

mlherd, 2019-11-28:
[image]

It is a turkey on a TB3 Burger robot.

If anyone wants to do tests with the model -> [turkey_burger.zip](https://github.com/ros-planning/navigation2/files/3899584/turkey_burger.zip)

mlherd, 2019-12-03:
Closing this issue since I can't reproduce it anymore. However, please feel free to reopen it, if you still have this issue :)

closed 2019-12-03 by mlherd