ros-navigation/navigation2 #1399: Circle CI tests sometimes use wrong cache key to restore overlay_ws

crdelsey, 2019-12-02:
## Bug report

Frequently Circle CI reports a test failure on the release or debug build while the other passes. Here is an example:
https://circleci.com/workflow-run/52ce72ee-bb06-4184-9924-96b52b2ded4b

Notice in this example that the `release_build` passed but the `release_test` failed. The test failure seems to be a result of the `restore_cache_overlay_ws` step which reports

```
No cache is found for key: overlay_ws-v1-arch1-linux-amd64-6_85-master-<no value>-u6AwsDLhugmiZ0B_Wl3o2sBFBcgYbeHki0IEMAAizBI=
```

This is not the same key used to store the overlay_ws used in the build. That step reports

```
Creating cache archive...
Uploading cache archive...
Stored Cache to overlay_ws-v1-arch1-linux-amd64-6_85-master-<no value>-qOTLX06YeIrLRuIbKs0P7eIKhiK3seugKoNnS1GUriA=-1575305752
  * /opt/overlay_ws
```

@ruffsl Am I correct in assuming the cache keys should match?

ruffsl, 2019-12-02:
The source of any cache break can be debugged by diffing the `checksum.txt` artifacts between jobs. Comparing the overlay `checksum.txt` files:

* https://6486-135363400-gh.circle-artifacts.com/0/opt/overlay_ws/checksum.txt
* https://6489-135363400-gh.circle-artifacts.com/0/opt/overlay_ws/checksum.txt

We can see the following first delta is the cause for the break in cache key change:

``` diff
  angles:
    type: git
    url: https://github.com/ros/angles.git
-    version: 60cdaa34b81b2dba7938546f011db1ac09238339
+    version: 1.12.1
```

@crdelsey , any clue as to why vcs_export returns a different version between the time the build jobs starts and finishes before the test jobs starts? Are the versions being pinned by release tag or by commit sha, or just branch name? Perhaps the branch was pushed to between jobs? I can look into CircleCI to see if they have pipeline passing feature to avoid the second checkout.

crdelsey, 2019-12-02:
> Are the versions being pinned by release tag or by commit sha, or just branch name?

Depends on the package. Most just use a branch name. The only exception is the behavior tree library which uses a release tag.

crdelsey, 2019-12-02:
Those two versions are the same; 1.12.1 is a release tag on 60cdaa

crdelsey, 2019-12-02:
It seems like a bug in `vcs`. The docs say if `--exact` is specified it should output the SHA hash. Instead it's outputting tags.

It looks like this block of code looks if a tag exists and reports that instead.

https://github.com/dirk-thomas/vcstool/blob/ca86a2699c8b23b3cfadf565e54d3f1bc44528dc/vcstool/clients/git.py#L149-L175

It must be occasionally failing. It depends on the network, I believe (`git ls-remote`), so if there is a glitch at that moment, it can't get the remote tag and will report the hash instead.

crdelsey, 2019-12-02:
I filed issue https://github.com/dirk-thomas/vcstool/issues/109 to report this. I think it should be an easy fix if Dirk agrees to this behavior.

crdelsey, 2020-01-13:
This will be fixed when `python3-vcstool` version 0.2.6 binaries are released.

SteveMacenski, 2020-01-13:
It appears to be out: https://pypi.org/project/vcstool/

crdelsey, 2020-01-13:
Hmm. The [official ROS 2 build instructions](https://index.ros.org/doc/ros2/Installation/Eloquent/Linux-Development-Setup/#install-development-tools-and-ros-tools) have us install vsctool using apt. That was still 0.2.5 when I grabbed it this morning.

I'm assuming that's what our build uses, but it's possible we grab vcs using pip.

SteveMacenski, 2020-01-13:
I never know with python package management... why support both apt and pip... that just makes this so much more difficult

crdelsey, 2020-01-13:
Yeah. I keep getting messed up because some pip package installs the pip version of colcon, which conflicts with my apt installed colcon, and then my build breaks some months later when the two versions get out of sync. Ugh!

ruffsl, 2020-01-13:
> and then my build breaks some months later when the two versions get out of sync.

That happened to me just after returning from break; a pip install slipped into my system:

https://answers.ros.org/question/339552/failed-to-build-with-colcon-cannot-import-name-has_ros_schema_reference/#339671

I'd suggest we ping the vcstool maintainers to release upstream in the apt repo, keeping it all in sync.

ruffsl, 2020-02-07:
I think this issue with vcs is resolved, as the output of vcs export seems to be the exact branch. 

* build https://8434-135363400-gh.circle-artifacts.com/0/opt/underlay_ws/checksum.txt
* test https://8439-135363400-gh.circle-artifacts.com/0/opt/underlay_ws/checksum.txt

Though, it seems that when the docker image is pulled (during `Spin Up Environment`) it sometimes isn't the exact same between build and test jobs, thus the test job fails to `Restore Cache underlay_ws`:

``` diff
ros_entrypoint
- 2020-02-07 09:22:56.000000000 +0000
- 61550284d74be21ef0ff1d7f39c5e92b2473176f8ace93e3ae65f159bf1a1d71  /opt/ros_ws/checksum.txt
+ 2020-02-07 21:04:11.000000000 +0000
+ 3d3267e255a3a8b51e6aa7517fec129e5395e532e5be69efdad9e52254c9f971  /opt/ros_ws/checksum.txt

```

Perhaps this flaking is from some internal aws caching? I'll look into how we could just pass the id sha from the parent jobs onto child jobs, so child jobs pull the exact image. Perhaps my feature request for pipeline variable has be implemented yet, e.g:

``` yaml
executors:
  debug_exec:
    docker:
      - image: rosplanning/navigation2@sha256:5b95f992dd45dafc25054892ef3b5cf2ee08bb375c4855071724d67f8bcf4da9
```

closed 2020-02-07 by ruffsl

SteveMacenski, 2020-02-07:
I'll take your word for it.

I'm working through the testing / CI issues and I'm seeing right now a pretty strange behavior where all the packages are failing with uncrustify (https://circleci.com/gh/ros-planning/navigation2/8454?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-build-link) but tests are passing otherwise. Its seemingly every single package but only in Circle, not locally.

ruffsl, 2020-02-07:
> Its seemingly every single package but only in Circle, not locally.

Can you reproduce locally via our development Dockerfile? Your local host setup may not have all the latest curstify rules from upstream. E.g. https://github.com/ros-planning/navigation2/issues/1465#issuecomment-574899182 , but with:

```
docker build --pull --no-cache -t nav2:latest .
```

SteveMacenski, 2020-02-07:
You could be correct: https://github.com/ament/ament_lint/blame/master/ament_uncrustify/ament_uncrustify/configuration/ament_code_style.cfg

That is a pain, ok, there goes my afternoon :wink: 

I _will_ get CI passing before monday...