ros-navigation/navigation2 #1404: Robot still bumping into obstacle during spin recovery

bpwilcox, 2019-12-04:
## Bug report

**Required Info:**
- Operating System: 
  - Ubuntu 18.04
- Version or commit hash: 
  - master
- DDS implementation: Fast-RTPS
  - Fast-RTPS

#### Steps to reproduce issue
- add an obstacle near robot in gazebo simulation (close enough such that a spin movement would hit the obstacle)
- trigger spin recovery

#### Expected behavior
- robot will stop executing spin before it hits obstacle
#### Actual behavior
- robot hits obstacle before collision checker stops spin
#### Additional information
It seems that when I am placing the obstacle very close to the robot (but not touching or colliding) that the costmap displays the footprint intersecting the obstacle laser scans while the cells themselves are not considered occluded. This might be some issue with the laser scans and obstacle layer.

SteveMacenski, 2019-12-04:
unless they were in the minimum distance so they weren't considered. Not all rays are valid. https://github.com/ROBOTIS-GIT/turtlebot3_simulations/blob/master/turtlebot3_gazebo/models/turtlebot3_burger/model.sdf#L109

gimait, 2020-03-17:
@SteveMacenski, I found an issue with the CollisionChecker that could be the source of this problem, could you take a look at my commit [d1b2754](https://github.com/gimait/navigation2/commit/d1b27548e61cb1fca9daf6f7d30e9ecf0a418988) and tell me what you think?

SteveMacenski, 2020-03-17:
That looks like it could be right, though I'd do some printouts to be sure. Sometimes you can implicitly cast `unsigned char *` to a `char *` which would manifest negative values. I'd make sure there's not some clever optimization going on there. 

You could very well be right, that's what it is. Devise some test you can run and see if that does the trick!

gimait, 2020-03-17:
I think there is definetely an issue, since the result of scorePose will always be >=0 (this function returns the maximum cost in the robot footprint, which is given as a unsigned char* and then converted to double). I run some tests using the navigation bringup and I can say that now the spin stops when there is an obstacle. 

What I am not certain of is how far does this obstacle have to be for it to stop. I was thinking of writing a simple plugin able to trigger the spin recovery from a service call so I can test this properly, but this might be already done. Is there a way I can trigger the recovery manually?

SteveMacenski, 2020-03-17:
> I run some tests using the navigation bringup and I can say that now the spin stops when there is an obstacle.

To be clear, as opposed to the current way, which you tested to not stop when in collision but this did?

> What I am not certain of is how far does this obstacle have to be for it to stop.

It should be if the value of the costmap is 254 (occupied) or 253 (inscribed radius). 

>  I was thinking of writing a simple plugin able to trigger the spin recovery from a service call so I can test this properly, but this might be already done. Is there a way I can trigger the recovery manually?

What I've done in the past is just modify the behavior tree to only have 1 recovery: spin. Turn off the sensor data so you just have the static map, and then localize the robot in some position you're interested in and trigger navigation. If you disabled planner / controller and the tree only contains spin, it'll execute.

gimait, 2020-03-18:
> To be clear, as opposed to the current way, which you tested to not stop when in collision but this did?

Yes, that’s correct.

> It should be if the value of the costmap is 254 (occupied) or 253 (inscribed radius).

I’ll update the condition to include the inscribed radius also then.

> What I've done in the past is just modify the behavior tree to only have 1 recovery: spin. Turn off the sensor data so you just have the static map, and then localize the robot in some position [...]

I’ll try to test it this way then, let’s make sure that this is the actual issue.

SteveMacenski, 2020-03-18:
Awesome! If it all works out, please do submit a PR

closed 2020-03-19 by SteveMacenski