ros-navigation/navigation2 #1405: Use collision checker in Dwb

bpwilcox, 2019-12-04:
Currently, the collision checker code is refactored from the obstacle footprint critic in dwb in order to allow for collision checking on remote nodes via the costmap and footprint topic. It would be good remove the duplicate code and have dwb use the collision checker itself.

Since the collision checker currently relies on the costmap subscriber and footprint subscriber, there could be an overloaded implementation that can directly take a shared pointer to the `Costmap2DROS` object and use its interface to get the needed `costmap2D` object and robot footprint. This way DWB would not be limited by the topic publish frequency if it has its own `Costmap2DROS` object.

shivaang12, 2020-01-13:
Any one doing this?

crdelsey, 2020-01-13:
I don't think so. If you're interested, go ahead. 

I think this is a relatively big change. After taking a look at this, and figuring out how you want to approach this, I recommend you quickly describe how you want to approach the problem here to see if there is any feedback before investing a lot of time.

shivaang12, 2020-01-14:
Cool! Before I move ahead, I need to clarify if I am understanding the problem in a right sense.

I can see there two problems here,

1. Collision checker code is refactored from obstacle footprint critic and now we want to run only collision checker code and want obstacle footprint to use the result from collision checker.

2. Currently, Collision checker itself is subscribed to costmap subscriber and footprint subscriber. Now we wanted it to work without subscription by internally passing `Costmap2DROS ` pointer.

SteveMacenski, 2020-01-14:
That would be correct.

Its a bit involved. Personally, I would air on the side of "meh, whatever" and close the ticket. Trying to factor out everything sounds good on paper but this is really a different usecase (subscriber vs having access to the costmaps directly) and would make both implementations messy, or result in unnecessary data transformations on big-ish costmaps into a `msg` datatype.

shivaang12, 2020-01-14:
I agree! You can close it if you want.

SteveMacenski, 2020-01-14:
Well, not really mine to close ;-)

bpwilcox, 2020-01-14:
If others would like to close this ticket, I am okay with it (maybe give an upvote if you agree).

closed 2020-02-15 by SteveMacenski