ros-navigation/navigation2 #1409: Create shared_from_this for nav2_util::LifecycleNode

bpwilcox, 2019-12-04:
Throughout the stack shared pointers of the lifecycle nodes are being passed through functions and objects, somewhat recklessly and inconsistent. There doesn't appear to be any consistent behavior passing a `rclcpp_lifecycle::LifecylcleNode::SharedPtr` vs the `nav2_util::LifecycleNode::SharedPtr`, however in either case, only a shared ptr to the base class (`rclcpp_lifecycle::LifecylcleNode::SharedPtr`) will be passed because the base class inherits from `std::enabled_shared_from_this`. 

In my view, this is problematic, as we are misleadingly passing pointers to the base class as opposed to our stack's derived version. If we are to continue to use our derived version in the stack (perhaps that is its own discussion), then I think it is more appropriate that we pass a shared pointer to the `nav2_util::LifecycleNode` when requested.

My proposal is to add the following on our derived class:
``` c++
  std::shared_ptr<nav2_util::LifecycleNode> shared_from_this()
  {
      return std::static_pointer_cast<nav2_util::LifecycleNode>(rclcpp_lifecycle::LifecycleNode::shared_from_this());
  }
```
Additionally, we can decide the modules in the stack that should accept the base class shared ptr vs the derived class (if for example we wanted a class to not depend on the nav2_util version of the lifecycle node).

One use-case I am particularly interested in this change, as an example, would be to add a `nav2_util::ParameterEventsSubscriber` object directly on the node and access its interface. This would make handling parameters that are spread across various classes in the stack much simpler, since each class no longer needs to keep track of its own parameter subscriber (or pass a reference/pointer to it) but can use the one off the node. I'm sure there are other conveniences we could introduce as well.

SteveMacenski, 2019-12-04:
I figured the `nav2_utils` wrapper for lifecycle was temporary (https://github.com/ros-planning/navigation2/blob/master/nav2_util/include/nav2_util/lifecycle_node.hpp) it looks like a pretty thin shim, other than the thread being spun up to spin (which could generalize to any node type?).

Is there benefit rather than overloading it, just removing it now that lifecycle nodes are a little more API stable?

But I think if you believe keeping the wrapper is useful, your proposal makes sense. To your option of vase class vs derived class, it should be base class. Lets try not to spread this too far down into the stack.

bpwilcox, 2019-12-04:
@SteveMacenski I'm not sure if we are in a stage yet to remove the thread yet. @mjeronimo @orduno might know better?

My position is that the wrapper would be useful, at least in the interim, as we could move convenient API in our stack to the wrapper that might receive resistance upstream. As I mentioned in my example, I think it's better code design and less intrusive to the code. 

> But I think if you believe keeping the wrapper is useful, your proposal makes sense. To your option of vase class vs derived class, it should be base class. Lets try not to spread this too far down into the stack.
>
If we pass the base class shared ptr, then we don't have access to any wrapper interfaces, so I think we could be selective about where we want to pass the wrapper. My example would be to pass the wrapper class to anywhere that is creating parameter callbacks. That is in contrast to the current implementation which has to pass in the `ParameterEventSubscriber` object and store callback handles on its own object.

SteveMacenski, 2019-12-04:
> I think it's better code design

I agree by common schools of thoughts you are correct. I, however, disagree with those thoughts :laughing:. I like to use the core tools, light wrappers I'd rather just use the tool itself.

> in a stage yet to remove the thread

I dont think we are either. I was just thinking that we have a node thread for the costmaps and such, and the lifecycle wrapper mostly just does that, so maybe rather than a separate lifecycle node wrapper, we homologate the node thread items to work for both. I dont think that's going to change any time soon.

> If we pass the base class shared ptr, then we don't have access to any wrapper interfaces,

There's not many interfaces, just 3 overloaded `add_parameter` that control+F tells me _only_ AMCL uses. I'd argue that with those overloads that parameter-like things happen at the highest level ROS-wrapper layer in most software so we don't have to and probably should try to make everyone fall in like with a light wrapper. By "ROS2 design" it seems like you actually just want to create a new parameter interface implementation (https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/node_interfaces/node_parameters_interface.hpp) not actually a full lifecycle node. I wouldn't nitpick this because I'd do exactly what you did to, but something to point out.

As much as possible, I think these projects (packages) should be thought of as separate projects living in the same repository. Any removal of `utils` like tools is a massive positive to me. There's alot of baggage, abet shrinking, for taking individual packages and working with them from light wrappers like this that don't add much additional utility.  

Simple action server does alot, that's a substantial wrapper that is good to have a utility for, but I'd argue that should really be its own project / repo not in a `utils` package. A `utils` package is considered an anti-pattern in ROS. So removing as many of these as possible has been a passive goal of mine and I think this is a good candidate for that.

Anyhow. Your suggestion would work, but I'd avoid making it too deep through the stacks except where actually required.

bpwilcox, 2019-12-05:
> > I like to use the core tools, light wrappers I'd rather just use the tool itself.
> 
For me it is case dependent.

> >  By "ROS2 design" it seems like you actually just want to create a new parameter interface implementation (https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/node_interfaces/node_parameters_interface.hpp) not actually a full lifecycle node.
> 
Yes, I think that would be preferable.  I'll try knocking on that door as soon as they merge my PR.

closed 2020-02-15 by SteveMacenski