ros-navigation/navigation2 #1410: Multiple gzclient instances launched

orduno, 2019-12-04:
Multiple gzclients are launched when running `nav2_tb3_simulation_launch.py`.

Bug was introduced in https://github.com/ros-planning/navigation2/pull/1349.

orduno, 2019-12-05:
For some reason, the condition for executing the process evaluates to `true` even when the `use_simulator` substitution is `false`. 

https://github.com/ros-planning/navigation2/blob/c2d93596084968376ccb3d6213621f1cf59fe7b5/nav2_bringup/bringup/launch/nav2_tb3_simulation_launch.py#L138-L141

orduno, 2019-12-05:
Seems like [compounding conditions](https://github.com/ros2/launch/issues/105) is still an open problem.

mlherd, 2019-12-05:
hmm.. interesting. I tested headless=true and false conditions, but I didn't test use_simulator argument since I didn't think anyone would launch tb3_simulation without simulation :) but you proved me wrong :D

PR #1411 (2019-12-05): Fix launching multiple gzclient instances
---

## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | #1410 |
| Primary OS tested on | Ubuntu 18.04 |
| Robotic platform tested on | Gazebo simulation of TB3 |

---

## Description of contribution in a few bullet points

Fixes #1410, the problem is here:

https://github.com/ros-planning/navigation2/blob/c2d93596084968376ccb3d6213621f1cf59fe7b5/nav2_bringup/bringup/launch/nav2_tb3_simulation_launch.py#L138-L141

Where the `and` operation over two conditions is not well defined.
closed 2019-12-05 by orduno