ros-navigation/navigation2 #1429: Waypoint follower fails to transition to inactive state during reset or pause

bpwilcox, 2019-12-10:
## Bug report

**Required Info:**

- Operating System: Ubuntu 18.04
- Version: master
- DDS implementation: Fast-RTPS,

#### Steps to reproduce issue
- launch navstack (e.g. `ros2 launch nav2_bringup nav2_tb3_simulation_launch.py`)
- reset/pause lifecycle nodes

#### Expected behavior
- nodes should transition to unconfigured or inactive
#### Actual behavior
- waypoint_follower node fails to transition to inactive

#### Additional information
Error logs:
```
[lifecycle_manager-5] [INFO] [1576008356.397068440] [lifecycle_manager]: Resetting managed nodes...
[lifecycle_manager-5] [INFO] [1576008356.397128412] [lifecycle_manager]: Deactivating waypoint_follower
[waypoint_follower-12] [INFO] [1576008356.397551297] [waypoint_follower]: Deactivating
[waypoint_follower-12] [ERROR] [1576008356.397649320] []: Caught exception in callback for transition 14
[waypoint_follower-12] [ERROR] [1576008356.397663382] []: Original error: std::future_error: No associated state
[waypoint_follower-12] [WARN] [1576008356.397677791] []: Error occurred while doing error handling.
[waypoint_follower-12] [FATAL] [1576008356.397688899] [waypoint_follower]: Lifecycle node entered error state
[lifecycle_manager-5] [ERROR] [1576008356.398766738] [lifecycle_manager]: Failed to change state for node: waypoint_follower
[lifecycle_manager-5] [ERROR] [1576008356.398811592] [lifecycle_manager]: Failed to reset nodes: aborting reset

```

SteveMacenski, 2019-12-10:
I think that's a simple action server issue https://github.com/ros-planning/navigation2/blob/master/nav2_util/include/nav2_util/simple_action_server.hpp#L183

If you see the `Deactivating` printout in waypoint follower:

```
nav2_util::CallbackReturn
WaypointFollower::on_deactivate(const rclcpp_lifecycle::State & /*state*/)
{
  RCLCPP_INFO(get_logger(), "Deactivating");

  action_server_->deactivate();

  return nav2_util::CallbackReturn::SUCCESS;
}
```

then its the action server I believe

bpwilcox, 2019-12-10:
@SteveMacenski Yes, I tracked it to this line in the `simple_action_server`:
https://github.com/ros-planning/navigation2/blob/3c736f87dfcbba029e92f2f40de332073aba742a/nav2_util/include/nav2_util/simple_action_server.hpp#L213

This is called during the `is_running` function inside `deactivate`

bpwilcox, 2019-12-10:
I was able to successfully deactivate the node and action server but only after sending an action goal for the waypoint follower. It looks like we are not initializing the future `execution_future_` until the goal handle has been accepted and it creates the async thread, so if deactivate is called before then, the future has no state, and the node crashes/goes to error state.

bpwilcox, 2019-12-10:
I'm thinking the (simplest) fix for this could just be to catch the exception and return from the deactivate function. I'll submit a PR.

SteveMacenski, 2019-12-10:
@bpwilcox you can just check if the future is valid in `is_running()`: https://en.cppreference.com/w/cpp/thread/future/valid.

Or if you want, you can just add

```
if (!execution_future_.valid()) {return;}
```

at the start of `deactivate()` and just miss everything

bpwilcox, 2019-12-10:
> ```
> if (!execution_future_.valid()) {return;}
> ```
> 
> at the start of `deactivate()` and just miss everything
>
I wasn't aware of this function. I'll update my PR with this method, I think it is a bit cleaner.

bpwilcox, 2019-12-18:
Resolved with #1430

closed 2019-12-18 by bpwilcox