ros-navigation/navigation2 #1433: Why do we have to take an absolute topic for the scan topic param in costmaps?

maxlein, 2019-12-12:
I don't know why we have to take an absolute topic name for the costmap scan topic param.  
Is this just a bug, or as intended?  

If I have namespaced robots, the costmap node runs already in a namespace.  
So if I use `scan` as scan topic and the node runs in namespace `robot_1`, I would want the resulting topic to be `/robot_1/scan`.

crdelsey, 2019-12-12:
That should be fixed in eloquent. I assume you are running into this problem in dashing?

SteveMacenski, 2019-12-12:
@crdelsey I think that's consistent with our multirobot items as well

maxlein, 2019-12-12:
No, I am on eloquent. But I need to check this repos branch to be sure.

SteveMacenski, 2019-12-12:
We decided on the global namespace for the lasers when we removed it from the launch remap so that we were only remapping TF and other topics like in rviz that we don't have control over the namespacing of. If you have `/robot1/scan` you should provide that in a configuration file.

maxlein, 2019-12-12:
I don‘t geht that. Where do we have no control over namespacing? Rviz config files? ... Ohh, I think I got it. Rviz is also used to look at multiple robots at the same time... That‘s what I don‘t do. I just start Rviz namespaced also, with relative paths in the config. So I can use the config for multiple robots.

crdelsey, 2019-12-12:
Ok. I see. 

> So if I use scan as scan topic and the node runs in namespace robot_1, I would want the resulting topic to be /robot_1/scan

Because our costmaps run in a namespace already (to differentiate between local and global costmap), if you just specified `scan`, you'd end up with `/robot_1/local_costmap/scan`

As @SteveMacenski said, you can override the topic using parameters as is done here:
https://github.com/ros-planning/navigation2/blob/c6d98d99c1c018c5d3e0dc2bf1105b10066639e9/nav2_bringup/bringup/params/nav2_multirobot_params_1.yaml#L130

I don't have a good fix for this yet. I'd like to default to something like `../scan` to remove the costmap namespace but keep any other scoping. I don't think ROS 2 supports that though

SteveMacenski, 2019-12-12:
I view this as a quirk to have many some documentation around, but I don't think its a bug requiring a fix. It's just how ROS2 is with namespacing and parameters now.

maxlein, 2019-12-13:
> Because our costmaps run in a namespace already

I forgot about this part, true!

SteveMacenski, 2019-12-13:
I would tend to agree, I haven’t looked at the namespaces recently, but I thought we’d removed that with the last batch of multirobot PRs?

crdelsey, 2019-12-13:
> It's a bit confusing to have two times the same name.

Yeah. I think that's my fault. It made sense at the time, I'm sure. :-)

We should do something like `/planner_server/costmap` and `/controller_server/costmap` now, given the new architecture. Can we have a node named `/planner_server` and a namespace with the same name at the same time?

SteveMacenski, 2019-12-13:
I still prefer the costmaps retain their namespaces at `global_costmap` and `local_costmap`. I would agree that if our intention for the long term future is to have the costmaps in the processes themselves, that would make sense to separate them out like what you suggest. However, memory recalls that there was resistance to me changing that and folks wanted a clear separation as to not bake in the structure for when it may eventually be moved out.

SteveMacenski, 2019-12-13:
```
/local_costmap/clearing_endpoints
/local_costmap/costmap
/local_costmap/costmap_raw
/local_costmap/costmap_updates
/local_costmap/footprint
/local_costmap/local_costmap/transition_event
/local_costmap/parameter_events
/local_costmap/published_footprint
/local_costmap/rosout
/local_costmap/voxel_grid

```

I don't see the double namespacing in non-multi-robot modes

crdelsey, 2019-12-18:
> I don't see the double namespacing in non-multi-robot modes

i believe @maxlein's concern was in the way you have to say /local_costmap/local_costmap anywhere you need to refer to a node by name. eg. `ros2 node info /local_costmap/local_costamp` where the first name is the namespace and the second is the node name.

SteveMacenski, 2020-01-14:
@crdelsey @maxlein reading over this now, are there any action items? I think we answered the question and I think its a feature, not a bug.

Is there an action item to leave this open for?

crdelsey, 2020-01-14:
I think we need an upstream feature - relative topic paths (eg. `../scan`) - to address @maxlein's issue. Without that, we can't address the main concern.

I'm OK with closing this and revisiting the topic when that feature exists.

SteveMacenski, 2020-01-14:
We could in theory make that feature, but that starts getting pretty involved. 

Closing

closed 2020-01-14 by SteveMacenski