ros-navigation/navigation2 #1447: Handle missing lifecycle node

maxlein, 2019-12-18:
Is there a reason not to detect missing / non-existant lifecycle nodes?  

See this example:  
I start up a lifecycle node like this:  
```
start_lifecycle_wo_loc_manager_cmd = Node(
        condition=IfCondition(PythonExpression(['not ', with_localization])),
        package='nav2_lifecycle_manager',
        node_executable='lifecycle_manager',
        node_name='lifecycle_manager',
        node_namespace=LaunchConfiguration('namespace'),
        output='screen',
        parameters=[{'use_sim_time': use_sim_time},
                    {'autostart': autostart},
                    {'node_names': ['map_server',
                                    'local_costmap',
                                    'planner_server',
                                    'recoveries_server',
                                    'bt_navigator',
                                    'waypoint_follower']}])
```  

All looks good, the only problem is `local_costmap` is wrong, the name is `controller_server`  
And for a long time I was looking at this:  
`[osg_1.lifecycle_manager] [INFO]   Configuring`  

So I thought something in costmap layer, parameters, lifecycle manager is wrong.  
Which was not, only the node name had changed and forgot to adapt launch file...

bpwilcox, 2019-12-18:
I think you are right that it should detect (or fail out) when a node is missing instead of being stuck hanging.

The `lifecycle_manager` changes the state of each lifecycle node using the `LifecycleServiceClient`, which I believe the current version will hang waiting on the response from the lifecycle node which in your case does not exist and forever waits.
https://github.com/ros-planning/navigation2/blob/a71ea592ca1d32aab300e180a298eb0eefd0fb72/nav2_lifecycle_manager/src/lifecycle_manager.cpp#L145

I think we want to use the other version of the `change_state` which uses a timeout and we could return if it was not found in the requested timeout: 
https://github.com/ros-planning/navigation2/blob/a71ea592ca1d32aab300e180a298eb0eefd0fb72/nav2_util/src/lifecycle_service_client.cpp#L48

SteveMacenski, 2019-12-18:
@bpwilcox I think as a general rule, we should always use a timeout for calling servers. I'd even recommend just deleting the other option and only using the timeout version

SteveMacenski, 2020-01-13:
@maxlein is this something we can expect a PR from you on?

maxlein, 2020-01-14:
Naive me thought this is just adding a parameter...  

Looking at the code here:  
https://github.com/ros-planning/navigation2/blob/a71ea592ca1d32aab300e180a298eb0eefd0fb72/nav2_util/src/lifecycle_service_client.cpp#L58-L66  

`wait_for_service` is called twice, in the function and when invoking. Don't know why.  
When I add a timeout to the first one, I can't trigger a timeout. It just blocks forever.  
So can't we just remove the `wait_for_service` call, as it's called anyway?  
I also like the repeating error message more when calling invoke..

SteveMacenski, 2020-01-18:
Are you sure that the `wait_for_service()` is implemented the way you think?

```
  void wait_for_service(const std::chrono::nanoseconds timeout = std::chrono::nanoseconds::max())
  {
    auto sleep_dur = std::chrono::milliseconds(10);
    while (!client_->wait_for_service(timeout)) {
      if (!rclcpp::ok()) {
        throw std::runtime_error(
                service_name_ + " service client: interrupted while waiting for service");
      }
      rclcpp::sleep_for(sleep_dur);
    }
  }
```

It looks like that's the timeout for the calls, in an infinite loop. I think you just need to change the meaning of the `timeout` parameter and remove the while loop, that's a really unintuitive implementation.

maxlein, 2020-02-05:
[Current state](https://github.com/maxlein/navigation2/commit/2768f3802f4e05105de1a62819c2813d04cb5986), made some comments there.   

So if I get you right by saying:  
>   In general I agree with adding timeouts to all calls, except lifecycle where I think its strictly required to succeed to bring up  

, you mean that `change_state()` should not timeout? Because there could be a long running transition, like a sensor booting up?

https://github.com/ros-planning/navigation2/blob/075208a23b50a4429406937412b2756738ed144e/nav2_lifecycle_manager/src/lifecycle_manager.cpp#L136-L148  

But only if the service is available, `invoke()` should take the time the server needs to transition?  
So calling `invoke()` should block, but wait for service can have a timeout because the service server should be reachable!?  

And `invoke()` doesn't need to have a `wait_for_service()` call also in itself, doesn' it?

SteveMacenski, 2020-02-05:
> Because there could be a long running transition, like a sensor booting up?

Sure, I think that's also a good reason. I was just thinking that if you've gone out of your way to use lifecycle nodes to get some kind of deterministic bringup and shutdown, you're specifying the nodes that should be up. In the case where one of these is incorrect, that should be an immediate failure (by lifecycle not coming up) and not just gloss over it. This could be masking a typo where then something important (like a safety system or a controller) isn't actually brought up.

SteveMacenski, 2020-06-29:
@maxlein with some distance, what do you think about this? It seems like with a typo with an invalid server, failing to come up seems like a reasonable behavior since that's a critical configuration error.

I'm inclined to close this as wontfix, but what do you think?

maxlein, 2020-06-30:
When I remember this right:

- we could fail when the service is not available 

- if the service is available, block on the change state call without timeout as the transition could take some time.

Is this in your opinion not a solution for the problem?  
Or am I missing something? Can the startup of the service server itself take more time?  

In our code, we just loop over it, if there is an error.   

All in all it‘s not a situation that comes very often but if it does, it can take you some time finding the bug.

SteveMacenski, 2020-06-30:
Well I think an assumption of lifecycle manager is that you give it lifecycle nodes to work with. If you give it a random string or an incorrect name, then I'm not sure what to expect lifecycle manager to do. 

We could add a timeout for checking if available, that would be reasonable, but what do we do after it tries and fails? Crash? 

If the service is available, then I agree, we wait until we get a response back. 

Overall, I think we could add a timeout for wait until available and then do [something] if it isn't a real service. This is though a result of a bad configuration so there's going to be odd behavior so just stalling out I think could be definitely made better, but not totally unexpected. 

What do you think the [something] is we should do? Fail via a crash, bring everything down and then exit?

maxlein, 2020-07-03:
I think bringing other nodes down first and then shutting down the manager.  
But it depends. If you are working with a single robot and no fleet control server or something, then it's not that of a problem to just have the lifecycle manager shut down.  
With a fleet controller you may send information on which nodes have problems and let the server act accordingly.

SteveMacenski, 2020-07-06:
OK. I think if it can't talk to the server (presumably doesn't exist by that name), then after a reasonable wait, it prints fatal errors, shuts everything down, and exits.

Is that something you'd be interested in implementing?

maxlein, 2020-07-07:
It was already a goal to have a pr for you as we don't want to have our own LifecycleManager running.   
So should be possible in the next two weeks.

SteveMacenski, 2020-12-14:
@maxlein did you get a chance to work on this?

SteveMacenski, 2021-01-28:
Closing - since this is misconfiguration, there's not much of an action item here if there isn't a strong desire to add user protections.

closed 2021-01-28 by SteveMacenski