ros-navigation/navigation2 #1450: nav2_bt_navigator ignores `/goal_pose` topic messages

hidmic, 2019-12-20:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Version or commit hash:
  - ea42a7d670ad80d03c4612c92bb4160665f072f8
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue

I had this happen on a private project. I was unable to bring up TB3 simulation myself for a simple repro (had issues with the lifecycle manager right out of the box), but it should boil down to:

 - Bring up TB3 navigation in Gazebo simulation.

 - Send a goal using RViz Set Goal plugin (instead of `navigation2`'s).

#### Expected behavior

`nav2_bt_navigator` and the navigation stack acknowledge the goal sent.

#### Actual behavior

Nothing happens.

#### Additional information

Issue seems to stem from an action name mismatch between action client and server in `nav2_bt_navigator`:

https://github.com/ros-planning/navigation2/blob/ea42a7d670ad80d03c4612c92bb4160665f072f8/nav2_bt_navigator/src/bt_navigator.cpp#L58-L59
 
https://github.com/ros-planning/navigation2/blob/ea42a7d670ad80d03c4612c92bb4160665f072f8/nav2_bt_navigator/src/bt_navigator.cpp#L73-L78

SteveMacenski, 2019-12-20:
https://github.com/ros-planning/navigation2/pull/1434

This PR is handling it, its in review

SteveMacenski, 2020-02-11:
@hidmic any movement for fixing that small typo and testing that it works?

closed 2020-02-13 by SteveMacenski