ros-navigation/navigation2 #1453: Is there a list of documentation items that we want to tackle?

yathartha3, 2019-12-24:
After the eloquent release, one of our priorities was to improve documentation.
Can we create a list of them so that it is clear what needs to be done?

SteveMacenski, 2020-01-14:
I think there's a few tickets open on this @yathartha3 
https://github.com/ros-planning/navigation2/issues/1396
https://github.com/ros-planning/navigation2/issues/1092
https://github.com/ros-planning/navigation2/issues/1317
https://github.com/ros-planning/navigation2/issues/1228

In addition to doxygen in all the remaining headers missing ( I think we're about 50% )

yathartha3, 2020-01-14:
@SteveMacenski Got it.

SteveMacenski, 2020-02-07:
Closing - doc tickets are still open (well the 2 that haven't already been dealt with)

closed 2020-02-07 by SteveMacenski