ros-navigation/navigation2 #1459: KinematicParameters dynamic parameters callback needs thread synchronization

crdelsey, 2020-01-13:
This needs some thread synchronization. The kinematic parameters are used on a different thread than the reconfigure call back.

The simple fix would be to add a mutex to the `KinematicParameters` class that we take in `on_parameter_event_callback` and every function that accesses these parameters. However, that's not going to be very efficient; every parameter has an individual accessor method.

Instead this class should be refactored so clients can grab the set of parameters instead of individual parameters.

_Originally posted by @crdelsey in https://github.com/ros-planning/navigation2/pull/1456#issuecomment-573881732_

gimait, 2020-03-24:
I have been taking a look at this issue, and I would like to help with it.

I don't have so much experience with multi threaded applications, so here goes a stupid question: Is it really necessary to protect these variables in this case? It looks like there is only two moments that these parameters are written: during initialization or in service request, and there isn't any other operations happening than read and write.

If I understand all this correctly, in this case we could be okay with some undefined behaviours: The two functions writing to these values are called rarely, and do not depend in other operations from parallel threads (they write, independently of the status of the other threads). 
There could be a problem with the getters though: it could be that they are accessed at the same time as a service request changes a value, and then we wouldn't know if they would get the old or new value. However, since they are only reading, they would definetely get the updated value in the next getter call. Is this right?

Assuming that indeed this was a quite stupid question, I'll ask about what would be the best way to go:
I believe that using atomic variables would be the cleanest solution here. I would avoid creating a new class including the different parameters to be able to pass all of them on a single getter, because that would be redundant (the ```KinematicParameters``` class is already used only to manage these parameters). And then, I think we all could agree that in this case the atomic variables would be cheaper and more comfortable than adding mutexes.

Should I then try with the atomic variables?

SteveMacenski, 2020-03-24:
They're used in several places if you search, outside of initialization and reconfiguration:
- https://github.com/ros-planning/navigation2/blob/f3a138124ae3d24a52c0963d971f29993a6cebf3/nav2_dwb_controller/dwb_plugins/src/standard_traj_generator.cpp#L184
- https://github.com/ros-planning/navigation2/blob/aaa97902c1e1bb9a509c4ee0d88b880afb528f20/nav2_dwb_controller/dwb_plugins/src/xy_theta_iterator.cpp#L65
- https://github.com/ros-planning/navigation2/blob/f3a138124ae3d24a52c0963d971f29993a6cebf3/nav2_dwb_controller/dwb_plugins/src/kinematic_parameters.cpp#L126

For software, the goal is to never have undefined behavior-- we need to ensure that we're not accessing or writing variables at the same time another thread may be writing to these variables. 

Atomic variables could work -- additionally, you could create `SomeStructWithKinemeticParameters` that itself is atomic. What that buys you then is the ability to do something like:

```
auto kine = kinematics->getKinematics();
kine.whatever_variable + kine.another_var;
```

Which allows the code to only try to access the locked down resource via atomic or mutex once, copy its contents and then use that new contents without an issue. Doing that, it may make sense to move the `getX()` `getParameter()` etc in the `.hpp` file to the struct so its `kine.getX()`. Then the class `KinematicParameters` could be renamed to `ReconfigurableKinematicParameters` because then its just a ROS reconfiguration wrapper on top of the struct.

gimait, 2020-03-25:
Makes sense! I'll try with the struct and see how it looks, then create a pr.

closed 2020-04-02 by SteveMacenski