ros-navigation/navigation2 #1465: System tests are failing

SteveMacenski, 2020-01-14:
```
    [INFO] [test_occ_grid_node-2]: process started with pid [6602]
    [map_server-1] [INFO] [1578971956.550386454] [map_server]: Creating
    [map_server-1] [INFO] [1578971957.544490694] [map_server]: Configuring
    [map_server-1] [INFO] [1578971957.544962477] [map_server]: OccGridLoader: Creating
    [map_server-1] [INFO] [1578971957.545020939] [map_server]: OccGridLoader: Configuring
    [map_server-1] [INFO] [1578971957.545069657] [map_server]: Loading yaml file: ../testmap.yaml
    [map_server-1] [INFO] [1578971957.545946159] [map_server]: Loading image_file: ../testmap.png
    [map_server-1] [INFO] [1578971957.549803282] [map_server]: Activating
    [map_server-1] [INFO] [1578971957.549823206] [map_server]: OccGridLoader: Activating
    [map_server-1] [INFO] [1578971957.551137303] [map_server]: OccGridLoader: Handling GetMap request
    [map_server-1] [INFO] [1578971957.551775461] [map_server]: Deactivating
    [map_server-1] [INFO] [1578971957.551793994] [map_server]: OccGridLoader: Deactivating
    [map_server-1] [INFO] [1578971957.551930737] [map_server]: Cleaning up
    [map_server-1] [INFO] [1578971957.551942753] [map_server]: OccGridLoader: Cleaning up
    [map_server-1] [INFO] [1578971958.556257508] [map_server]: Configuring
    [map_server-1] [INFO] [1578971958.556468803] [map_server]: OccGridLoader: Creating
    [map_server-1] [INFO] [1578971958.556493557] [map_server]: OccGridLoader: Configuring
    [map_server-1] [INFO] [1578971958.556506727] [map_server]: Loading yaml file: ../testmap.yaml
    [map_server-1] [INFO] [1578971958.556655667] [map_server]: Loading image_file: ../testmap.png
    [map_server-1] [WARN] [1578971964.560066947] [rcl_lifecycle]: No transition matching 1 found for current state inactive
    [map_server-1] [ERROR] [1578971964.560107536] []: Unable to start transition 1 from current state inactive: Transition is not registered., at /home/jenkins-agent/workspace/packaging_linux/ws/src/ros2/rcl/rcl_lifecycle/src/rcl_lifecycle.c:327
    [map_server-1] [WARN] [1578971970.563852749] [rcl_lifecycle]: No transition matching 1 found for current state inactive
    [map_server-1] [ERROR] [1578971970.563885840] []: Unable to start transition 1 from current state inactive: Transition is not registered., at /home/jenkins-agent/workspace/packaging_linux/ws/src/ros2/rcl/rcl_lifecycle/src/rcl_lifecycle.c:327
    [map_server-1] [WARN] [1578971976.567907673] [rcl_lifecycle]: No transition matching 1 found for current state inactive
    [map_server-1] [ERROR] [1578971976.567939133] []: Unable to start transition 1 from current state inactive: Transition is not registered., at /home/jenkins-agent/workspace/packaging_linux/ws/src/ros2/rcl/rcl_lifecycle/src/rcl_lifecycle.c:327
    [map_server-1] [INFO] [1578971976.568176760] [map_server]: Activating
    [map_server-1] [INFO] [1578971976.568198005] [map_server]: OccGridLoader: Activating
    [map_server-1] [INFO] [1578971976.569512998] [map_server]: OccGridLoader: Handling LoadMap request
    [map_server-1] [INFO] [1578971976.569558213] [map_server]: Loading yaml file: /opt/overlay_ws/src/navigation2/nav2_map_server/test/invalid_image.yaml
    [map_server-1] [INFO] [1578971976.569776768] [map_server]: Loading image_file: /opt/overlay_ws/src/navigation2/nav2_map_server/test/invalid.png
    [map_server-1] [ERROR] [1578971976.570004786] [map_server]: Failed to load image file /opt/overlay_ws/src/navigation2/nav2_map_server/test/invalid.png for reason: Magick: Unable to open file (/opt/overlay_ws/src/navigation2/nav2_map_server/test/invalid.png) reported by magick/blob.c:3002 (OpenBlob)
    [map_server-1] [INFO] [1578971976.570400269] [map_server]: Deactivating
    [map_server-1] [INFO] [1578971976.570413057] [map_server]: OccGridLoader: Deactivating
    [map_server-1] [INFO] [1578971976.570539651] [map_server]: Cleaning up
    [map_server-1] [INFO] [1578971976.570546088] [map_server]: OccGridLoader: Cleaning up
    [ERROR] [test_occ_grid_node-2]: process has died [pid 6602, exit code 1, cmd '/opt/overlay_ws/build/nav2_map_server/test/component/test_occ_grid_node'].
```

mkhansen-intel, 2020-01-14:
Can you provide the template info: what branch of nav2, what branch of ros2, etc? How to reproduce?

:)

SteveMacenski, 2020-01-14:
Ahhhhhh ok ;-)

I'm seeing this in CI for most of the PRs. So Foxy / master, navigation2 master. Reproduce by looking at CI. I see this on at least the following:

https://circleci.com/gh/ros-planning/navigation2/7690?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-build-link

https://circleci.com/gh/ros-planning/navigation2/7682?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-build-link

mkhansen-intel, 2020-01-15:
So I ran this manually and it is passing on my system, however I haven't updated to the latest ros2 repos.

Before I do that can you just try one thing on your system:

`cd` into your `build/nav2_map_server` folder

`ctest -V` 

Tell me if it's passing for you, also how recently you updated your ros2 repos.

SteveMacenski, 2020-01-15:
Yeah things are passing on my end but I haven't updated since before ROSCon. I'm not sure its too productive to test on my local machine since the error that's displayed above is in CI. Things are working fine (well, except other known issues) locally otherwise

mkhansen-intel, 2020-01-15:
I'll update my ros2.repos today and test this again.

mkhansen-intel, 2020-01-15:
The error message from magick is expected, as the test is testing the invalid file condition:
```
[map_server-1] [ERROR] [1578971976.570004786] [map_server]: Failed to load image file /opt/overlay_ws/src/navigation2/nav2_map_server/test/invalid.png for reason: Magick: Unable to open file (/opt/overlay_ws/src/navigation2/nav2_map_server/test/invalid.png) reported by magick/blob.c:3002 (OpenBlob)
```
I don't think there's anything wrong with the test, but don't know why this error is occuring:
```
[ERROR] [test_occ_grid_node-2]: process has died [pid 6602, exit code 1, cmd '/opt/overlay_ws/build/nav2_map_server/test/component/test_occ_grid_node'].
```
What would cause the node to die in CI?

mkhansen-intel, 2020-01-15:
Just to confirm, I pulled the latest ros2.repos and re-ran on my system and the nav2_map_server tests are passing. I don't know how to reproduce this issue. 

Maybe @ruffsl can look into it?

SteveMacenski, 2020-01-15:
Are they moved to the install space with the same permissions? Maybe it just can't be opened. 

Also, it makes sense that it crashes @mkhansen-intel. If you look at occ grid loader where that log is coming from, L161, it returns false. That triggers L176 to throw an exception inside of the `on_configure` step, which won't have exception handlers so it crashes.

mkhansen-intel, 2020-01-15:
This isn't happening during the `on_configure` step. That error is in the `load_map_callback` lambda function, see the message printed here:
https://github.com/ros-planning/navigation2/blob/94283482d4e61ab323f5d9fb134efa5dbf3cc3b2/nav2_map_server/src/occ_grid_loader.cpp#L208

It's expected to fail when it tries to load "invalid.png".

SteveMacenski, 2020-01-15:
Same logic applies: https://github.com/ros-planning/navigation2/blob/master/nav2_map_server/src/occ_grid_loader.cpp#L210

It still returns false, throws exception, nothing handles it.

mkhansen-intel, 2020-01-15:
The exception is caught here:
https://github.com/ros-planning/navigation2/blob/7097e6944c8617cccbf93286b0d61ff70b580518/nav2_map_server/src/occ_grid_loader.cpp#L159
Then the response value is set, the function returns false and the lambda returns the response.
https://github.com/ros-planning/navigation2/blob/7097e6944c8617cccbf93286b0d61ff70b580518/nav2_map_server/src/occ_grid_loader.cpp#L163

I don't see an exception not being caught.

ruffsl, 2020-01-15:
> Maybe @ruffsl can look into it?

I just did a fresh build/test against 6fa310c1e3fc8fed5a291bb4f0fc3eea3648c7dc using `osrf/ros2:nightly` like so:

``` terminal
git clone https://github.com/ros-planning/navigation2.git
cd navigation2
docker build --pull -t nav2:latest .
mkdir -p /tmp/overlay_ws/log
docker run -it --rm \
    -v /tmp/overlay_ws/log:/opt/overlay_ws/log \
    nav2:latest bash
colcon test --retest-until-pass 2
colcon test-result --verbose
exit
cd /tmp/overlay_ws/log/latest_test/
```

Side note: Why are the tests now so painfully long (~40min of tests), could we have them fail a bit quicker? It kind of bogs down the CI, and given its mostly `nav2_system_tests`, test parallelization via splitting over packages won't help much in this respect.

Output:

<details>

```
# colcon test-result --verbose
build/nav2_map_server/Testing/20200115-2227/Test.xml: 1 test, 0 errors, 1 failure, 0 skipped
build/nav2_map_server/test_results/nav2_map_server/test_occ_grid_node.xml: 1 test, 0 errors, 1 failure, 0 skipped
- nav2_map_server test_occ_grid_node.missing_result
  <<< failure message
    The test did not generate a result file:
    
    [INFO] [launch]: All log files can be found below /root/.ros/log/2020-01-15-22-27-47-691619-3407e193aef4-972
    [INFO] [launch]: Default logging verbosity is set to INFO
    [INFO] [map_server-1]: process started with pid [975]
    [INFO] [test_occ_grid_node-2]: process started with pid [976]
    [map_server-1] [ERROR] [1579127270.836034826] [map_server]: YAML file name is empty, can't load!
    [ERROR] [test_occ_grid_node-2]: process has died [pid 976, exit code -11, cmd '/opt/overlay_ws/build/nav2_map_server/test/component/test_occ_grid_node'].
    [INFO] [map_server-1]: sending signal 'SIGINT' to process[map_server-1]
    [map_server-1] [foonathan::memory] Allocator foonathan::memory::heap_allocator (at (nil)) leaked 39696 bytes.
    [map_server-1] [INFO] [1579127267.798440857] [map_server]: Creating
    [map_server-1] [INFO] [1579127268.794432203] [map_server]: Configuring
    [map_server-1] [INFO] [1579127268.795135297] [map_server]: OccGridLoader: Creating
    [map_server-1] [INFO] [1579127268.795183651] [map_server]: OccGridLoader: Configuring
    [map_server-1] [INFO] [1579127268.795200514] [map_server]: Loading yaml file: ../testmap.yaml
    [map_server-1] [INFO] [1579127268.796609834] [map_server]: Loading image_file: ../testmap.png
    [map_server-1] [INFO] [1579127268.801859837] [map_server]: Activating
    [map_server-1] [INFO] [1579127268.801890334] [map_server]: OccGridLoader: Activating
    [map_server-1] [INFO] [1579127268.804344498] [map_server]: OccGridLoader: Handling GetMap request
    [map_server-1] [INFO] [1579127268.805589519] [map_server]: Deactivating
    [map_server-1] [INFO] [1579127268.805611568] [map_server]: OccGridLoader: Deactivating
    [map_server-1] [INFO] [1579127268.806011871] [map_server]: Cleaning up
    [map_server-1] [INFO] [1579127268.806030965] [map_server]: OccGridLoader: Cleaning up
    [map_server-1] [INFO] [1579127269.816167126] [map_server]: Configuring
    [map_server-1] [INFO] [1579127269.816509443] [map_server]: OccGridLoader: Creating
    [map_server-1] [INFO] [1579127269.816537529] [map_server]: OccGridLoader: Configuring
    [map_server-1] [INFO] [1579127269.816550596] [map_server]: Loading yaml file: ../testmap.yaml
    [map_server-1] [INFO] [1579127269.816782421] [map_server]: Loading image_file: ../testmap.png
    [map_server-1] [INFO] [1579127269.818683619] [map_server]: Activating
    [map_server-1] [INFO] [1579127269.818713403] [map_server]: OccGridLoader: Activating
    [map_server-1] [INFO] [1579127269.820777146] [map_server]: OccGridLoader: Handling LoadMap request
    [map_server-1] [INFO] [1579127269.820818623] [map_server]: Loading yaml file: /opt/overlay_ws/src/navigation2/nav2_map_server/test/testmap.yaml
    [map_server-1] [INFO] [1579127269.821099969] [map_server]: Loading image_file: /opt/overlay_ws/src/navigation2/nav2_map_server/test/testmap.png
    [map_server-1] [INFO] [1579127269.822522463] [map_server]: Deactivating
    [map_server-1] [INFO] [1579127269.822551997] [map_server]: OccGridLoader: Deactivating
    [map_server-1] [INFO] [1579127269.823007924] [map_server]: Cleaning up
    [map_server-1] [INFO] [1579127269.823037334] [map_server]: OccGridLoader: Cleaning up
    [map_server-1] [INFO] [1579127270.832411177] [map_server]: Configuring
    [map_server-1] [INFO] [1579127270.832763022] [map_server]: OccGridLoader: Creating
    [map_server-1] [INFO] [1579127270.832791108] [map_server]: OccGridLoader: Configuring
[... 39 lines cut ...]
build/nav2_system_tests/test_results/nav2_system_tests/test_planner_costmaps.xml: 1 test, 0 errors, 1 failure, 0 skipped
- nav2_system_tests test_planner_costmaps.missing_result
  <<< failure message
    The test did not generate a result file.
  >>>
build/nav2_system_tests/test_results/nav2_system_tests/test_planner_random.xml: 1 test, 0 errors, 1 failure, 0 skipped
- nav2_system_tests test_planner_random.missing_result
  <<< failure message
    The test did not generate a result file.
  >>>
build/nav2_system_tests/test_results/nav2_system_tests/test_waypoint_follower.xml: 1 test, 0 errors, 1 failure, 0 skipped
- nav2_system_tests test_waypoint_follower.missing_result
  <<< failure message
    The test did not generate a result file.
  >>>
build/nav2_util/Testing/20200115-2225/Test.xml: 1 test, 0 errors, 1 failure, 0 skipped
build/nav2_util/test_results/nav2_util/test_actions.gtest.xml: 5 tests, 0 errors, 1 failure, 0 skipped
- nav2_util.ActionTest test_simple_action_preemption_after_succeeded
  <<< failure message
    unknown file
    C++ exception with description "Goal handle is not known to this client." thrown in the test body.
  >>>

Summary: 1158 tests, 0 errors, 18 failures, 0 skipped
```

SteveMacenski, 2020-01-15:
Ah got it, well, in `on_configure` there is an uncaught exception: https://github.com/ros-planning/navigation2/blob/7097e6944c8617cccbf93286b0d61ff70b580518/nav2_map_server/src/occ_grid_loader.cpp#L177

We catch one exception, return false, then use that false to throw another exception. That seems excessive.  I agree there you linked to does not, sorry, I jumped the gun from seeing how `on_configure` worked.

closed 2020-01-16 by SteveMacenski