ros-navigation/navigation2 #1466: Question: Why do we have a custom Node launch class?

SteveMacenski, 2020-01-14:
In `nav2_common` we have a separate `Node` that may make it hard to transition to lifecycle nodes. Why is this here? I'm not seeing alot of documentation about it. I assume it has to do with the remappings, but this should be explicit somewhere and what a path might be back to using the base-Node launch tool.

crdelsey, 2020-01-15:
I believe you are right. We needed to pass remappings from one launch file to another for the multirobot use case, similar to the way namespaces are passed down.

There are two paths back to base Nodes that I can see:
1. Incorporate this capability in the base Node. We looked into that. I don't remember why we didn't do that.
2. Rewrite the multi-robot launch files in way that avoids the need to pass remappings from one launch file to another.

SteveMacenski, 2020-01-15:
I’m not sure what’s preferable to yal. Any opinions? Both sound reasonable.

crdelsey, 2020-01-15:
Well, `2.` is easier said than done. I don't have a method in mind to accomplish this, and @orduno didn't find a reasonable way of doing `2.` either when he wrote this code originally, I presume. But, maybe we can find a way now.

SteveMacenski, 2020-01-15:
https://github.com/ros2/launch_ros/blob/18cc7fc699692e040dde3406c8170fe9f269c254/launch_testing_ros/test/examples/talker_listener_launch_test.py#L44

It appears remapping is supported? The only init item I don't see is `use_remappings` but that can be an easy conditional about whether to call the `Node()` object with the field `remappings` or not (or call with it either way, but just remap global namespace to itself).

crdelsey, 2020-01-15:
As I recall, the problem wasn't that remappings weren't supported in the Node class. The problem was that the remappings were computed up at the multirobot launch file level and needed to be used by all subordinate launch files. There was no good way to pass those remappings on to subordinate launch files.

There is a mechanism in ros2 launch that supports this kind of thing for namespaces - something like `PushNamespace`. At the top level, you can add a PushNamespace action, and all Nodes and launch files below that automatically use that namespace. However, support for remappings didn't exist.

We discussed implementing the remappings mechanism in the upstream Node class using the namespace mechanism as a template, but various issues kept arising. I think one of the problems was that the remappings are a list instead of a single string, so the namespace mechanism wouldn't work without changes.

SteveMacenski, 2020-01-15:
OK. I'll admit I didn't follow all of that but it sounds compelling. Do you feel its worth trying to homologate back to the original `Node()` in some fashion or we should leave as is?

crdelsey, 2020-01-16:
Let me give a concrete example of what we'd ideally like if we were to use the existing base Node action.

1. The multirobot launch file defines two sets of remapping:
   * `robot1_remappings = [('/tf, /robot1/tf'), ('/tf_static, /robot1/tf_static')]`
   * `robot2_remappings = [('/tf, /robot1/tf'), ('/tf_static, /robot2/tf_static')]`

2. It launches two instances of `nav2_tb3_simulation_launch`, each with it's own set of remappings
   * `IncludeLaunchDescription('nav2_tb3_simulation_launch.py'), launch_arguments={'remappings': robot1_remappings}` and similar for robot 2

3. `nav2_tb3_simulation_launch` reads the launch parameter:
   * `my_remappings = LaunchConfiguration('remappings')`

4. `nav2_tb3_simulation_launch` creates the node action with the passed in remappings
   * `talker_node = launch_ros.actions.Node(remappings=my_remappings)`

----

Something like this could be made to work, however, it's not as straightforward as we'd like. At the start and the end, the remappings are a complex datatype - a list of tuples. However, all that structure gets lost as we try to pass the data as parameter; it gets collapsed to a string.

Converting strings back and forth would be fairly easy in straight python, but all the conversions need to happen in the launch context, meaning it needs to be done with launch.actions and substitutions.

-----

To address your question, yes, I think we should address this by fixing the shortcomings of launch in some way. We are unlikely to be the only ones to encounter this problem. 

I've got a long list of launch fixes that really *should* be done, but it's not a code base I like working in. I can make it do what I need, but making a general solution that will work for everybody and has all the required support code always looks like more work than I have time for.

orduno, 2020-01-16:
The custom node class is to support *conditionalizing* the remappings. 

I agree with what @crdelsey has explained.

I looked into possible implementations for:
> Rewrite the multi-robot launch files in a way that avoids the need to pass remappings from one launch file to another.

Like having a [PushRemapping](https://github.com/ros2/launch_ros/issues/56) action or passing the sets of remappings down to included launch files (as Carl explains above). I didn't see a clear path for implementing either.

My recommendation would be to push changes to the changes upstream for now.

SteveMacenski, 2020-01-16:
OK, I like that as well. Is this something @orduno you have the cycles to start? I have no issues pulling the work over and submitting the PR, but frankly if they had followup, I'm not sure I could do it.

crdelsey, 2020-01-16:
I'll look at it. I've got no choice at the moment. There's a bug in the class right now that is preventing me from running the stack at all. 

I'm not sure why nobody else is seeing it, but it looks like a legit bug to me.

SteveMacenski, 2020-01-16:
Can you describe it?

crdelsey, 2020-01-16:
See  #1474

PR #1476 (2020-01-27): Make the nav2_common Node action use the launch_ros Node as super class
## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | fixes #1474, fixes #1466 |
| Primary OS tested on | Ubuntu |
| Robotic platform tested on | TB3 in Gazebo|

---

## Description of contribution in a few bullet points
* Rather than reimplement the launch_ros Node action, our custom class now derives from it.
closed 2020-01-27 by SteveMacenski