ros-navigation/navigation2 #1470: Can't pass map to nav2_bringup_launch.py file

mkhansen-intel, 2020-01-15:
## Bug report

**Required Info:**

- Operating System:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2020-01-15:
@mkhansen-intel its this line https://github.com/ros-planning/navigation2/commit/2873bfc3468a25a47483f22d17a232a83355dfc6#diff-9ced65f168bb2611c9a7bb195a719d11R50

I resolved it in this PR https://github.com/ros-planning/navigation2/pull/1464

It was part of my "why the hell isn't this working for me?" discussion in that PR which I  resolved.

mkhansen-intel, 2020-01-15:
This line of code seems to be the culprit:
https://github.com/ros-planning/navigation2/blob/1ec7b205ddee240273ac6ba5d218d050e8ab4f4c/nav2_bringup/bringup/launch/nav2_bringup_launch.py#L50

I'm not sure why this was changed to the default, or what the side effects of changing it back might be. @orduno can you tell me?

orduno, 2020-01-16:
Changing the default value to an empty string should fix the problem without side effects -- but haven't been able to confirm as am having trouble running nav2.

SteveMacenski, 2020-01-16:
I have confirmed, its in #1464

yathartha3, 2020-01-22:
I tried what @orduno suggested, and changing the default value to an empty string __did solve__ the problem. I am able to run the stack and navigate.

PR #1477 (2020-01-22): fix map loading
Fixes #1470 

### Description of contribution in a few bullet points

Simply set the default value to an empty string as discussed in #1470.
Tested on simulation. Map loads and the robot is able to navigate.
closed 2020-01-22 by SteveMacenski