ros-navigation/navigation2 #1472: LIfeycle manager to python launch_ros LifecycleNode feature

SteveMacenski, 2020-01-16:
Right now, `launch_ros` has a way to transition into lifecycle active states. Its a mess, but it exists. You can request state transitions based on current states and the states of other nodes. If you have even a modestly complex launch file, this will take up 80%+ of your launch file over the actual functional code. Also I haven't figured out how to transition to shutdown on control-C, but I'm not convinced its not possible, but challenging. 

Long-term, I think there should be a reimplementing of the lifecycle manager as a python launch system item. One thought is the `LifecycleNode()` has another flag for `add_to_manager` and in the background it aggregates the *ordered* lifecycle nodes and brings them all up in that fashion. Still need to figure out how to handle the SIGs to transition to shutdown before exiting. 

This would have the same capabilities as today, integrated into launch, and generically usable for all lifecycle node users. Really, I think its **nuts** that launch_ros doesn't have a way to automatically transition nodes into active with a flag, nor a way to easily step through all of your launch nodes through the transitions at the same time. 

CC @dirk-thomas @wjwwood for feedback on this. If this isn't something they like or would merge, I don't want to waste anyone's time.

dirk-thomas, 2020-01-16:
I am not the right person to give you feedback on this. @Karsten1987 @hidmic FYI.

wjwwood, 2020-02-06:
> This would have the same capabilities as today, integrated into launch, and generically usable for all lifecycle node users. Really, I think its nuts that launch_ros doesn't have a way to [...]

The problem, as I see it is: "who is responsible for transitioning lifecycle states"

It's technically possible that:

- a node transitions itself
- a node is transitioned by code in the same process (the `main()` function for instance)
- a node is transitioned via the ROS interface (Services) by a separate ROS node (same or different process), like a "lifecycle manager" or something
- a node is transitioned via the ROS interface (Services) by launch

Each of those cases might make sense in different scenarios, and so we hesitated to flesh out the tools for managing lifecycle nodes (also launch wasn't around at the time). The hesitation is about how those various cases interact with each other and how they're configured.

This is one of the important ways in which I consider lifecycle's implementation to be incomplete.

We could add launch tools for interacting with lifecycle nodes, and I think that would be nice and helpful and we would consider taking them as a pr. But you should just keep in mind this issue that launch might not always be the way people want to manage the lifecycle state of nodes as you're implementing the features.

SteveMacenski, 2020-02-06:
> This is one of the important ways in which I consider lifecycle's implementation to be incomplete.

Fair enough. From what I've seen of Lifecycle it seems to be a work in progress which is fine :smile:. This is just another case to add to your study next time there's being work done in this area. 

I think there's many ways, like you mention, that people might want to play with lifecycle nodes. I think I'd add the following bullet, though:

- Transition from launch system

If the launch system could take in the nodes, in order, with a flag to transition them, that would greatly alleviate what we're trying to do. Long term, we would like to remove the lifecycle manager from Navigation2. We'd likely spin it out into a separate repository since it would have use still if we generalized and cleaned up the code a little. 

We're essentially just using lifeycycle manager to take in the `N` lifecycle nodes, and transition each of them through the steps in the order provided. This is something that seems like a reasonable addition to the launch system. While that may not be the way everyone wants to interact with lifecycle nodes, if some flag is not set to do it, the other 4 options are still available to use to be manually transitioned.

I tried in the ouster ros2 drivers to use the lifecycle / launch system to transition it to the active state, and it was a real pain. If you were to try to create the same logic of stepping them through in order with 3+ nodes, that would get very messy, very quick. Something like:

```
    driver_node = LifecycleNode(package='ros2_ouster',
                                node_executable='ouster_driver',
                                node_name=node_name,
                                output='screen',
                                emulate_tty=True,
                                parameters=[parameter_file],
                                node_namespace='/',
                                auto_start=True,
                                )
```

Where `auto_start` is the new flag, would be relatively easy under the hood to support and drastically simplify the launch files for large lifecycle enabled systems.

wjwwood, 2020-02-06:
Just to expand on a few of those cases that might make sense to me, while I'm thinking about it...

----

> a node transitions itself

In this case we talked about have a setting in each lifecycle node which I called a "lifecycle policy", which would break down into things like:

- do nothing (all externally requested state transitions)
- automatically transition to level X
  - the node will try to transition itself up to a certain level, like "goto to inactive and wait"
  - external managers could then be like "wait for foo_node to get to inactive, then wait for some other condition, then transition foo_node to active, ..."
- all state transitions happen automatically
  - this would be useful when you just want to use the lifecycle state transitions for observation

Maybe others... This would be the first configuration point, and you could imagine various combinations of this with external managers and launch, etc...

----

> a node is transitioned by code in the same process

Imagine the lifecycle manager or launch but in process and maybe safer/real-time due to being just function calls.

> a node is transitioned via the ROS interface (Services) by a separate ROS node (same or different process), like a "lifecycle manager" or something

This is like a user written application which perhaps manages the state of many lifecycle nodes and perhaps changes behavior based on domain specific information.

> a node is transitioned via the ROS interface (Services) by launch

Could fulfill all of the roles below this, but then would tie those features/behaviors to launch. Sometimes that ok and even desired, but not necessarily always. For instance, R&D might start with launch, but move to not using launch, and in that case having lifecycle management be separate from launch. Or just from a performance perspective, having nodes transition themselves might reduce communication overhead on start up compared to having launch transition each one itself.

wjwwood, 2020-02-06:
> I think I'd add the following bullet, though (unless that's what you mean by bullet 4 and I'm misinterpreting):

>    Transition from launch system

Yeah, that's what I meant.

wjwwood, 2020-02-06:
> We're essentially just using lifeycycle manager to take in the N lifecycle nodes, and transition each of them through the steps in the order provided.

Unless you need coordination between them, that might be replaced with the policy I described.

Even if you need coordination, the in-process lifecycle manager might be a good solution when you are composing them anyways.

Otherwise, a separate node that manages the lifecycles ("lifecycle manager") is what I think makes the most sense. If there are performance or scaling issues with that then we should aim to address those, but launch will have the same issues.

If you would want to have launch capture this transition information, then maybe it could dump a config file which it passes to a boilerplate lifecycle manager, rather than doing the transitions itself. Again, the goal being that you can decouple launch's Python implementation from managing the transitions, mostly from a long-looking gaze towards safety and real-time systems where launch's python implementation may not be acceptable.

wjwwood, 2020-02-06:
> If this new flag is defaulted to False, would that resolve your concerns about supporting users that don't want to interact with lifecycle nodes in that way?

I think an option like that would be fine, but I'd like to see the implementation of it be "pass an argument to the node so it transitions itself" rather than "launch will monitor and transition the node for you". But that's just my perspective.

iluetkeb, 2020-06-25:
I'm coming a bit late to the party here, but I definitely agree with putting better lifecycle management into launch. I think it's the right place for it.

We can do the auto_start thing in lifecycle, but having written an autostart into the ros1_lifecycle, I'm a bit afraid this might cause people to just ignore the whole thing by auto-starting everything. Then we would have the same fault-handling within the nodes as we had before.

This should be easy to do right! :-)

My initial idea was to define a dependency structure, like, "A depends on B", "B depends on C". Then we can either make it bottom-up -- once C is there, B automatically activates and so on. Or we make it top-down, where activating A causes first B to be activated, which first causes C to be activated, and then it goes back up. Or both.

SteveMacenski, 2020-06-25:
Well my hope was that the autostart would look at the aggregated lifecycle nodes in the launch tree and bring them up in order (so transition A to configuring, B to configuring, ... N to configuring; A to active, B to active, and so on). So rather than just autostarting themselves from inactive -> active independently, they'd still be tied together by their order. We've have to document that the order in the launch files matter, not that they didn't matter before, but many applications could get by without realizing that.

Would that help with some of your auto-start concerns since there would be a depend / order to bringup. That's how our lifecycle manager in this repo works; you give it an ordered list of nodes and transitions them all to active by stepping them all through each state one at a time. We could also make some object to help the users define those dependencies if we wanted to be more explicit about it, I have no issues with that.

SteveMacenski, 2020-06-25:
We'd definitely be open to spinning out the lifecycle manager to a new repo if there were some lifecycle tooling efforts underway and we could "donate" that (even has > 90% test coverage!). Because its C++, it wasn't trivial to integrate it in its current form into the launch system.

iluetkeb, 2020-06-25:
> Would that help with some of your auto-start concerns since there would be a depend / order to bringup.

Yes, it would.

My main question is whether you'd prefer to annotate this with each node? I would prefer to have it a separate element in the launch file. This make re-using launch files easier -- the planner might depend on the localization, but it wouldn't need to care which localization system there is, because this is specified in a separate element.

> We'd definitely be open to spinning out the lifecycle manager to a new repo if there were some lifecycle tooling efforts underway and we could "donate" that (even has > 90% test coverage!). Because its C++, it wasn't trivial to integrate it in its current form into the launch system.

I was thinking more in the direction of having some macro-like Python functions that would reduce the boilerplate in the launch file. I mean, launch already has the functionality to listen for state changes and react to it, but you have to write down the _entire thing_ _every single time_, when, what we want most of the time is "once node X is started, start node Y".

btw, just a question: I noticed that the current lifecycle manager first configures everything, then activates everything. Is there a particular reason not just transition the first node all the way to active, and _then_ move on to the next?

SteveMacenski, 2020-06-25:
I'm not sure I agree with a new element, though that's a way we could do it and I could live with it. I think we just need another field in the lifecycle node function if we want to explicitly define what a node depends on rather than implicit by ordering. Ex.

```
        LifecycleNode(
            package='nav2_waypoint_follower',
            executable='waypoint_follower',
            name='waypoint_follower',
            output='screen',
            parameters=[configured_params],
            remappings=remappings),
            autostart=True,
            depends=['nodeA', 'nodeB']
```

As you mention, that's less general if you want to swap out one of many localization options that the node may depend on (unless you mask all the names to the same).  The implied order would work well if we just added the include call for the localization launch file after this one in the launch description, then no requirement that it knows anything about the items below. We'd have to have some documentation about best practices for designing launch systems so that the hierarchy of launching is also hierarchy of dependencies. 

>  I noticed that the current lifecycle manager first configures everything, then activates everything. Is there a particular reason not just transition the first node all the way to active, and then move on to the next?

Wasn't my design choice, but intuitively, it makes more sense to me like that. I'd ideally like nothing to be active until the "system" is active, so this leaves less ambiguity when one thing comes online when everything else will be online. I want to think about "the system" being online more than I want to think about individual servers being online. If A,B,C,...N all make it through configuring and start activating, once I see A active, I'm more confident the system will properly activate completely relative to individually where node F could fail configuring due to some bogus configuration file. 

A slightly more structured argument (still not uber strong) would be that I don't want A to be fully active and start showering topics / data at nodes B,C,D,E while activating in case there's some subtle async issues in bringup where a message can get through before that server is prepared to deal with it. If they're all configured before activated, then you know all the memory has been allocated already and the networking interfaces are fully setup. When that first message hits the callback queue, I'm far more confident that it won't crash due to some hard to track 10 microsecond window where not everything is ready.

iluetkeb, 2020-06-26:
Are you going to work on this? I notice you put a "summer sprint" label on it?

SteveMacenski, 2020-06-26:
Not something I have an immediate plan to work on.

SteveMacenski, 2021-07-30:
Closing - as we integrate more and more with the lifecycle manager in C++ it seems to be something that will stick around for a while. I'd love to move to a fully launch-based lifecycle transitioning system, but that is slightly out of the scope of this project and will close this ticket as a result.

closed 2021-07-30 by SteveMacenski

iluetkeb, 2021-07-30:
Yeah, this doesn't seem a navigation issue but more a launch system issue. I still think it is relevant, and we've been using nav2's lifecycle manager enough to also be aware of its limits (which are not severe, but there are some, particularly with respect to starting nodes in response to lifecycle state changes). Maybe we'll pick it up eventually, but it's nothing urgent.