ros-navigation/navigation2 #1475: Fix CircleCI Test Failures on PRs

SteveMacenski, 2020-01-16:
See all PRs active ;-)

crdelsey, 2020-01-16:
I looked through the Circle logs of the latest build on master, and they seem to be failing in the same way as #1474.

SteveMacenski, 2020-02-01:
I started looking at the `nav2_utils` failure first, since a failure in the action server would cause other failures.

Its in the `test_simple_action`, the last PR was https://github.com/ros-planning/navigation2/pull/1356, soon after we stopped passing CI.

it crashes on this line (https://github.com/ros-planning/navigation2/blob/master/nav2_util/test/test_actions.cpp#L240) which is odd. I removed it from the assertion, even removed the assertion totally, and it still crashes on just `rclcpp::spin_until_future_complete(node_, future_result);` as a single line.

From the executor: https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/executor.hpp#L225, I don't see any exception to be thrown, so I think its a legit crash, but I don't know how to get the failure printouts in gtest logging. 

Some other tests that are failing also include a line similar to this one. I wonder what's going on with that or if its related

SteveMacenski, 2020-02-04:
OK just the system tests failing are left

SteveMacenski, 2020-02-07:
I went back through and found some errors in the simple action server that helped stabilize a number of the flaky tests to 100%.

Still working on the action tests.

SteveMacenski, 2020-02-07:
@shivaang12 I'm now seeing the behavior your described:

```
No valid trajectories out of ...!
```

from DWB.

closed 2020-02-10 by SteveMacenski