ros-navigation/navigation2 #1478: Understanding the Custom BT Nodes: Back to fundamentals

SteveMacenski, 2020-01-24:
Hi,

I did a bunch of reading on behavior trees and got a good feeling of the building blocks. It occurs to me that the BT we use in Navigation doesn't contain mostly fundamental building blocks from literature. I think this has the unintended effect of the following:
- BT experts that would like to use the `NavigateToPose` BT as a subtree in their applications won't be able to understand what exactly is happening
- The format of components in Navigation may not comply with fundamental actions that can have reuse.

I see 3 major nodes used that seem to not comply with literature:
- Pipeline
- Round robbin
- Recovery*

[* though it can be found in specific expansions of literature, though this implementation doesn't comply to it, to my understanding].

It seems they can be represented as operations of fundamental building blocks. 

Recovery: Triggers both children but always returns the code of the first node. This can be implemented as a Parallel control node with a decorator node (M=1) on the second child forcing always successful. 

Pipeline: Seems a subtree where we have a sequence control node with child 1 being first action and child 2 is another sequence node. The children of that sequence node is action 2 and another sequence node, recursively.

Round Robbin: That one is more complex, I'm not sure off hand, that may actually be something that can't be assembled (or not easily).

I wanted to get some thoughts on this. Ideally, I'd like to revert to the simplest base operations as possible. If we create sub-trees that are named as above, but implemented using the primitives, that would be also OK.

crdelsey, 2020-01-24:
> It seems they can be represented as operations of fundamental building blocks.

I believe this is true. In fact, I'm pretty sure we made a version out of fundamental building blocks for all three before replacing them with a custom node. In all those cases, the tree made with fundamental building blocks was much more complicated and difficult to understand. 

I've come to the conclusion that it's better to make custom nodes and require people to learn what they do than to have a complicated tree that they need to decode over and over again every time they work with it.

As for these specific nodes:
* Recovery - @mhpanah can speak to this one better than I, but I think this replaced a block of 3 fundamental nodes. We had that in early versions of the behavior tree so you can see what the equivalent is that we replaced. As I recall, the tree became much easier to understand at a glance.

* Pipeline - This is exactly the same as a [reactive sequence](https://www.behaviortree.dev/sequencenode/), except in the case where a child returns running. In our behavior tree library, there is only one node that allows two children to be in the running state at the same time -Parallel. To replace pipeline, you either need to guarantee that the planning subtree can never return running (in which case, use reactive sequence), or you need to build the replacement using the parallel node and enough other nodes to get equivalent behavior. It was pretty ugly when I tried. I got close, but I didn't get all the way. If you take a look at #1336 and see what I did with the "FirstCompletion" node, you'll see some of what's required to make the parallel node work for us. 

* Round Robin - I believe for this one, we came up with an equivalent where the round robin is replaced with sequence and each child becomes a subtree where we use blackboard_check_int and set_blackboard to control which node gets run on a particular tick.

SteveMacenski, 2020-01-24:
Given the BT library V3 allows for subtrees, could we instead create subtrees that implement the same things that we can call Pipeline, RoundRobbin, etc? 

That way its still made of the building blocks but abstracted away into the larger tree?

cdelsey, 2020-01-28:
I was thinking that would be nice as well, but I doubt we can do that with the current implementation of the library. We'd need a way to attach nodes below a subtree. I don't think we can do that. I think a subtree has to be a leaf of the "super-tree"

closed 2020-08-03 by SteveMacenski