ros-navigation/navigation2 #1480: {spatio-temporal, nonpersistent}-voxel eloquent debians don't play nice with nav2 debians

Jconn, 2020-01-31:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 18.04
Eloquent Debians

I tried installing every eloquent package with 

```
sudo apt install ros-eloquent-desktop
sudo apt install "ros-eloquent*"
```

and got a few errors:
```
dpkg: error processing archive /var/cache/apt/archives/ros-eloquent-nonpersistent-voxel-layer_2.1.0-2bionic.20200121.234728_amd64.deb (--unpack):
 trying to overwrite '/opt/ros/eloquent/share/costmap_plugins.xml', which is also in package ros-eloquent-nav2-costmap-2d 0.3.2-1bionic.20200121.234411
Preparing to unpack .../ros-eloquent-spatio-temporal-voxel-layer_2.1.2-2bionic.20200121.234728_amd64.deb ...
Unpacking ros-eloquent-spatio-temporal-voxel-layer (2.1.2-2bionic.20200121.234728) ...
dpkg: error processing archive /var/cache/apt/archives/ros-eloquent-spatio-temporal-voxel-layer_2.1.2-2bionic.20200121.234728_amd64.deb (--unpack):
 trying to overwrite '/opt/ros/eloquent/share/costmap_plugins.xml', which is also in package ros-eloquent-nav2-costmap-2d 0.3.2-1bionic.20200121.234411
Preparing to unpack .../ros-eloquent-plansys2-simple-example-dbgsym_0.0.5-1bionic.20200121.233516_amd64.deb ...
Unpacking ros-eloquent-plansys2-simple-example-dbgsym (0.0.5-1bionic.20200121.233516) ...
dpkg: error processing archive /var/cache/apt/archives/ros-eloquent-plansys2-simple-example-dbgsym_0.0.5-1bionic.20200121.233516_amd64.deb (--unpack):
 trying to overwrite '/usr/lib/debug/.build-id/33/2a432c1181b707c7dc8a37cd227b60bea66bd6.debug', which is also in package ros-eloquent-plansys2-multidomain-example-dbgsym 0.0.5-1bionic.20200121.233711
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/ros-eloquent-nonpersistent-voxel-layer_2.1.0-2bionic.20200121.234728_amd64.deb
 /var/cache/apt/archives/ros-eloquent-spatio-temporal-voxel-layer_2.1.2-2bionic.20200121.234728_amd64.deb
 /var/cache/apt/archives/ros-eloquent-plansys2-simple-example-dbgsym_0.0.5-1bionic.20200121.233516_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

```

I'm filing this here because nav2 is head honcho of all these plugins.

SteveMacenski, 2020-01-31:
> I'm filing this here because nav2 is head honcho

Its all me anyway. The semantics of Nav2 repo vs. STVL repo is purely organizational :wink: 

I just copy pasted the template on all of these, so its no surprise if one's broken, they're all broken (and I need to update my template).

That is really not good that this is being installed outside the package. The line in question is below:

https://github.com/ros-planning/navigation2/blob/master/nav2_costmap_2d/CMakeLists.txt#L145

Now the question is, should that be `share/` or `share/${PROJECT_NAME}`

The other question is "where else might have we messed that up?" across the navigation2 stack.

SteveMacenski, 2020-01-31:
`share/${PROJECT_NAME}` should work, I checked that the rviz plugins debians are correct and they use that.

https://github.com/ros2/rviz/blob/ros2/rviz_default_plugins/CMakeLists.txt#L309

A PR for nav2_costmap_2d, STVL, and NPVL is appreciated.

Jconn, 2020-01-31:
output of  `grep -rni "destination share"` gives me these files with the same issue:
```
nav2_recoveries/CMakeLists.txt:121:  DESTINATION share
nav2_navfn_planner/CMakeLists.txt:71:  DESTINATION share
nav2_costmap_2d/CMakeLists.txt:145:  DESTINATION share
nav2_dwb_controller/dwb_critics/CMakeLists.txt:69:        DESTINATION share
nav2_dwb_controller/dwb_plugins/CMakeLists.txt:72:        DESTINATION share

```

Can I change those too?

SteveMacenski, 2020-01-31:
Yes please

closed 2020-01-31 by SteveMacenski