ros-navigation/navigation2 #1485: Navigation Branding

SteveMacenski, 2020-01-31:
The navigation stack hasn't traditionally had much branding associated with it. [MoveIt](https://moveit.ros.org/) was at one point Arm Navigation, has done an exceptionally good job at creating an active brand presence, logos, website, etc.

I'd love to see the phrasing `Nav Stack`, a derivative, or a complete fresh re-branding take place as part of a V1.0.0 release of the ROS2 Navigation Stack, which I hope to have come out later this year or early 2021 after some roadmap planning.

This may include things like:
- Logo
- Website / graphics
- Branding templates / color scheme

This is a clearly different type of undertaking than typical technical tasks, but I wanted to get some feedback on this idea, thoughts folks had, and generally if there was excitement around something like this.

Why be excited?

**tee shirts** 
**socks**
**keychains**
**stickers**
**tattoos** (ok, that may be a bridge too far, but you get the idea)

mikeferguson, 2020-01-31:
Honestly, the word "stack" could die - it's a carryover from when we organized packages into stacks (which are now metapackages).

SteveMacenski, 2020-01-31:
And metapackages weren't carried over to ROS2 :wink: 

I was looking for something more flashy than `Nav Stack` anyhow. MoveIt is both fun and relevant. I'd like to have some new name for it that has a similar theme

davetcoleman, 2020-01-31:
Let's just roll Navigation into the feature set of MoveIt and call it a day ;-) MoveIt is already suppose to support base planning (and leg planning, and all types of planning).

There's plenty of online cheap logo sites like https://99designs.com/ you could use for Navigation.

caelinsutch, 2020-04-28:
What features would we want on the website? I use [docusaurus](https://v2.docusaurus.io/) for most of the projects I work on, it has modules for documentation, blog, landing page, and pages are written in markdown. 

Is branding also wanted? I can propose some logos and branding guides along with website designs :)

SteveMacenski, 2020-04-28:
Wow, I've never heard of that before, some of those showcase sites are _really_ good. 

I think there's a big opportunity for creativity or asserting your vision on this one. I'd want to have one of the tabs go to the tread the docs that we have for things like tutorials or developer guides, but we could move stuff from RtD to a main website similar to what MoveIt does. 

> Is branding also wanted? I can propose some logos and branding guides along with website designs :)

Definitely, have a shot at it!

caelinsutch, 2020-04-29:
What names are you thinking of? Nav stack does seem outdated, maybe something like `Navvy`, I'd love to hear some suggestions to create some logos with

SteveMacenski, 2020-04-30:
`Nav` had come to mind - but I'm also terrible at naming things. (or `NAV` as an acronym for Navigation for Autonomous Vehicles).

Perhaps `Navigation2` (though I don't particularly like). @mikeferguson any ideas :wink:

shivaang12, 2020-04-30:
How about NAV BASE ? But it sounds more like Move Base.

caelinsutch, 2020-04-30:
@SteveMacenski  I personally like NAV as an acronym, NAV Base I don't think really gets to the core of the project. Navigation2 makes sense, but isn't as cool as the acronym

caelinsutch, 2020-04-30:
Here's three concepts:

https://www.dropbox.com/sh/31d3wefa5qmen8o/AABgGAIsaYBpEOtwN_eDkyVHa?dl=0

I think the red color scheme with the darkmode style would look good

SteveMacenski, 2020-04-30:
Perhaps we should consider a logo that's a little more abstract. [Moveit](https://moveit.ros.org/)'s 2 bar-linkage is simple, clean, and has some elements of a manipulator in it. The "MoveIt" is just tacked onto the side (which also lets us separate the issue of finding a name from finding a logo). 

I've thought about something like a wheel or something, but that implies it won't work for legged robots (which maybe that's just being too nit-picky on my part, so ignore me if that's actually a good idea). Things that are pretty "navigation-y"
- Voxels
- Plans
- Wheels
- cameras / lidars
- the behavior tree
- mobile robot

shivaang12, 2020-05-01:
@caelinsutch commenting on your logos,

I think the logo with black background looks cool but the font is kinda uncool. I think graphically created font with some robot effect might look awesome :wink:

SteveMacenski, 2020-05-01:
Back on naming, I'm starting to come around a little on "Nav" (the non-acronym version) as a short hand. I already use "I'm doing Nav work" as my short hand for navigation as is, and people referred to the old "nav stack" as such. 

Also open for more suggestions. I think that's not a very inspired name change, but I think its in the right direction; something short and catchy ("MoveIt", "Fuse", other good examples in the community)

caelinsutch, 2020-05-01:
> A cool abstract-ish turtle

I gotta say a robot turtle would look dope. I think we just need to figure out direction a little more. I used to be a professional logo/graphic designer, so I'm sure I could handle it. I'll take a shot at the turtle and a few of those other ideas and upload them here.

SteveMacenski, 2020-05-01:
It also doesn't need to be a turtle, it could also be something related to or part of a turtle; like an homage. I'm trying to think of options on that front, obviously a shell or the cool patterns on turtles, the topical sea.

SteveMacenski, 2020-05-08:
Hey just checking in @caelinsutch any fun updates on your adventure?

Shru, 2020-05-11:
Turtle is a really cool idea; like all of the ROS Distro logos. The challenge would be to make it more navigation-y. Can we use the ROS nine dot logo in any form? Found some literature here(https://www.ros.org/press-kit/) but still unclear. 

I was thinking if we should go away from the navigation gps pin marker & the compass arrow symbol or not? It would be great if we can convey "robotics" somehow. 

Here is an idea I sketched out to consider: 
[image]
[image]
The letters "N A V" are the walls/obstacles and the robot plans a rough(can make it more accurate lol) A* path from start to finish :D

SteveMacenski, 2020-05-11:
> Can we use the ROS nine dot logo in any form?

I can ask for permission, but I'm nearly entirely confident the answer will be yes. Though, I would avoid it. I think the Turtle + Navigation themes is enough for a logo, squashing in a ROS logo in there is getting a little too busy.

I personally would avoid any logo that makes use of the name, given that we don't have a name in mind at the moment. Nav / Nav2 are just placeholders. I think good logos should stand on their own!

caelinsutch, 2020-05-12:
> Turtle is a really cool idea; like all of the ROS Distro logos. The challenge would be to make it more navigation-y. Can we use the ROS nine dot logo in any form? Found some literature [...]

Might be cool to explore these ideas with a tiny turtle instead of those shapes.

caelinsutch, 2020-05-12:
> Hey just checking in @caelinsutch any fun updates on your adventure?

Coming out with some today. Work beens busy the last few days, sorry for the delayed response

caelinsutch, 2020-05-12:
I added 4 and 5 into [this folder](https://www.dropbox.com/sh/31d3wefa5qmen8o/AABgGAIsaYBpEOtwN_eDkyVHa?dl=0)

I'm trying to think of a navy symbol, besides the obvious tracing lines for P2P navigation

shrijitsingh99, 2020-05-12:
WIll put in my 2 cents too on this:

> Perhaps we should consider a logo that's a little more abstract. [Moveit](https://moveit.ros.org/)'s 2 bar-linkage is simple, clean, and has some elements of a manipulator in it. 
I like this idea a lot of a subtle reference to navigation, not something very obvious. 
MoveIt's manipulator bar linkage has this very subtle link to arms like Steve mentioned. So people who worked with robotics arms understand the reference but not any random person.

> Nav had come to mind
The acronym is good but I feel the word is too general, 'I am working on nav' might feel like you are working on something like google maps 😅 , I feel people will end up prefixing ROS to it and saying "I work with ROS Nav".

> I gotta say a robot turtle would look dope
I really like the idea of having some sort of turtle reference

> I personally would avoid any logo that makes use of the name
Yeah, the logo should be enough to identify. Like the ROS logo is enough to identify it is ROS

Awesome work till now by @caelinsutch and @Shru 💯

SteveMacenski, 2020-05-12:
Honestly "I'm working on nav" is something I literally say daily. I always refer to this field as "nav" in casual conversation. 

Something subtle like the 2 bar linkage would be awesome. The turtle idea is another direction to go down. 

I really like the logo 5, really clean. Though nothing navigation-y about it. Doesn't need to be much, just a little detail. 4 I think is too much nav detail. I generally don't like the maps pin. Maybe some small pattern in the shell or something around the turtle?

SteveMacenski, 2020-05-13:
Putting turtle in thread for visibility 

[image]

caelinsutch, 2020-05-13:
> Maybe some small pattern in the shell or something around the turtle?

For Turtle 4 or 5? That two bar linkage is hard to match. Maybe like some sort of robot treads? A simple google search for autonomous robot may raise some ideas

Shru, 2020-05-13:
@caelinsutch , @SteveMacenski , @shrijitsingh99 agree with all your comments especially keeping logo separate from name so we could use the logo all by itself. 

A company called [Fetch Robotics](https://fetchrobotics.com/wp-content/uploads/2018/06/WorldEconomicForum_TechnologyPionoeerFetchRobotics_FINAL.png) has a logo that is similar to Move it. Simple clean and speaks of their product. So I understand the inclination towards Move it. :D 

@caelinsutch love the color of logo 5. It works with any background, any shade of gray. The turtle also looks like its moving forward. 
As an inspiration, here are examples of subtle modifications to turtle: http://wiki.ros.org/Distributions.
May be a distance sensor in the front near the eyes or say, wheels in place of hind legs. 

I'll think of other forms of actuators/sensors and ways to represent them thematically.

SteveMacenski, 2020-05-13:
@caelinsutch I was talking about 5, the one I posted above. Of the 2, I definitely like this one better. Its less "busy" with alot of stuff going on. Just clean, simple. It is a little nondescript as to relationship to navigation though.

@Shru in my opinion, I wouldn't want something super mechanical like sensors just mounted on a turtle. The MoveIt and Fetch logos work well because they're simple geometry that relates to the subject. I don't think there's a simple way of doing that and would also make it pretty "busy". 

If we also came up with a number of suggestions, I might be able to rope together some money and hire a designer to make a logo / style guide on 99Designs like Dave mentioned above. I've also had good experiences with that in the past. 

There are many options available.

caelinsutch, 2020-05-13:
> May be a distance sensor in the front near the eyes or say, wheels in place of hind legs.

An actuator might look nice in place of one of the arms, like a robot turtle. 

@SteveMacenski maybe the shell could be a map or a lidar scan or something? Might look cool...

SteveMacenski, 2020-05-13:
Give it a try then!

SteveMacenski, 2020-05-15:
Maybe a weird idea, but what about a turtle that's made of voxels?

SteveMacenski, 2020-05-20:
Cool mechanical turtle here: https://discourse.ros.org/t/roscon-jp-2020-call-for-sponsors/12515/3?u=smac

ruffsl, 2020-10-01:
I'm a fan of sticking with `Nav2`, for at least keeping in close proximity with the established package names that folks may have grown accustomed to. When apt installing, `apt install ros-$ROS_DISTRO-nav<tab-tab>` is the first thing I see folks use to find what navigation packages ship with there current distro.

> Let's just roll Navigation into the feature set of MoveIt and call it a day ;-) MoveIt is already suppose to support base planning (and leg planning, and all types of planning).

We could also called it `Navit`? We could make the `<intransitive verb> +  it` a common project naming pattern across large projects within the ROS ecosystem. E.g. I've MoveIt, NavIt, SlamIt, FindIt installed, so my golden snitch seeking robot is ready for a test run. 

Technologic: https://youtu.be/fnWypQz6X2U

<details>

> Technologic
Daft Punk
Buy it, use it, break it, fix it,
Trash it, change it, mail, upgrade it,
Charge it, point it, zoom it, press it,
Snap it, work it, quick, erase it,
Write it, cut it, paste it, save it,
Load it, check it, quick, rewrite it
Plug it, play it, burn it, rip it,
Drag and drop it, zip, unzip it,
Lock it, fill it, curl it, find it,
View it, code it, jam, unlock it
Surf it, scroll it, pose it, click it
Cross it, crack it, twitch, update it,
Name it, read it, tune it, print it,
Scan it, send it, fax, rename it,
Touch it, bring it, pay it, watch it,
Turn it, leave it, stop, format it.

SteveMacenski, 2020-10-01:
@ruffsl the problem with that is even if the _stack_ is called Nav2, that doesn't mean the packages would be to be found with tab complete. In fact, I'd really like to remove the `nav2_` naming we use, I think that was mostly done in the early days because some of the packages existed in ROS1 and OSRF was (maintaining? developing? I don't actually know that state) a ROS Nav1 port to ROS2 so it needed the prefix to differentiate. 

That has since stopped, so we're the only `costmap_2d` package around.

ruffsl, 2020-10-01:
I think the prefix helps associate individual packages with the project/repo. And is perhaps one of the few ways we can uniquely name the packages with simple and straightforward descriptors within a global package namespace in ROS. E.g:

* `ament_*`
* `autoware_*`
* `control_*`
* `rosbag2_*`
* `rqt_*`
* `tb3_*`
* `tf2_*`
* `ros_ign _*`
* `webots_*`

If we end up creating new packages with scopes beyond navigation2, changing it's name prefix would be appropriate. But taking the canonical names for packages would be tricky, especially if they were already taken: we'd then have to prefix anyway.

SteveMacenski, 2020-10-27:
New logo has been selected and will be announced at ROS World 2020!

closed 2020-10-27 by SteveMacenski