ros-navigation/navigation2 #1489: Bug may have been introduced into simple action server

SteveMacenski, 2020-02-04:
After PR https://github.com/ros-planning/navigation2/pull/1356, we see a bunch of tests become flaky. I was able to track down the flakiness to the simple action servers not receiving action requests when called in quick succession. 

I added some sleeps between calls in the tests to patch it form 50% success to 80% success, but it should probably be dealt with. It seems to be caused by this PR and not from the upstream action server library. We should fix this ASAP.

Jconn, 2020-02-04:
I am getting a prohibitive amount of bt_navigator crashes off the head of `eloquent-devel` that look something like this:

```
[bt_navigator-10] [INFO] [bt_navigator]: Begin navigating from current location to (0.01, 0.08)
[ERROR] [bt_navigator-10]: process has died [pid 29440, exit code -11, cmd '/home/johnconn/ros2_outdoor_ws/install/nav2_bt_navigator/lib/nav2_bt_navigator/bt_navigator --ros-args -r __node:=bt_navigator --params-file /tmp/tmpffvcz4c4        '].

```
Maybe related? I'm going to look into it on Thursday or Friday

SteveMacenski, 2020-02-04:
Can't say -- personally, I don't use eloquent, I'm on master. Master should still work with eloquent binaries (that's my current development environment).

If you see those problems still in Master, let me know. This could just be the case that Eloquent-devel is a little stale. I filed a reminder #1490 to sync them up.

shivaang12, 2020-02-07:
I am having some what same problems with current master. After two-three preemption request I get this error. This is my log. Please ignore "[SHIVANG_DEBUG]"
```
[controller_server-4] [ERROR] [1581033129.528207137] [controller_server]: No valid trajectories out of 420! 
[controller_server-4] [WARN] [1581033129.528289747] [controller_server_rclcpp_node]: [follow_path] [ActionServer] Aborting handle.
[controller_server-4] [INFO] [1581033129.529241309] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[controller_server-4] [INFO] [1581033129.530341565] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[planner_server-5] [INFO] [1581033129.532002360] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[bt_navigator-7] [INFO] [1581033133.863067073] [bt_navigator]: Received goal preemption request
[bt_navigator-7] [INFO] [1581033133.863359744] [bt_navigator]: Begin navigating from current location to (-0.53, 0.96)
```
It gets stuck at above log and does nothing after that. If I hit the ctrl-c then
```
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[waypoint_follower-8] [INFO] [1581033262.238866324] [rclcpp]: signal_handler(signal_value=2)
[waypoint_follower-8] [INFO] [1581033262.259823894] [waypoint_follower]: Destroying
[lifecycle_manager-9] [INFO] [1581033262.238929132] [rclcpp]: signal_handler(signal_value=2)
[lifecycle_manager-9] [INFO] [1581033262.273417492] [lifecycle_manager_navigation]: Destroying
[INFO] [map_server-1]: process has finished cleanly [pid 22889]
[INFO] [lifecycle_manager-3]: process has finished cleanly [pid 22891]
[INFO] [controller_server-4]: process has finished cleanly [pid 22892]
[INFO] [planner_server-5]: process has finished cleanly [pid 22893]
[INFO] [recoveries_server-6]: process has finished cleanly [pid 22894]
[planner_server-5] [INFO] [1581033262.239017569] [rclcpp]: signal_handler(signal_value=2)
[lifecycle_manager-3] [INFO] [1581033262.242537858] [rclcpp]: signal_handler(signal_value=2)
[lifecycle_manager-3] [INFO] [1581033262.305224048] [lifecycle_manager_localization]: Destroying
[map_server-1] [INFO] [1581033262.242665425] [rclcpp]: signal_handler(signal_value=2)
[amcl-2] [INFO] [1581033262.242890786] [rclcpp]: signal_handler(signal_value=2)
[amcl-2] [INFO] [1581033262.292088963] [amcl]: Destroying
[recoveries_server-6] [INFO] [1581033262.243170057] [rclcpp]: signal_handler(signal_value=2)
[controller_server-4] [INFO] [1581033262.245335915] [rclcpp]: signal_handler(signal_value=2)
[bt_navigator-7] [INFO] [1581033262.287245170] [rclcpp]: signal_handler(signal_value=2)
[bt_navigator-7] [INFO] [1581033262.295232993] [DEBUG]: [SHIVANG_DEBUG] AFTER BTEngine Loop
[bt_navigator-7] [INFO] [1581033262.295252568] [bt_navigator]: [SHIVANG_DEBUG] AFTER the BT status
[bt_navigator-7] [INFO] [1581033262.320687074] [bt_navigator]: Destroying
[bt_navigator-7] [ERROR] [1581033262.295257636] [bt_navigator]: Navigation failed
[bt_navigator-7] [WARN] [1581033262.295274698] [bt_navigator]: [NavigateToPose] [ActionServer] Aborting handle.
[bt_navigator-7] [ERROR] [1581033262.315193521] [bt_navigator]: Action server failed while executing action callback: "Asked to publish result for goal that does not exist"
[INFO] [lifecycle_manager-9]: process has finished cleanly [pid 22909]
[INFO] [waypoint_follower-8]: process has finished cleanly [pid 22905]
[INFO] [amcl-2]: process has finished cleanly [pid 22890]
[ERROR] [bt_navigator-7]: process has died [pid 22901, exit code -11, cmd '/home/shivang/nav2_my_ws/install/nav2_bt_navigator/lib/nav2_bt_navigator/bt_navigator --ros-args -r __node:=bt_navigator --params-file /tmp/tmp1x3akgdk -r /tf:=tf -r /tf_static:=tf_static'].

```
What I noticed (with like 10 trials) is that when it finds "No valid trajectories out of 420! " then "follow path" action server drops "Aborting handle" log. I don't know much about how action server works but it creates problem and turtlebot does not move forward.

SteveMacenski, 2020-02-07:
What's in the control-c print outs that you're trying to highlight? It looks pretty typical to me, things just shutting down. 

For your `No valid trajectories out of 420! `,  have you debugged why that's happening? How may it relate to this ticket about the action server? 

I haven't seen this issue at the moment, but that may be also because I'm a little underwater trying to fix CI and documentation right now. I'd appreciate any digging you can do to narrow things down.

SteveMacenski, 2020-02-07:
In line with this ticket on the action servers, it may be upstream. 

I'm seeing most of the test issues due to:

```
  auto future_goal_handle = node_->action_client_->async_send_goal(goal);
  EXPECT_EQ(rclcpp::spin_until_future_complete(node_,
    future_goal_handle), rclcpp::executor::FutureReturnCode::SUCCESS);
```

and 

```
    auto future_result = client_->async_get_result(goal_handle_);
    rclcpp::spin_until_future_complete(node_lifecycle_, future_result);
```
spinning just never returning. I check with prints and the goal never makes it to the action server. So the async will return success, but the action server callback is never called. In the first case, it just spins forever trying to submit the goal.

SteveMacenski, 2020-02-07:
Found issue with an uninitiated boolean. It was registering with random values creating highly non-deterministic behavior that affected **all action servers**. This is a huge stability boost. Fix in #1509

SteveMacenski, 2020-02-07:
@shivaang12 @Jconn those may be issues -- please file new tickets for them

closed 2020-02-07 by SteveMacenski