ros-navigation/navigation2 #1500: turtlebot's not moving form the start pose when provided goal pose

shivaang12, 2020-02-06:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04 
- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior
Should run towards goal pose
#### Actual behavior
Controller-server say's 0 trajectory found from the start.
#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

shivaang12, 2020-02-06:
@SteveMacenski Do you know about this?

SteveMacenski, 2020-02-06:
You didn't fill out any of the ticket. I can't really do or say anything with that.

I however was testing with stuff this morning, so my guess is its something with your setup.

shivaang12, 2020-02-06:
It's working fine with my old repo.

SteveMacenski, 2020-02-06:
Feel free to circle back when you have more information to share.

shivaang12, 2020-02-06:
cool! If it's working for you then it's my system's fault. Thanks, I am closing the ticket then.

closed 2020-02-06 by shivaang12