ros-navigation/navigation2 #1516: reduce time it takes to process NavigateToPose request

Jconn, 2020-02-11:
merging in https://github.com/ros-planning/navigation2/pull/1515 adds another 200-300ms of latency to handling a NavigateToPose request, which is a lot. 

This comes from creating a new behavior tree every time a NavigateToPose request is accepted. That change is done to handle CoroActionNode's properly. But there should be a smarter way that doesn't take so much time.

SteveMacenski, 2020-02-15:
Did you file a ticket with BehaviorTree.CPP on the reset?

Jconn, 2020-02-15:
it is here https://github.com/BehaviorTree/BehaviorTree.CPP/issues/146

SteveMacenski, 2020-02-15:
Awesome, thanks for linking it. 

Is this ticket something you have a a business reason to look into, or should I put this in the pool of tasks to try to find some interested parties to look at?

Jconn, 2020-02-15:
I would put it in the pool - we aren't prioritizing performance issues yet

SteveMacenski, 2020-02-15:
Ok, thanks for the update.

facontidavide, 2020-03-18:
Hi,

I probably need to join this. I am sorry that my library is creating issues of this kind. I will try to dedicate time to this in the next days.

SteveMacenski, 2020-03-25:
#1606 resolves

closed 2020-03-25 by SteveMacenski