ros-navigation/navigation2 #1519: simple_action_server race condition, deadlock

Jconn, 2020-02-13:
## Bug report

**Required Info:**

- Operating System:
ubuntu 18.04
- Version or commit hash:
hash 2a37b51aa (master on 02/11)
- DDS implementation:
Fast-RTPS
#### Steps to reproduce issue
I'm hitting this when sending NavigateToPose goals in a script that fires off a new goal as soon as the previous goal succeeds. 

There's a race condition between when the `work()` function finishes and when a new goal begins executing.

What I think is happening: 

NavigateToPose request is accepted, 
simple_action_server [spawns  thread A](https://github.com/ros-planning/navigation2/blob/master/nav2_util/include/nav2_util/simple_action_server.hpp#L135) to handle the work
`          execution_future_ = std::async(std::launch::async, [this]() {work();});
`

thread A is [finishing the goal](https://github.com/ros-planning/navigation2/blob/master/nav2_bt_navigator/src/bt_navigator.cpp#L218), and calls
`      action_server_->succeeded_current();
`

then thread A is pre-empted, and a new request comes in, and the server accepts it
we enter `handle_accepted`, and grab `update_mutex_`, there is no active goal, simple_action_server again calls
`          execution_future_ = std::async(std::launch::async, [this]() {work();});
`

This fires off the destructor of the previous object that `execution_future_` was referencing, 
[which causes it to block on thread A completing](https://stackoverflow.com/a/27992092/10981836). `handle_accepted` is still holding onto  `update_mutex_`

`execute_callback_()` in thread A returns, and thread A blocks when [trying to lock](https://github.com/ros-planning/navigation2/blob/master/nav2_util/include/nav2_util/simple_action_server.hpp#L154) `update_mutex_`

SteveMacenski, 2020-02-13:
Does this describe your issue https://github.com/ros2/rclcpp/issues/861?

Jconn, 2020-02-13:
I think its an issue with the nav wrapper. After I hit this case, the navstack 'zombies' and doesn't respond to new NavigateToPose goals, and also stops logging anything. I'm working on a fix

crdelsey, 2020-02-13:
Looking forward to the PR. In general, I think our current simple_action_server/client code is overly complex and prone to race conditions. I'd like to rewrite at some point. Since you're looking at this specifically today, I'd be interested in any thoughts on alternate architecture for this component if any ideas come up.

closed 2020-02-13 by SteveMacenski