ros-navigation/navigation2 #1521: RGBD intersects laser scan

KadynCBR, 2020-02-14:
## Bug report

**Required Info:**

- Operating System:
  - 
  - Ubuntu 18.04 -- Eloquent
- Version or commit hash:
  - 
  - 1f6d5a509865936c0537b1a454ae656d2dfe1783
- DDS implementation:
  - 
  - Fast-RTPS
#### Steps to reproduce issue

```
export TURTLEBOT3_MODEL=waffle
TB3_GAZEBO_PATH=`ros2 pkg prefix turtlebot3_gazebo`
export GAZEBO_MODEL_PATH=$TB3_GAZEBO_PATH/share/turtlebot3_gazebo/models:$GAZEBO_MODEL_PATH
ros2 launch nav2_bringup tb3_simulation_launch.py
```

#### Expected behavior
[image]

#### Actual behavior
[image]

#### Additional information

On a fresh install running the commands below will result in the robot spawning in gazebo but the laser scan will be 'flickering' This becomes problematic because the nav goal gets rejected. 

Upon closer inspection it seems the collision for the realsense is incorrectly placed (And this lines up where the flicker is happening so the laser is hitting the RGBD.: 
[image] 

Moving the collision of the RGBD camera down seems to fix this:
[image]

I can PR this if needed just wanting to confirm this

fmrico, 2020-02-14:
Hi @KadynCBR 

I experienced the same problem. I change the robot model to raise the laser a little bit.

I can do a PR, but I would like to know more opinions also.

SteveMacenski, 2020-02-14:
1. Very good ticket. Thank you. Pictures tell a thousand words.
2. You're 100% right, this is a problem, a simple shifting up on the URDF should give that a fix - do submit a PR.

I saw the fluttering of the laser scan for a year and just assumed it was Gazebo's noise model at work or the front end just not projecting all the beams before the next one came along. I never actually looked into it.

closed 2020-02-14 by SteveMacenski