ros-navigation/navigation2 #1530: Check up on nightly builds to ensure they're turning over again

SteveMacenski, 2020-02-15:
From #1527 we need to see if we're fixing the issue. Retriggering nightly on the day of isnt getting the recent commit hashes, so I'll have to wait a day between them without digging into the CircleCI code more to clear the cache and build. No worries, I'm not in a hurry.

SteveMacenski, 2020-02-16:
Update: Changes for the system tests make them now no longer fail on shutdown, but are not returning so the test is timing out.

PR #1535 fixes this as well for the waypoint follower, but that has an explicit `exit(0)` so I'm curious if tomorrow's nightly build will have that one still fail or not. If not, then we can add the explicit call and be done with it, but I kind of doubt it'll be that trivial

SteveMacenski, 2020-02-18:
https://8954-135363400-gh.circle-artifacts.com/0/opt/overlay_ws/log/test/nav2_system_tests/stdout.log

Same issue still with waypoint follower. All tests are passing but they're failing the nightly builds by just not returning cleanly.

SteveMacenski, 2020-02-18:
@ruffsl any ideas why the nightly builds would act differently than the normal CI / master branch builds? This is not seen in either of those in CircleCI.

ruffsl, 2020-02-18:
> any ideas why the nightly builds would act differently than the normal CI / master branch builds?

Nightly shouldn't act different than any merge commit that's tested after it's pushed to master.

https://github.com/ros-planning/navigation2/blob/4836c7ff40e28497b350a5cc67e2e0a26488df51/.circleci/config.yml#L468-L483

Nightly also has the same type of jobs as build_and_test (but more of them), only a different trigger:

https://github.com/ros-planning/navigation2/blob/4836c7ff40e28497b350a5cc67e2e0a26488df51/.circleci/config.yml#L484-L490

I'm currently seeing green on the readme:

[image]

Though I just re-triggered commit job for 4836c7f

SteveMacenski, 2020-02-18:
I'm not sure what that is pulling from. There are times (most times) when at least 1 of the tests fail when merging into master, which I would think should make it say failing, but it'll be passing. Other times, it'll be failed with the same conditions. Is it related to nightly at all? That's not immediately clear to me.

The nightly for Cyclone and Connext always fail this way, but we also don't run either of those in normal CI, so maybe why its being seen in nightly. I didn't think the way Circle dispatched jobs should matter, but I am starting to grasp at straws.

ruffsl, 2020-02-19:
> I'm not sure what that is pulling from.

From the "Checkout code" step, you can verify the commit hash it clones/checkout:

```
Cloning into '.'...
Warning: Permanently added the RSA host key for IP address '140.82.113.3' to the list of known hosts.
remote: Enumerating objects: 138, done.        
remote: Counting objects: 100% (138/138), done.        
remote: Compressing objects: 100% (102/102), done.        
remote: Total 20573 (delta 60), reused 71 (delta 30), pack-reused 20435        
Receiving objects: 100% (20573/20573), 37.51 MiB | 39.08 MiB/s, done.
Resolving deltas: 100% (13493/13493), done.
HEAD is now at b3f079c3 adding robots table, clearing up ros1 conversion and adding dox files (#1549)
HEAD is now at b3f079c3 adding robots table, clearing up ros1 conversion and adding dox files (#1549)
```
> https://app.circleci.com/jobs/github/ros-planning/navigation2/9019/parallel-runs/0/steps/0-102

> The nightly for Cyclone and Connext always fail this way, but we also don't run either of those in normal CI, so maybe why its being seen in nightly.

The workflows named "build_and_test" only runs a release and debug build/test using the default RMW (FastRTPS currently), while the "nightly" workflow also tests Cyclone and Connext.

I think I may see the issue in your use of the web UI vs your expectations:

> Retriggering nightly on the day of isnt getting the recent commit hashes

If you go to an old nightly job, and click on "Rerun" then that rerun will rerun the exact same workflow on the exact same commit, reproducing the same result (in most of our cases). That rerun button is helpful if there was something flaky, a remote resource couldn't be fetched from the web, or if you wanted "Return to job with SSH" to debug the same commit manually. Having "Rerun" pull from the head of the branch instead of exact commit from before would be as helpful for the scenario just mentioned, like ssh debugging.

Instead, if you like to trigger a fresh job manually, pointed the head of branch or any commit in particular, I think you could use API via curl:

* https://circleci.com/docs/2.0/api-job-trigger/
* https://circleci.com/docs/2.0/triggers/

Though, for (nightly) workflows (vs individual jobs in a workflow), I think the docs say you have to use the Trigger a Build by Project endpoint:

* https://circleci.com/docs/api/#trigger-a-new-build-by-project-preview

SteveMacenski, 2020-02-19:
I'm saying I'm not what the CI badge is pulling from.

ruffsl, 2020-02-19:
>I'm saying I'm not what the CI badge is pulling from.

Ok, I had no idea this was all about badges. I'd guess that the depicted state of the badge derives from the success or failure of the most recent job on the specified branch. 

https://circleci.com/docs/2.0/status-badges

Though perhaps there's some latching issues in hosting the badge endpoint, as both circlci and shields.io show the same status, although the last workflow on the repo has failed. 

```
[![CircleCI](https://circleci.com/gh/ros-planning/navigation2/tree/master.svg?style=svg)](https://circleci.com/gh/ros-planning/navigation2/tree/master)
```
[image]

```
![CircleCI](https://img.shields.io/circleci/build/github/ros-planning/navigation2/master)
```
[image]

Perhaps it has something to do with the recent commits with [website sprint3] being tripped by the badge suppressing logic else ware for [skip ci]:
https://discuss.circleci.com/t/status-badges-for-skipped-builds-show-as-failed/30182/5

However, it seems all badges cached by github are currently acting a bit strange:
[image]

SteveMacenski, 2020-02-19:
Huh, that's unusual. Interesting.

Anyhow, that's here nor there on this ticket -- this is on why cyclone / connext decided that they dislike me. I poked Joe to see if he or his team had thoughts.

SteveMacenski, 2020-03-07:
Fixed!

Now what's left is usual flakiness of tests. Will make new ticket for that.

closed 2020-03-07 by SteveMacenski