ros-navigation/navigation2 #1533: Tutorials for creating all plugins

SteveMacenski, 2020-02-15:
Need a set of tutorials with examples of creating new plugins. These shouldn't take long and should be very simple implementations as an example

- [ ] costmap 2d: gradient layer
- [ ] Planner: RRT-connect
- [ ] Controller: Pure pursuit 
- [ ] Recovery: Send a slack / sms message
- [ ] BT: Call new server to trigger an external action

SteveMacenski, 2020-02-15:
CC #1524 @AlexeyMerzlyakov I'm just moving the discussion over here so that its all together.

AlexeyMerzlyakov, 2020-02-17:
I've submitted a PR for a `gradient_layer` plugin. In terms of this issue, there are might be need to update a directory structure to put all plugins' tutorials in one place and rename `writing_new_costmap2d_plugin` to something else as well. But I do not know your original intention, so remained the commit as-is in order to start a concrete discussion about.

SteveMacenski, 2020-02-17:
Awesome, I'll take a look tomorrow (today is an American holiday). One thing I did notice if that you included the C++ files in the documentation itself. 

Can you create a ros package `nav2_gradient_costmap_plugin` or something similar and put this info in the newly created navigation2_tutorials repo? This lets us separate out tutorial code from the main codebase, and also creating a full stand-along package with just 1 plugin will give people a good example of how to deal with the plugin registration, linking to the headers, etc. Also people can then clone the package and use it as an example starting point for their application.

https://github.com/ros-planning/navigation2_tutorials

AlexeyMerzlyakov, 2020-02-21:
Thank you for so detailed review! I've made all changes you've mentioned in the PR #1451 and re-submitted new patch there.

ardabbour, 2020-03-15:
I would like do the controller/planner tutorial, but is there a simple sim-based navigation testbed?

SteveMacenski, 2020-03-15:
Hi! 

If you look at the getting started page, you’ll see how to run the TB3 in simulation! 

https://ros-planning.github.io/navigation2/

shivaang12, 2020-04-02:
@SteveMacenski I can write tutorial for RRT. Let me know if its taken.

SteveMacenski, 2020-04-02:
If you make it, its not taken ;-) 

I got side-tracked. I'm working on a planner and intended to write a tutorial on a basic implementation of it but I got sucked into the real thing and haven't been spit back out yet. 

I think for RRT, OMPL will make your life easier and give a demo of OMPL in navigation. I don't know if @davetcoleman has any input on that though.

this is someone's implementation, I see some missing / TODO / wrong things, but there's alot right here too : https://github.com/windelbouwman/move-base-ompl/ (like working in the velocity space rather than XY space)

shivaang12, 2020-04-02:
@SteveMacenski thanks! I think using OMPL would be a good thing to start, it will be easy to make and also it will pose a great example to show how to wrap an existing planner library for Nav 2.

SteveMacenski, 2020-04-02:
Well the point of the tutorial is to show how to create a global planner plugin, not explain under the hood of how you created it. 

For instance, look at Alexey’s in costmap. It creates some gradient pattern that doesn’t matter as a toy example. 

If you were To create an RRT example, my feeling is people would just use it so it could be made well. Keep in mind that pkg doesn’t properly encapsulate the library In a non-ROS class so it looks more complex. The Ompl stuff should be wrapped into a class with simpler interfaces and used in the plugin. See NavFn for an example.

shivaang12, 2020-04-02:
Okay! Got it!

SteveMacenski, 2020-04-02:
My recommendation: if your goal is to build a tutorial, a good toy example is to just make a straight line path. If your goal is to work with OMPL, maybe you should just make a legit global planner and we can worry about the tutorial later. 

If your goal is both, awesome! 😎

shivaang12, 2020-04-09:
@SteveMacenski I was writing RRTConnect plugin and I found that the costmap query return 0 when querying the unexplored state. Shouldn't be 255? Can you guide me if this is due to default setting of the costmap or something?

Thanks!

SteveMacenski, 2020-04-09:
If you'd like to treat unknown space differently than free space, you need to set the parameter `track_unknown_space` for the costmap to true (in fact, I dont remember if we include that at all in the default configs, maybe we shuld add that to the .yaml file for people to see that). That will differentiate unknown from free.

Then for something like NavFn, there's a param for `allow_unknown` to then ask the question: "if you see unknown space, should you treat it as navigable or not?" You should also add that to your planner.

ardabbour, 2020-04-12:
Regarding the global planner tutorial, I guess I'm a bit late to the party, but I wrote a path planner cpp class that I intend for to be only dependent on OMPL [here](https://github.com/ardabbour/omplgp), for which ROS1 and ROS2 global planner plugins (which I'm currently working on) can be written. I had a plan to patch together some excrepts and create a tutorial. Can it be of some assistance here?

SteveMacenski, 2020-04-14:
@ardabbour you should take a look at https://github.com/ros-planning/navigation2_tutorials/pull/2 and review your thoughts. 

Maybe it would make alot of sense for you and @shivaang12 to work together on building a high-quality OMPL navigation planner.

SteveMacenski, 2020-08-26:
I'm working on the BT node now

SteveMacenski, 2020-08-27:
Did recoveries too. @shrijitsingh99 if we can finish up your PR for the controller we'd be done with this ticket and have an example for every major plugin :1st_place_medal:

davetcoleman, 2020-08-27:
OMPL is generic and can be used for any kind of planning, including nav. @mamoll is a good one to have input on this.

SteveMacenski, 2020-08-27:
There's another ticket https://github.com/ros-planning/navigation2/issues/1646 where we're discussing adding an OMPL planner. We've had a couple of folks working on that part time for awhile now and so far haven't been able to get acceptable performance. I'd love to have an OMPL planner available, but it needs to be suitable for realistic use.

closed 2020-09-16 by SteveMacenski