ros-navigation/navigation2 #1556: reduce missing frame warning output

relffok, 2020-02-21:
## Feature request

#### Feature description
When running the nav2 bringup I run into the problem that a lot of missing map frame warning messages are published **until a 2D pose is estimated**, like the following: 

`[planner_server-13] Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist`

These warnings come up in no time and stop when a 2D pose was estimated (obviously) but they make debugging other things really hard. Is it possible by any chance to only publish this warning once or with a **different frequency**?

SteveMacenski, 2020-02-21:
Good timing, I merged that a few days ago https://github.com/ros-planning/navigation2/pull/1553

relffok, 2020-02-21:
Oh I didn't see that one. Great, thanks!

closed 2020-02-21 by relffok